gbionics/bipedal-locomotion-framework #24: Create a custom cmake function for compiling the components

GiulioRomualdi, 2020-03-11:
Every time a new component is added the very similar `CMakeLists.txt` has to be copied from one component to the other (i.e. [here](https://github.com/dic-iit/bipedal-locomotion-controllers/blob/e26ec3444e7837f17b3cdd0156ed15690ff08496/src/Estimators/CMakeLists.txt) or [here](https://github.com/dic-iit/bipedal-locomotion-controllers/blob/master/src/ParametersHandler/CMakeLists.txt)). I really don't like this copy-paste of code because it may lead to errors.   
I think that it could be a nice idea to implement a `CMake` function similar to the one written by @S-Dafarra for the [`unit tests` ](https://github.com/dic-iit/bipedal-locomotion-controllers/blob/master/cmake/AddBipedalLocomotionControllersUnitTest.cmake).

cc @traversaro

S-Dafarra, 2020-03-11:
I agree. I guess that the main difference would be between libraries and executables, but this may be an option to a hypothetic function.

traversaro, 2020-03-11:
I think it is a good idea. The amount of duplicate boilerplate CMake code for each library is indeed risky and prone to errors. We did something similar in gazebo-yarp-plugins, and the only reason we did not do that for https://github.com/robotology/how-to-export-cpp-library is that it would complexify readability for a tutorial such as that..

traversaro, 2020-03-11:
Interesting takes on "we need to reduce the complexity and variability of C++ libraries build system": 
* https://vector-of-bool.github.io/2020/01/06/new-decade.html
* https://api.csswg.org/bikeshed/?force=1&url=https://raw.githubusercontent.com/vector-of-bool/pitchfork/develop/data/spec.bs

PR #25 (2020-03-12): Add a custom cmake function for enhancing the components compilation
This PR implements the `add_bipedal_component` cmake function. 
This should closes #24
closed 2020-03-12 by GiulioRomualdi