gbionics/bipedal-locomotion-framework #30: CI integration is failing on Windows

GiulioRomualdi, 2020-03-23:
The Windows CI integration is failing with the following error
```
Starting package 2/8: ace:x64-windows
Building package ace[core]:x64-windows...
-- Downloading https://github.com/DOCGroup/ACE_TAO/releases/download/ACE%2BTAO-6_5_8/ACE-src-6.5.8.zip...
-- Extracting source C:/vcpkg/downloads/ACE-src-6.5.8.zip
-- Using source at C:/vcpkg/buildtrees/ace/src/ACE-src-6-a3686a23cb
-- Downloading http://strawberryperl.com/download/5.30.0.1/strawberry-perl-5.30.0.1-32bit.zip...
-- Building C:/vcpkg/buildtrees/ace/src/ACE-src-6-a3686a23cb/ace/ace.sln for Release
CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:72 (message):
    Command failed: msbuild C:/vcpkg/buildtrees/ace/src/ACE-src-6-a3686a23cb/ace/ace.sln /p:Configuration=Release /t:Rebuild /p:Platform=x64 /p:PlatformToolset=v142 /p:VCPkgLocalAppDataDisabled=true /p:UseIntelMKL=No /p:WindowsTargetPlatformVersion=10.0.18362.0 /m /p:ForceImportBeforeCppTargets=C:/vcpkg/scripts/buildsystems/msbuild/vcpkg.targets /p:VcpkgTriplet=x64-windows
    Working Directory: C:/vcpkg/buildtrees/ace/x64-windows-rel
    Error code: 1
    See logs for more information:
      C:\vcpkg\buildtrees\ace\build-x64-windows-rel-out.log

Call Stack (most recent call first):
  scripts/cmake/vcpkg_build_msbuild.cmake:119 (vcpkg_execute_required_process)
  ports/ace/portfile.cmake:98 (vcpkg_build_msbuild)
```

Running `vcpkg.exe update` should solve the problem

https://github.com/dic-iit/bipedal-locomotion-controllers/runs/527453730

traversaro, 2020-03-23:
Interestingly, I had the same error in iDynTree even using the  vcpkg tag 2020.01, so it is something that changed in GitHub Actions Windows image, not only on vcpkg.

traversaro, 2020-03-23:
Upstream issue: https://github.com/actions/virtual-environments/issues/605 .

S-Dafarra, 2020-03-23:
Interesting PRs:
- https://github.com/robotology/whole-body-estimators/pull/52
- https://github.com/robotology/idyntree/pull/655

traversaro, 2020-03-23:
It turns out that there is a bug in Visual Studio 2019 16.5 that prevents  to correctly compile ACE ( see https://github.com/actions/virtual-environments/issues/605#issuecomment-602772318 and https://developercommunity.visualstudio.com/content/problem/954522/internal-compiler-error-1925286104.html), and GitHub Actions recently updated to VS2019 16.5. The fix has been already prepared, but it will be only released with Visual Studio 2019 16.6 . For the time being, there are two possible workaround: 
* Remove VS2019 16.5 from the GitHub Action image, as suggested in https://github.com/actions/virtual-environments/issues/605#issuecomment-602772318
* Use a pre-compiled vcpkg archive, as done in https://github.com/robotology/idyntree/pull/655

traversaro, 2020-03-24:
Another workaround I used in https://github.com/iit-danieli-joint-lab/idjl-software-dependencies-vcpkg/pull/9 (that is one repo used to produce the pre-compiled vcpkg archives) is to bootstrap the vcpkg used in the CI with a vcpkg compiled locally that just contains as installed port `ace:x64-windows` .

closed 2020-03-24 by GiulioRomualdi