gbionics/bipedal-locomotion-framework #34: Unable to find the ParameterHandler yarp implementation

GiulioRomualdi, 2020-03-30:
The header files of the `ParametersHandlerYarpImplementation` component are installed in
```
-- Up-to-date: /home/gromualdi/robot-code/robotology-superbuild/build/install/include/BipedalLocomotionControllers/ParametersHandlerYarpImplementation/YarpImplementation.h
-- Up-to-date: /home/gromualdi/robot-code/robotology-superbuild/build/install/include/BipedalLocomotionControllers/ParametersHandlerYarpImplementation/YarpImplementation.tpp
```
This is in contrast with the `#include` directive in 
https://github.com/dic-iit/bipedal-locomotion-controllers/blob/1835a1d1f5e9bc5e080447c12ec934e64484ca52/src/ParametersHandler/YarpImplementation/include/BipedalLocomotionControllers/ParametersHandler/YarpImplementation.tpp#L14

The bug has been introduced by https://github.com/dic-iit/bipedal-locomotion-controllers/pull/25. Indeed now the files are installed in a folder named as the component (i.e. `ParametersHandlerYarpImplementation`)

I would change the path and the file name 
```
BipedalLocomotionControllers/ParametersHandlerYarpImplementation/YarpImplementation.h
```
into
```
BipedalLocomotionControllers/ParametersHandlerYarpImplementation/Implementation.h
```

@traversaro @S-Dafarra

S-Dafarra, 2020-03-30:
I would also explore solutions which could allow us to keep the current include structure. I think that
```cpp
#include <BipedalLocomotionControllers/ParametersHandler/YarpImplementation.h> 
```
is pretty neat.

What about modifying the CMake structure to allow a custom install directory?

GiulioRomualdi, 2020-03-30:
> What about modifying the CMake function to allow a custom install directory?

deal

S-Dafarra, 2020-03-30:
Great! Anyways, I don't know if it is possible, but in principle, you could make that parameter optional. In this way, it would be sort of retro-compatible :thinking:

PR #37 (2020-03-30): Add the possibility to specify the installation folder AddBipedalLocomotionControllersComponent function
This PR closes #34. In details:
-  `AddBipedalLocomotionControllersComponent` cmake function accepts also the name of a custom installation folder. If the folder is not specified the default one will be used. The name of the default folder is equal to name of the component.
- The header of `ParameterHandlerYarpImplementation` are now installed in `BipedalLocomotionControllers/ParameterHandler`
closed 2020-03-30 by GiulioRomualdi