gbionics/bipedal-locomotion-framework #39: Introduce a Signal block

GiulioRomualdi, 2020-04-23:
I noticed that every time I need to read the data from a `yarp port`, I always need the following ingredients
```cpp
yarp::os::BufferedPort<T> port; 
U data; /**< Vector containing the data. */
std::unique_ptr<Smoother> smoother; 
std::mutex mutex // <---- Useful if we are going to introduce threads in the final application 
```
Where the `Smoother` is an optional class that smooth the data. In general a low pass filter or a minimum jerk trajectory generator. 

Last but not least, a signal may be streamed not only through a yarp port. For instance, the signal may arrive from a file (i.e. @prashanthr05 dataset for testing the estimation)

We can try to abstract the concept of the signal and introduce an interface for streaming data through the network/logfile. 

A possible idea is to adapt what @xenvre did in [robots-io](https://github.com/xEnVrE/robots-io).

traversaro, 2020-04-23:
For reference, this is also similar to the YARPRead block available in WB-Toolbox: 
* https://github.com/robotology/wb-toolbox/blob/v5/toolbox/library/src/YarpRead.cpp

traversaro, 2020-04-23:
Also Drake's LCM systems are related (they also include a system to playback logs): 
* https://github.com/RobotLocomotion/drake/tree/master/systems/lcm