gbionics/bipedal-locomotion-framework #44: Discussion on the possible use of QT in the framework.

S-Dafarra, 2020-05-13:
While working on a GUI for a separate project, I started using QT. Let me start with the premise that I am not an expert at all with this framework and I just started learning it, but along the way, I started appreciating some features that I believe may be useful in our case. 

First of all, Qt is pretty easy and tailored to be used within a ``CMake`` project while having multi-platform support. In addition, it is possible to use both in C++ and Python. The documentation is very detailed and, even if there may be some dark spots, online you can find countless tutorials and answers on Stack Overflow on 95% of the possible doubts you may have while using it. Nevertheless, these are not the reasons why I considered using it in this project.

One of the core features in QT is the definition of signals and slots: https://doc.qt.io/qt-5/signalsandslots.html. These allow defining an interface between different objects. Once a signal is "emitted", the corresponding slots on the other objects are called. On their own, they are not so much different from normal interfaces and methods call. The interesting thing is that multiple objects may be called after a single "signal". In addition, threading becomes easy. See for example https://doc.qt.io/qt-5/qthread.html#details where an object can be put in a separate thread with a single method call. In addition, if there is no shared data between objects (they do not point to a third object) ``mutex`` are not even necessary (see https://doc.qt.io/qt-5/threads-synchronizing.html). 

An example in our case would be the handling of planners and controllers. Planners can be in a separate thread, and once done, they may simply "emit" a signal which is handled by the controller class. Without forgetting that Qt is used to build GUIs. The signal emitted by the planner may be also detected by a logger which displays the trajectories on a GUI.

One of the downsides is that templates are not digested well in a Qt object. See for example https://stackoverflow.com/questions/4397478/qt-templated-q-object-class. The main reason is that Qt requires to autogenerate a lot of stuff, hence the definition of templates makes this passage prohibitive.

I wanted to use this issue to trigger a discussion on this, even if it is not clear to me either which first step we may eventually take in this direction. Also, this partially overlaps with #39.

@traversaro @GiulioRomualdi @prashanthr05 what do you think?

fjandrad, 2020-05-13:
cc @fjandrad

traversaro, 2020-05-13:
Just to understand, were you thinking of using signal and slots in the libraries such as  the Planner library, or just to use them in specific executable programs, such as the  https://github.com/robotology/walking-controllers/tree/master/src/WalkingModule ?  

Furthermore, while Qt5 is a really nice library under a lot of aspects,  it has certain aspects (its license, not being trivial to install on all  OSs) for which it may not be ideal to have it as a  dependency for a non-GUI tool. One possible option, as the Signal&Slot is quite a general pattern, it to use a  dedicated  library that implements the Signal&Slot pattern, for example:
* https://github.com/TheWisp/signals
* https://github.com/tripleslash/rocket
* https://github.com/NoAvailableAlias/nano-signal-slot

S-Dafarra, 2020-05-13:
> Just to understand, were you thinking of using signal and slots in the libraries such as the Planner library, or just to use them in specific executable programs, such as the https://github.com/robotology/walking-controllers/tree/master/src/WalkingModule ?

I was thinking not to use it within a library necessarily, but when putting things together like in the ``WalkingModule`` maybe.

Indeed, the main appeal was the signal/slot paradigm. Those alternatives are very interesting indeed. What I liked about Qt5 was the possibility to insert GUIs where needed, an aspect that we often overlook in my opinion. Not that adding GUIs is simple anyways :grin:

S-Dafarra, 2020-05-21:
I think we can close this for the time being. The main aim was to start the discussion and to make people aware. The main takeaway is that when we will be going to put things together, QT or other signal/slot libraries may be useful to keep the communication between blocks clean.

closed 2020-05-21 by S-Dafarra

traversaro, 2020-05-25:
Related blog post: https://medium.com/@julienjorge/testing-c-signal-slot-libraries-1994eb120826 .

S-Dafarra, 2020-05-25:
> Did you know that every day several new C++ signal-slot libraries are created? There are so many of them that we estimate there is 1.14 libraries for each C++ developer.

:joy: :joy: