gbionics/bipedal-locomotion-framework #67: Start thinking of using MeshCat for visualizing the floating base system

GiulioRomualdi, 2020-07-08:
The visualization is appealing and modern. In this issue, I would like to track the investigation of porting [`MeshCat`](https://github.com/rdeits/MeshCat.jl) in our framework

traversaro, 2020-07-08:
Related issues: 
* https://github.com/search?q=org%3Adic-iit+meshcat&type=Issues
* https://github.com/search?q=org%3Aicub-tech-iit+meshcat&type=Issues

traversaro, 2020-07-10:
One cool feature I was not aware of and I just saw used  in https://github.com/RobotLocomotion/drake/pull/13666 is the possibility of exporting 3D animations as self contained `.html` files that can be navigated played and restarted by anyone with a browser, see for example: http://people.csail.mit.edu/russt/uploads/quadrotor_lqr.html . I am not sure how it works, but it seems cool.

GiulioRomualdi, 2023-06-26:
The visualizer is used for  the example https://github.com/ami-iit/bipedal-locomotion-framework/blob/ba95103f95745ae604ff14dc4d191cd07e18ca6e/examples/InverseKinematics/inverse_kinematics.ipynb 

We can consider this activity concluded

closed 2023-06-26 by GiulioRomualdi