gbionics/bipedal-locomotion-framework #88: [YCM] CI integration is failing in all the platforms

GiulioRomualdi, 2020-08-31:
The error is due to `YCM`

```
2020-08-31T08:46:01.5203225Z Cloning into 'ycm'...
2020-08-31T08:46:02.1291538Z Note: switching to 'a1fc6ad0d76ae18c8cce638c72516dcf1d7665cc'.
2020-08-31T08:46:02.1291668Z 
2020-08-31T08:46:02.1292222Z You are in 'detached HEAD' state. You can look around, make experimental
2020-08-31T08:46:02.1292557Z changes and commit them, and you can discard any commits you make in this
2020-08-31T08:46:02.1292716Z state without impacting any branches by switching back to a branch.
2020-08-31T08:46:02.1292808Z 
2020-08-31T08:46:02.1292944Z If you want to create a new branch to retain commits you create, you may
2020-08-31T08:46:02.1293261Z do so (now or later) by using -c with the switch command. Example:
2020-08-31T08:46:02.1293357Z 
2020-08-31T08:46:02.1293625Z   git switch -c <new-branch-name>
2020-08-31T08:46:02.1293817Z 
2020-08-31T08:46:02.1293948Z Or undo this operation with:
2020-08-31T08:46:02.1294030Z 
2020-08-31T08:46:02.1294267Z   git switch -
2020-08-31T08:46:02.1294351Z 
2020-08-31T08:46:02.1294489Z Turn off this advice by setting config variable advice.detachedHead to false
2020-08-31T08:46:02.1294572Z 
2020-08-31T08:46:03.1888255Z -- YCM Version: 0.11.0 (0.11.0)
2020-08-31T08:46:03.1909899Z CMake Warning (dev) at /usr/local/share/cmake-3.17/Modules/GNUInstallDirs.cmake:225 (message):
2020-08-31T08:46:03.1910311Z   Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
2020-08-31T08:46:03.1910574Z   target architecture is known.  Please enable at least one language before
2020-08-31T08:46:03.1910817Z   including GNUInstallDirs.
2020-08-31T08:46:03.1911050Z Call Stack (most recent call first):
2020-08-31T08:46:03.1911261Z   CMakeLists.txt:40 (include)
2020-08-31T08:46:03.1911700Z This warning is for project developers.  Use -Wno-dev to suppress it.
2020-08-31T08:46:03.1911948Z 
2020-08-31T08:46:03.2040330Z CMake Warning (dev) at find-modules/CMakeLists.txt:64 (message):
2020-08-31T08:46:03.2040978Z   A "FindGLUT.cmake" module already exists in CMAKE_ROOT.  Please remove it,
2020-08-31T08:46:03.2041373Z   or move it to proposed.
2020-08-31T08:46:03.2042042Z This warning is for project developers.  Use -Wno-dev to suppress it.
2020-08-31T08:46:03.2042350Z 
2020-08-31T08:46:03.2042978Z CMake Warning (dev) at find-modules/CMakeLists.txt:64 (message):
2020-08-31T08:46:03.2043378Z   A "FindGSL.cmake" module already exists in CMAKE_ROOT.  Please remove it,
2020-08-31T08:46:03.2043730Z   or move it to proposed.
2020-08-31T08:46:03.2044347Z This warning is for project developers.  Use -Wno-dev to suppress it.
2020-08-31T08:46:03.2044625Z 
2020-08-31T08:46:03.2045302Z CMake Warning (dev) at find-modules/CMakeLists.txt:64 (message):
2020-08-31T08:46:03.2045694Z   A "FindOpenGL.cmake" module already exists in CMAKE_ROOT.  Please remove
2020-08-31T08:46:03.2046046Z   it, or move it to proposed.
2020-08-31T08:46:03.2046665Z This warning is for project developers.  Use -Wno-dev to suppress it.
2020-08-31T08:46:03.2046968Z 
2020-08-31T08:46:03.3536555Z -- Configuring done
2020-08-31T08:46:03.4278165Z -- Generating done
2020-08-31T08:46:03.4329061Z -- Build files have been written to: /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/ycm/build
2020-08-31T08:46:03.4501023Z Scanning dependencies of target modules
2020-08-31T08:46:03.4824884Z [  2%] Files from modules installed in CMake build directory
2020-08-31T08:46:03.4867509Z [  2%] Built target modules
2020-08-31T08:46:03.4909742Z Scanning dependencies of target find-modules
2020-08-31T08:46:03.5802942Z [  5%] Files from find-modules installed in CMake build directory
2020-08-31T08:46:03.5846430Z [  5%] Built target find-modules
2020-08-31T08:46:03.5888423Z Scanning dependencies of target cmake-proposed
2020-08-31T08:46:03.6100300Z [  8%] Files from cmake-proposed installed in CMake build directory
2020-08-31T08:46:03.6143227Z [  8%] Built target cmake-proposed
2020-08-31T08:46:03.6183877Z Scanning dependencies of target 3rdparty-qt-gstreamer
2020-08-31T08:46:03.6243005Z [ 10%] Downloading file cmake/modules/FindGObject.cmake from qt-gstreamer git repository (robotology-dependency fork) (ref 383c41d310b14c328ab6d5d02f478f9cca182bfb)
2020-08-31T08:46:03.9944059Z [ 13%] Downloading file cmake/modules/COPYING-CMAKE-SCRIPTS from qt-gstreamer git repository (ref a0e95b202a72b6d9e48bd1949ab6811c0f3c91c3)
2020-08-31T08:46:04.2579149Z [ 16%] Downloading file cmake/modules/FindGLIB2.cmake from qt-gstreamer git repository (ref a0e95b202a72b6d9e48bd1949ab6811c0f3c91c3)
2020-08-31T08:46:04.4249374Z [ 18%] Downloading file cmake/modules/FindGStreamerPluginsBase.cmake from qt-gstreamer git repository (ref a0e95b202a72b6d9e48bd1949ab6811c0f3c91c3)
2020-08-31T08:46:04.6023988Z [ 21%] Downloading file cmake/modules/MacroFindGStreamerLibrary.cmake from qt-gstreamer git repository (ref a0e95b202a72b6d9e48bd1949ab6811c0f3c91c3)
2020-08-31T08:46:04.7395110Z [ 24%] Downloading file cmake/modules/FindGStreamer.cmake from qt-gstreamer git repository (robotology-dependency fork) (ref 383c41d310b14c328ab6d5d02f478f9cca182bfb)
2020-08-31T08:46:04.9298408Z [ 27%] Files from qt-gstreamer git repository (ref a0e95b202a72b6d9e48bd1949ab6811c0f3c91c3) downloaded
2020-08-31T08:46:04.9344531Z [ 27%] Built target 3rdparty-qt-gstreamer
2020-08-31T08:46:04.9384807Z Scanning dependencies of target 3rdparty-qgv
2020-08-31T08:46:04.9602218Z [ 29%] Files from 3rdparty-qgv installed in CMake build directory
2020-08-31T08:46:04.9642334Z [ 29%] Built target 3rdparty-qgv
2020-08-31T08:46:04.9681340Z Scanning dependencies of target 3rdparty-cmake-wiki
[... 48 lines cut ...]
2020-08-31T08:46:06.0653368Z       for file: [/home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/downloads/cmake/FindEigen3.cmake]
2020-08-31T08:46:06.0653877Z         expected hash: [5ad2b8e1ddbd9f0468b21e9d5343d05eda6b9dd1]
2020-08-31T08:46:06.0654311Z           actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
2020-08-31T08:46:06.0655726Z                status: [22;"HTTP response code said error"]
2020-08-31T08:46:06.0656400Z CMakeFiles/Makefile2:1369: recipe for target '3rdparty/CMakeFiles/3rdparty-eigen.dir/all' failed
2020-08-31T08:46:06.0656987Z 
2020-08-31T08:46:06.0657274Z 
2020-08-31T08:46:06.0657716Z make[2]: *** [3rdparty/eigen3/cmake/FindEigen3.cmake] Error 1
2020-08-31T08:46:06.0658476Z make[1]: *** [3rdparty/CMakeFiles/3rdparty-eigen.dir/all] Error 2
2020-08-31T08:46:06.0659302Z Makefile:145: recipe for target 'all' failed
2020-08-31T08:46:06.0668412Z make: *** [all] Error 2
2020-08-31T08:46:06.0682286Z ##[error]Process completed with exit code 2.
2020-08-31T08:46:06.0758435Z Post job cleanup.
2020-08-31T08:46:06.1779422Z [command]/usr/bin/git version
2020-08-31T08:46:06.1839184Z git version 2.28.0
2020-08-31T08:46:06.1873522Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-08-31T08:46:06.1913844Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-08-31T08:46:06.2221555Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-08-31T08:46:06.2256377Z http.https://github.com/.extraheader
2020-08-31T08:46:06.2265085Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-08-31T08:46:06.2306348Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-08-31T08:46:06.2639122Z Cleaning up orphan processes
```

@traversaro @S-Dafarra @prashanthr05

GiulioRomualdi, 2020-08-31:
Related news https://bitbucket.org/blog/sunsetting-mercurial-support-in-bitbucket

GiulioRomualdi, 2020-08-31:
@traversaro fixed the issue https://github.com/robotology/ycm/commit/a50ca05f3ea7c6afd97a066061fe4fb1d6507ceb

S-Dafarra, 2020-08-31:
Wasn't this fixed with https://github.com/dic-iit/bipedal-locomotion-framework/pull/86?

GiulioRomualdi, 2020-08-31:
> Wasn't this fixed with #86?

Ops I didn't see it. Sorry. I've to rebase my PR on master :smile:

closed 2020-08-31 by GiulioRomualdi