gbionics/bipedal-locomotion-framework #94: Discuss the usage of proper types to handle heterogeneous elements through the RobotInterface

prashanthr05, 2020-09-04:
In general, we adhere to the usage of Eigen types in the public API provided by the libraries in this repository.
However, when it comes to dealing with heterogeneous types of data, which may further be of fixed size or variable size, single-dimensional or multi-dimensional, be classes or structs, it becomes unclear how these types must be handled.

We saw this ambiguity arise with the introduction of the `ISensorBridge` interface, where we had to deal with,
- fixed-size quantities  (eg. a three-axis gyroscope measurement is of known size 3)
- pre-determined variable size quantities (eg. the number of joints to be read from the robot can be variable but is fixed during initialization)
- variable size quantities (variable set of points)
- matrices, structs, etc (images, point clouds)

Bipedal Locomotion Framework also provides the `GenericContainer` class which was meant to have a unique way to deal with multiple vector implementations that could have also been resized. This was coming in handy when dealing with parameters. Depending on the implementation and on the needs, we may have needed a `Yarp::Sig::Vector`, or `iDynTree::VectorDynSize` for example. This was before moving to Eigen in the main API. So, it would be suggestive to use `GenericContainer::Vector` in case one needs to deal with different vector implementation at the same time, especially in case you need to resize them (something that is not possible with an `Eigen::map`). 

However, `GenericContainer::Vector` cannot be used for datatypes, for instance, images, that are usually stored as matrices. 

Additionally, we must choose the types efficiently so that we do not introduce any additional overheads while converting from one type to another type, and passing it on to the user (who might/might not have to again convert to the desired type). This issue mainly comes into the picture in a SensorBridge scenario, where proper/inexpensive conversions to `cv::Mat` or `pcl::PointCloud` must be done. 

Given the above pointers, in particular, it is not clear:
- when to use Eigen vectors vs GenericContainer vs templates
- how to deal with matrices
- how to deal with particular input classes, like point clouds

cc @S-Dafarra @GiulioRomualdi @traversaro @diegoferigo

S-Dafarra, 2020-09-04:
Probably we can simply turn this into an epic and open an issue for each subpoint.

S-Dafarra, 2020-09-04:
Eventually, we can use this specific issue to discuss mainly about the third point. @prashanthr05 did you have something in mind to deal with 
> how to deal with particular input classes, like point clouds

?

prashanthr05, 2020-09-07:
Sorry that I come back to this a bit late.

> @prashanthr05 did you have something in mind to deal with
> 
> > how to deal with particular input classes, like point clouds
> 
> ?

At this point, no. We must take a use-case scenario like a YARP implementation (or similar, what is practiced in the sensing and perception group) and try to understand how it sits within our SensorBridge interface design.

prashanthr05, 2020-12-07:
In order to properly handle images and make use of the OpenCV library, I refactored the Camera interface from the `ISensorBridge` into a new class [`ICameraBridge`](https://github.com/prashanthr05/bipedal-locomotion-framework/blob/feature/CameraBridge/src/RobotInterface/include/BipedalLocomotion/RobotInterface/ICameraBridge.h) which can be compiled only if OpenCV is found as a dependency of BLF.  This allows us to bypass https://github.com/dic-iit/bipedal-locomotion-framework/issues/110.

The Yarp Implementation of the Camera bridge currently implements the bridge methods for `IRGBDSensor` and `IFrameGrabber` interfaces allowing us to read RGB and depth images. It is done so in the [`YarpCameraBridge`](https://github.com/prashanthr05/bipedal-locomotion-framework/blob/feature/CameraBridge/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpCameraBridge.h) class.

It must be noted that in order to keep `YarpCameraBridge` storage light, it does not inherit from `Advanceable` and the user can initialize the camera bridge,  set polydrivers list and is allowed to call the getters without running advance every time. (Mimics the Advanceable structure without the advance() method). This approach allows us to maintain only minimal buffers for transferring the image from the interface to the user. This is in line with the issue https://github.com/dic-iit/bipedal-locomotion-framework/issues/109.

I have tested the camera bridge with a Realsense D435i camera used with YARP realsense2withIMU driver.

[image]

The color of the RGB image seems to be incorrect. I remember the the Realsense device itself does some warm-up phase, until which the color remains as seen in the image, however it auto-corrects itself. This does not seem to be happening in this case, which needs to be verified.
However, the interface streaming seems to be working as expected.

S-Dafarra, 2020-12-07:
All the points above seem pretty reasonable @prashanthr05 :rocket:

traversaro, 2020-12-07:
🚀🚀🚀

prashanthr05, 2020-12-15:
> The color of the RGB image seems to be incorrect. 

Looks like realsense gives BGR image. I noticed that in https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L133-L136

GiulioRomualdi, 2024-04-05:
@prashanthr05 implemented camerabridge back in time ❤️ This is currently used by many project in the lab! I'm going to close this Epic.

Thanks again all!

closed 2024-04-05 by GiulioRomualdi