gbionics/bipedal-locomotion-framework #96: Decide on how to handle matrices in the public interfaces

S-Dafarra, 2020-09-04:
This issue wants to address one of the points raised in https://github.com/dic-iit/bipedal-locomotion-framework/issues/94, in particular, how to deal with the case where we have to input a matrix to an interface.

The problems here are similar to the ones presented in https://github.com/dic-iit/bipedal-locomotion-framework/issues/95. In addition, compared to https://github.com/dic-iit/bipedal-locomotion-framework/issues/95, there is no equivalent of ``GenericContainer::Vector``.

S-Dafarra, 2020-09-04:
We already discussed this back in the past: https://github.com/robotology/idyntree/issues/430. In particular, I would drop here the same idea I had back then (which I forgot and got again today) https://github.com/robotology/idyntree/issues/430#issuecomment-380123607

In particular, a possibility could be to have a ``GenericContainer::Matrix`` class which contains a pointer to the vectorized version of the matrix, plus a couple of variables for the size and to discern if the matrix is row or column major. 

Do you think it may be worth going down this road? @prashanthr05 @GiulioRomualdi @traversaro

GiulioRomualdi, 2024-04-05:
We have been using Eigen::Matrix for a while. I will close this issue. In case of needed we can reopen it or create a new one

closed 2024-04-05 by GiulioRomualdi