gbionics/bipedal-locomotion-framework #99: GenericContainer compilation error in macOS 10.14.2, Apple LLVM version 10.0.0 (clang-1000.11.45.5)

prashanthr05, 2020-09-08:
I get the following error trying to compile BLF in a Mac OS X system

``` icub$ make install
Scanning dependencies of target CatchTestMain
[  2%] Building CXX object CMakeFiles/CatchTestMain.dir/Testing/Catch2Main.cpp.o
[  4%] Linking CXX shared library lib/libCatchTestMain.dylib
[  4%] Built target CatchTestMain
Scanning dependencies of target GenericContainerPlusYarpUnitTests
[  6%] Building CXX object src/GenericContainer/tests/CMakeFiles/GenericContainerPlusYarpUnitTests.dir/GenericContainerTestPlusYarp.cpp.o
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/tests/GenericContainerTestPlusYarp.cpp:35:34: error: no viable constructor or deduction guide for deduction of template arguments of 'Vector'
        GenericContainer::Vector container(iDynTree::make_span(vector));
                                 ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:130:5: note: candidate template ignored: substitution failure [with T = double]
    Vector(iDynTree::Span<T> span)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:33:7: note: candidate template ignored: could not match 'Vector' against 'Span'
class Vector;
      ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:146:5: note: candidate template ignored: could not match 'Vector' against 'Span'
    Vector(const Vector<T>& other) = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:154:5: note: candidate template ignored: could not match 'Vector' against 'Span'
    Vector(Vector<T>&& other)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:111:5: note: candidate function template not viable: requires 0 arguments, but 1 was provided
    Vector() = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:118:5: note: candidate function template not viable: requires 2 arguments, but 1 was provided
    Vector(iDynTree::Span<T> span, resize_function_type resizeLambda)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/tests/GenericContainerTestPlusYarp.cpp:63:34: error: no viable constructor or deduction guide for deduction of template arguments of 'Vector'
        GenericContainer::Vector container(iDynTree::make_span(vector));
                                 ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:130:5: note: candidate template ignored: substitution failure [with T = double]
    Vector(iDynTree::Span<T> span)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:33:7: note: candidate template ignored: could not match 'Vector' against 'Span'
class Vector;
      ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:146:5: note: candidate template ignored: could not match 'Vector' against 'Span'
    Vector(const Vector<T>& other) = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:154:5: note: candidate template ignored: could not match 'Vector' against 'Span'
    Vector(Vector<T>&& other)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:111:5: note: candidate function template not viable: requires 0 arguments, but 1 was provided
    Vector() = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:118:5: note: candidate function template not viable: requires 2 arguments, but 1 was provided
    Vector(iDynTree::Span<T> span, resize_function_type resizeLambda)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:645:20: error: no viable constructor or deduction guide for deduction of template arguments of 'Vector'
            return Vector(span, DefaultVectorResizer(input));
                   ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/tests/GenericContainerTestPlusYarp.cpp:39:74: note: in instantiation of function template specialization 'BipedalLocomotion::GenericContainer::make_vector<yarp::sig::VectorOf<double> >' requested here
        GenericContainer::Vector containerToBeCopied = GenericContainer::make_vector(copiedIn);
                                                                         ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:118:5: note: candidate template ignored: substitution failure [with T = double]
    Vector(iDynTree::Span<T> span, resize_function_type resizeLambda)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:33:7: note: candidate function template not viable: requires 1 argument, but 2 were provided
class Vector;
      ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:111:5: note: candidate function template not viable: requires 0 arguments, but 2 were provided
    Vector() = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:130:5: note: candidate function template not viable: requires single argument 'span', but 2 arguments were provided
    Vector(iDynTree::Span<T> span)
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:146:5: note: candidate function template not viable: requires single argument 'other', but 2 arguments were provided
    Vector(const Vector<T>& other) = delete;
    ^
/Users/icub/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:154:5: note: candidate function template not viable: requires single argument 'other', but 2 arguments were provided
    Vector(Vector<T>&& other)
    ^
3 errors generated.
make[2]: *** [src/GenericContainer/tests/CMakeFiles/GenericContainerPlusYarpUnitTests.dir/GenericContainerTestPlusYarp.cpp.o] Error 1
make[1]: *** [src/GenericContainer/tests/CMakeFiles/GenericContainerPlusYarpUnitTests.dir/all] Error 2
make: *** [all] Error 2
```

The version of gcc 
```
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin18.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin```

CMake version
```
cmake --version
cmake version 3.18.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

Any idea? @S-Dafarra @traversaro

traversaro, 2020-09-08:
As in GitHub actions we are testing only macOS 10.15 (see https://docs.github.com/en/actions/reference/virtual-environments-for-github-hosted-runners#supported-runners-and-hardware-resources), should we just update the macOS machine and support only macOS >= 10.15 ?

S-Dafarra, 2020-09-08:
A quick test could be to change https://github.com/dic-iit/bipedal-locomotion-framework/blob/26f45d714097ad3fe299025122cae1864cbef497/src/GenericContainer/tests/GenericContainerTestPlusYarp.cpp#L35 to 
```
GenericContainer::Vector<double> container(iDynTree::make_span(vector)); 
```

prashanthr05, 2020-09-08:
> As in GitHub actions we are testing only macOS 10.15 (see https://docs.github.com/en/actions/reference/virtual-environments-for-github-hosted-runners#supported-runners-and-hardware-resources), should we just update the macOS machine and support only macOS >= 10.15 ?

I am running an update for that. 

However, looks like C++17 is not being enforced by the gcc version that was existing. I will see if there is a bump in the gcc version after the upgrade and see if it allows the use of C++17.

prashanthr05, 2020-09-09:
I updated the macOS to Catalina 10.15. Still the gcc version was only 8. I upgraded the gcc. The version was bumped to 10.0.
So apparently, this version of gcc fully supports C++17 standards. And I was able to compile without any errors.

I will close this issue.

closed 2020-09-09 by prashanthr05

traversaro, 2020-09-09:
Interesting: I would expect that on macOS with XCode by default clang is used, not gcc.

prashanthr05, 2020-09-09:
> Interesting: I would expect that on macOS with XCode by default clang is used, not gcc.

Uhm, not sure. But I was compiling it from the terminal.