gbionics/bipedal-locomotion-framework #109: Understand if current design of YarpSensorBridge class is feasible enough for large-scale sensor applications

prashanthr05, 2020-09-18:
Copying from https://github.com/dic-iit/bipedal-locomotion-framework/pull/106

> I Inherited the YarpSensorBridge from the Advanceable class as well. However, I have a feeling that to poll the measurements at each advance step and have a huge set of measurements stored internally would be expensive than polling for the specific measurement whenever the user calls the getXXX() method on a very specific sensor of a specific measurement type. Also since within the advance() step all the sensor reads are done serially (no sense of parallelism in internal reads as well), this might cause a huge overhead in the pipeline. I hope this doesn't turn out to be a huge design flaw in the future. If it does, we might have to reuse the read individual sensor methods in either a parallel way or to call the individual sensor reads while the user calls the getters.
(Imagine a single instance of SensorBridge configured with all sensor interfaces including cameras and multiple interfaces!)

We decided for the current requirements, we will stick with this workflow. But I will keep this issue here so that we can discuss potential implementation design changes in case the current design of YarpSensorBridge does not scale up efficiently for our future experiments.

prashanthr05, 2020-12-03:
cc @Yeshasvitvs

Yeshasvitvs, 2020-12-03:
As discussed this morning with @prashanthr05 one of the options that would be nice to have from a client perspective is the ability to make a boolean option for [advance function (https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L127), which when set to false, individual sensor streams such as [camera data](https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L229) or [imu data](https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L159) can be accessed through specific sensor api (which at the moment are private), instead of accessing through only [readAllSensors](https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L137)

@prashanthr05 please feel free to correct any details that I got wrong. Thanks.