gbionics/bipedal-locomotion-framework #111: Add some tests to the Gihub Actions for the Yarp implementations of the RobotInterface

S-Dafarra, 2020-09-18:
After a meeting with @prashanthr05 @GiulioRomualdi @kouroshD and @isorrentino we converged on the following.

In order to test these implementations, we decided to use Gazebo with the ``gazebo-yarp-plugins``. 

These interfaces do not need precise synchronization since, for the time being, we just need to set simple commands and read the output of the sensors. Hence, we thought about keeping Gazebo as a separate process (without visualization). The clock can be controlled as in the [Simulator synchronizer](https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/SimulatorSynchronizer.cpp) of WB-Toolbox.

We identified the following action points:
- Edit the Github actions in order to install Gazebo at least on Ubuntu and prepare a world with the robot in it. 
- Prepare a CMake script to automatize the generation of tests involving Gazebo
- Define a utility class to communicate with Gazebo in a test.

These action points will be on separate issues.

S-Dafarra, 2020-09-18:
This may also be the first step towards https://github.com/dic-iit/component_wholebody-teleoperation/issues/240

GiulioRomualdi, 2020-09-18:
cc @traversaro

traversaro, 2024-04-05:
cc @xela-95