gbionics/bipedal-locomotion-framework #129: Implement the machinery for the generation fo Python bindings

GiulioRomualdi, 2020-10-19:


GiulioRomualdi, 2020-10-19:
Today I tried to play with SWIG for the generation of the python bindings. 
I tried to adapt the code of 
-  [`superquadric-lib`](https://github.com/eleramp/superquadric-lib/tree/6ce39ebe20621a444c0170a3c12075efac052bd2/bindings)
- [`iDynTree`](https://github.com/robotology/idyntree/tree/0e1949b6a9be0c61adf40a0a55dda9df719af7b6/bindings)

I wrote the `BipedalLocomotionFramework.i` file
https://github.com/dic-iit/bipedal-locomotion-framework/blob/b76c17ff50090012840e6231276ebdc60f580b26/bindings/BipedalLocomotionFramework.i#L1

I've also introduced the the `eigen.i` and `numpy.i` files [here](https://github.com/dic-iit/bipedal-locomotion-framework/blob/add_bindings/bindings/python)

When I tried to generate the bindings I didn't get any error, however the `BipedalLocomotion.py` does not contains any class of the framework.

<details><summary>Click here for check the content of <b>BipedalLocomotion.py</b></summary>
<p>
```py
# This file was automatically generated by SWIG (http://www.swig.org).
# Version 3.0.12
#
# Do not make changes to this file unless you know what you are doing--modify
# the SWIG interface file instead.

from sys import version_info as _swig_python_version_info
if _swig_python_version_info >= (2, 7, 0):
    def swig_import_helper():
        import importlib
        pkg = __name__.rpartition('.')[0]
        mname = '.'.join((pkg, '_BipedalLocomotion')).lstrip('.')
        try:
            return importlib.import_module(mname)
        except ImportError:
            return importlib.import_module('_BipedalLocomotion')
    _BipedalLocomotion = swig_import_helper()
    del swig_import_helper
elif _swig_python_version_info >= (2, 6, 0):
    def swig_import_helper():
        from os.path import dirname
        import imp
        fp = None
        try:
            fp, pathname, description = imp.find_module('_BipedalLocomotion', [dirname(__file__)])
        except ImportError:
            import _BipedalLocomotion
            return _BipedalLocomotion
        try:
            _mod = imp.load_module('_BipedalLocomotion', fp, pathname, description)
        finally:
            if fp is not None:
                fp.close()
        return _mod
    _BipedalLocomotion = swig_import_helper()
    del swig_import_helper
else:
    import _BipedalLocomotion
del _swig_python_version_info

try:
    _swig_property = property
except NameError:
    pass  # Python < 2.2 doesn't have 'property'.

try:
    import builtins as __builtin__
except ImportError:
    import __builtin__

def _swig_setattr_nondynamic(self, class_type, name, value, static=1):
    if (name == "thisown"):
        return self.this.own(value)
    if (name == "this"):
        if type(value).__name__ == 'SwigPyObject':
[... 309 lines cut ...]

    def assign(self, n, x):
        return _BipedalLocomotion.IntVector_assign(self, n, x)

    def resize(self, *args):
        return _BipedalLocomotion.IntVector_resize(self, *args)

    def insert(self, *args):
        return _BipedalLocomotion.IntVector_insert(self, *args)

    def reserve(self, n):
        return _BipedalLocomotion.IntVector_reserve(self, n)

    def capacity(self):
        return _BipedalLocomotion.IntVector_capacity(self)
    __swig_destroy__ = _BipedalLocomotion.delete_IntVector
    __del__ = lambda self: None
IntVector_swigregister = _BipedalLocomotion.IntVector_swigregister
IntVector_swigregister(IntVector)

# This file is compatible with both classic and new-style classes.
```
</p>

cc @traversaro and @diegoferigo

traversaro, 2020-10-19:
I guess that the software headers need to be included two times, once with `#include` and once with `%include`, see https://github.com/robotology/idyntree/blob/0e1949b6a9be0c61adf40a0a55dda9df719af7b6/bindings/iDynTree.i#L38 vs https://github.com/robotology/idyntree/blob/0e1949b6a9be0c61adf40a0a55dda9df719af7b6/bindings/iDynTree.i#L162 . The `#include` specify the header file to include in the generated source file, while the `%include` the headers that SWIG should process.

traversaro, 2020-10-19:
You may be interested in https://github.com/rdeits/swig-eigen-numpy as well.

GiulioRomualdi, 2020-10-19:
Thank you @traversaro. 

Now swig fails with this nice error regarding templates
```
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:20: Error: Syntax error in input(1).
bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/build.make:80: recipe for target 'bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx' failed
make[2]: *** [bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx] Error 1
CMakeFiles/Makefile2:984: recipe for target 'bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all' failed
make[1]: *** [bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

https://github.com/dic-iit/bipedal-locomotion-framework/blob/b76c17ff50090012840e6231276ebdc60f580b26/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h#L20

traversaro, 2020-10-19:
I am afraid that the SWIG support for C++17 syntax is still incomplete: https://github.com/swig/swig/issues/1863 .

diegoferigo, 2020-10-19:
I fear that heavily templated libraries, especially with C++17, are quite challenging to bind with SWIG :/

diegoferigo, 2020-10-19:
I think that using pybind11 as done in https://github.com/robotology/idyntree/pull/741 would provide a much greater freedom if you plan to target only Python.

traversaro, 2020-10-19:
If that code is not actually useful to define the interface that you want to wrap with SWIG (perhaps because it is used just internally) you can try to disable it with some #ifdef SWIG or similar.

diegoferigo, 2020-10-19:
> If that code is not actually useful to define the interface that you want to wrap with SWIG (perhaps because it is used just internally) you can try to disable it with some #ifdef SWIG or similar.

Or, if they do not contain APIs, you can just `#include` them without using `%include`. This means that the header is imported while compiling the autogenerated class but SWIG does not parse it to expose its functions to the target language.

GiulioRomualdi, 2020-10-19:
> I think that using pybind11 as done in robotology/idyntree#741 would provide a much greater freedom if you plan to target only Python.

The idea was to start with python and then extend the bings to Matlab. 

> If that code is not actually useful to define the interface that you want to wrap with SWIG (perhaps because it is used just internally) you can try to disable it with some #ifdef SWIG or similar.

Unfortunately, it's not so simple, for instance, I need the definition of `BipedalLocomotion::GenericContainer::Vector<T>::Ref`
https://github.com/dic-iit/bipedal-locomotion-framework/blob/b76c17ff50090012840e6231276ebdc60f580b26/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h#L906-L907
because it is required in 
https://github.com/dic-iit/bipedal-locomotion-framework/blob/b76c17ff50090012840e6231276ebdc60f580b26/src/ParametersHandler/include/BipedalLocomotion/ParametersHandler/StdImplementation.h#L119

however
```
[ 94%] Swig source
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:228: Warning 362: operator= ignored
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:245: Warning 362: operator= ignored
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:262: Warning 362: operator= ignored
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:316: Warning 389: operator[] ignored (consider using %extend)
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:326: Warning 389: operator[] ignored (consider using %extend)
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:912: Error: Undefined scope 'BipedalLocomotion::GenericContainer::Vector<(T)>'
/home/gromualdi/robot-code/bipedal-locomotion-framework_swig/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h:973: Error: Syntax error in input(3).
bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/build.make:80: recipe for target 'bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx' failed
make[2]: *** [bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx] Error 1
CMakeFiles/Makefile2:984: recipe for target 'bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all' failed
make[1]: *** [bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

Adding 
```
%include "BipedalLocomotion/GenericContainer/Vector.h"

%template(StringVector) BipedalLocomotion::GenericContainer::Vector<std::string>;
%template(IntVector) BipedalLocomotion::GenericContainer::Vector<int>;
%template(DoubleVector) BipedalLocomotion::GenericContainer::Vector<double>;

%template(StringVectorRef) BipedalLocomotion::GenericContainer::Vector<std::string>::Ref;
%template(IntVectorRef) BipedalLocomotion::GenericContainer::Vector<int>::Ref;
%template(DoubleVectorRef) BipedalLocomotion::GenericContainer::Vector<double>::Ref;
```
did not solve the problem.

traversaro, 2020-10-19:
For dealing with Eigen, did you checked https://github.com/rdeits/swig-eigen-numpy ? It could contain some interesting pointers. 

Anyhow, given the problems that you are experiencing with C++17 and SWIG, and given the interest that I guess we have for also having MATLAB bindings that prevents us from using alternative tools, I wonder if it could make to take a step back and wonder if there could be a better strategy to tackle this problem. 

A lot of code in BLF is already strongly block-oriented, in the sense that you have the `initialize` to pass fixed parameters, you have some methods (not structured) to set the inputs of the dynamical system, you have the `advance` method to advance the time in the dynamical system, and you have some (not structured) methods to retrieve the outputs. Perhaps a possible strategy is to introduce some structures ways of setting the inputs and the outputs (i.e. a "Block" interface) and then in Python and MATLAB provide only a generic APIs to deal with blocks. In this way the API surface that should be compatible with SWIG (or that needs to be manually exposed to Python and MATLAB) would be minimized.

GiulioRomualdi, 2020-10-20:
Here some updates
1. I had to change this line 
    https://github.com/dic-iit/bipedal-locomotion-framework/blob/216d2b0503302d8628129490406cb9652a77b0a2/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h#L70
    into 
    ```cpp
    using reference = value_type&
    ``` 
    without changing this line the following error is thrown by swig

    <details><summary>Click here to check the content of <b>the error</b></summary>
    <p>

    ```
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx: In instantiation of ‘struct SwigValueWrapper<std::__cxx11::basic_string<char>&>::SwigMovePointer’:
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:28:5:   required from ‘class SwigValueWrapper<std::__cxx11::basic_string<char>&>’
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:4885:64:   required from here
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:24:8: error: forming pointer to reference type ‘std::__cxx11::basic_string<char>&’
         T *ptr;
            ^~~
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:25:5: error: forming pointer to reference type ‘std::__cxx11::basic_string<char>&’
         SwigMovePointer(T *p) : ptr(p) { }
         ^~~~~~~~~~~~~~~
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx: In instantiation of ‘class SwigValueWrapper<std::__cxx11::basic_string<char>&>’:
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:4885:64:   required from here
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:35:6: error: forming pointer to reference type ‘std::__cxx11::basic_string<char>&’
       T *operator&() { return pointer.ptr; }
          ^~~~~~~~
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx: In function ‘PyObject* _wrap_StringVector_at__SWIG_1(PyObject*, Py_ssize_t, PyObject**)’:
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:4911:208: error: new cannot be applied to a reference type
       resultobj = SWIG_NewPointerObj((new BipedalLocomotion::GenericContainer::Vector< std::string >::reference(static_cast< const BipedalLocomotion::GenericContainer::Vector< std::string >::reference& >(result))), SWIGTYPE_p_iDynTree__SpanT_std__string_t__reference, SWIG_POINTER_OWN |  0 );
                                                                                                                                                                                                                    ^
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:1093:89: note: in definition of macro ‘SWIG_NewPointerObj’
     #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                             ^~~
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx: In function ‘PyObject* _wrap_StringVector___call____SWIG_1(PyObject*, Py_ssize_t, PyObject**)’:
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:5037:208: error: new cannot be applied to a reference type
       resultobj = SWIG_NewPointerObj((new BipedalLocomotion::GenericContainer::Vector< std::string >::reference(static_cast< const BipedalLocomotion::GenericContainer::Vector< std::string >::reference& >(result))), SWIGTYPE_p_iDynTree__SpanT_std__string_t__reference, SWIG_POINTER_OWN |  0 );
                                                                                                                                                                                                                    ^
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:1093:89: note: in definition of macro ‘SWIG_NewPointerObj’
     #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                             ^~~
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx: In function ‘PyObject* _wrap_IntVector_at__SWIG_1(PyObject*, Py_ssize_t, PyObject**)’:
    /home/gromualdi/robot-code/bipedal-locomotion-framework_swig/build/bindings/python/BipedalLocomotionFrameworkPYTHON_wrap.cxx:6169:65: error: ‘result’ declared as reference but not initialized
       BipedalLocomotion::GenericContainer::Vector< int >::reference result;
                                                                     ^~~~~~
[... cut ...]
    CMakeFiles/Makefile2:984: recipe for target 'bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all' failed
    make[1]: *** [bindings/python/CMakeFiles/_BipedalLocomotion_python.dir/all] Error 2
    Makefile:148: recipe for target 'all' failed
    make: *** [all] Error 2
    ```
   </p>
   </details>
    Indeed the `Ref` default constructor is deleted in 

    https://github.com/dic-iit/bipedal-locomotion-framework/blob/216d2b0503302d8628129490406cb9652a77b0a2/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h#L914

@traversaro, @diegoferigo do you know why this is happening?

traversaro, 2020-10-20:
Note that the SWIG 4 documentation is available at http://www.swig.org/Doc4.0/SWIGPlus.html#SWIGPlus_nested_classes .

traversaro, 2020-10-20:
I am afraid that SWIG needs to create a temporary object of Type ref for its code generation to work. A possible way to tackle this is to directly go on and provide typemaps (http://www.swig.org/Doc4.0/SWIGDocumentation.html#Arguments_nn2) to convert directly Ref elements to the equivalent Python types, or ignore the methods that take in input or output Ref types (if this is possible at all).

GiulioRomualdi, 2020-10-21:
In https://github.com/GiulioRomualdi/swig-eigen-example I wrote a simple python wrapper that uses swig+eigen. 

Related issue: https://github.com/GiulioRomualdi/swig-eigen-example/issues/1

traversaro, 2020-10-21:
Another codebase using Eigen+SWIG was https://github.com/robot-locomotion/dwl/blob/b45e6babfcace928af2601f3d453e3f27e2e76c8/swig/eigen.i#L626 , but I guess also there they use typemaps.

GiulioRomualdi, 2020-11-23:
#134 closes the issue

closed 2020-11-23 by GiulioRomualdi