gbionics/bipedal-locomotion-framework #133: Error compiling the framework

diegoferigo, 2020-11-03:
I installed the following dependencies: , and without passing any custom CMake option I get the following compilation error:

```
bipedal-locomotion-framework/build on  master [!] via 🐍 v3.8.2 (venv) 
✦2 ❯ ninja
[45/46] Linking CXX shared library lib/libBipedalLocomotionFrameworkPlanners.so.0.0.1
FAILED: lib/libBipedalLocomotionFrameworkPlanners.so.0.0.1 
: && /usr/lib/ccache/clang++-10 -fPIC -fdiagnostics-color=always -O3 -DNDEBUG   -shared -Wl,-soname,libBipedalLocomotionFrameworkPlanners.so.0.0.1 -o lib/libBipedalLocomotionFrameworkPlanners.so.0.0.1 src/Planners/CMakeFiles/Planners.dir/src/ConvexHullHelper.cpp.o src/Planners/CMakeFiles/Planners.dir/src/DCMPlanner.cpp.o src/Planners/CMakeFiles/Planners.dir/src/TimeVaryingDCMPlanner.cpp.o src/Planners/CMakeFiles/Planners.dir/src/QuinticSpline.cpp.o src/Planners/CMakeFiles/Planners.dir/src/SwingFootPlanner.cpp.o  -Wl,-rpath,/home/dferigo/git/bipedal-locomotion-framework/build/lib:/rl/local/lib:/iit/sources/robotology-superbuild/build/install/lib:  lib/libBipedalLocomotionFrameworkSystem.so.0.0.1  lib/libBipedalLocomotionFrameworkContact.so.0.0.1  /rl/local/lib/libqhullcpp.a  /rl/local/lib/libqhullstatic_r.a  /rl/local/lib/libcasadi.so.3.6  lib/libBipedalLocomotionFrameworkContactModels.so.0.0.1  lib/libBipedalLocomotionFrameworkParametersHandler.so.0.0.1  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-high-level.so  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-sensors.so  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-modelio-xml.so  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-model.so  /iit/sources/robotology-superbuild/build/install/lib/libidyntree-core.so  -lm  -ldl && :
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(global_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(stat_r.c.o): relocation R_X86_64_32S against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(geom2_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(poly2_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(merge_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(libqhull_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(geom_r.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(poly_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(qset_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(mem_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(random_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(usermem_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(io_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /rl/local/lib/libqhullstatic_r.a(user_r.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

I tried to add the following to the main CMakeLists.txt without any change:

```cmake
# Always use -fPIC
set(CMAKE_POSITION_INDEPENDENT_CODE ON)
```

S-Dafarra, 2020-11-03:
Thanks @diegoferigo! Feel free to open a PR

diegoferigo, 2020-11-03:
The workaround I tried did **not** solve the problem. I also tried using `Unix Makefiles` and the problem persists, it seems unrelated to the selected generator.

S-Dafarra, 2020-11-03:
Ah sorry, I misunderstood. I wonder if it is a problem of ``qhull`` :thinking:

S-Dafarra, 2020-11-03:
Can you try to install ``qhull`` from source?

GiulioRomualdi, 2020-11-03:
Hi @diegoferigo. Which version of qhull did you install?

S-Dafarra, 2020-11-03:
> Ah sorry, I misunderstood. I wonder if it is a problem of `qhull`

Indeed https://github.com/qhull/qhull/issues/62

S-Dafarra, 2020-11-03:
> > Ah sorry, I misunderstood. I wonder if it is a problem of `qhull`
> 
> Indeed [qhull/qhull#62](https://github.com/qhull/qhull/issues/62)

Indeed indeed https://github.com/dic-iit/element_dcm-walking/issues/176#issuecomment-645440318 :grin:

traversaro, 2020-11-03:
Yes, that is the tipical problem that you have if you try to use a static library compiled without -fPIC when linking a shared library.

diegoferigo, 2020-11-03:
Thanks everyone, compiling with current `master` of qhull passing `CMAKE_POSITION_INDEPENDENT_CODE` and then recompiling the framework works fine. What's strange is that I'm using the most recent `master` of qhull, that is ahead of the `2020.1` tag that supposedly solves the problem.

closed 2020-11-03 by diegoferigo