gbionics/bipedal-locomotion-framework #135: Track the feature upgrades related to FloatingBaseEstimators library and related libraries

prashanthr05, 2020-11-06:
This issue will track the pending PRs that will introduce feature updates to `BipedalLocomotion::FloatingBaseEstimators` library relevant to the online base estimation using state of the art filters and the DILIGENT filter.

Currently, the library contains the implementation of [InvariantEKF](https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/InvariantEKFBaseEstimator.h). 

The pending PRs include,
- [ ] Improvement of InvEKF UnitTest. I improved the unit test [here](https://github.com/prashanthr05/bipedal-locomotion-framework/commit/6c604a7e04e6e58433c6f041a3b044a907298a99) by properly adding noise to the measurements. PR is pending to be open.
- [ ] YARP device wrapper for online estimation A draft PR is open in https://github.com/dic-iit/bipedal-locomotion-framework/pull/130. I need to fix one part related to the `YarpSensorBridge` code to make it functional. And the device would be ready for review.
 - [ ] `IMUBipedMatrixLieGroup` This class was dependent on `manif::SE_2_3` which was merged recently. The current implementation is available at https://github.com/prashanthr05/bipedal-locomotion-framework/tree/feature/IMUBipedMatrixLieGroup.  A proper unit test is missing for this class. Once written, we can open a PR for code review.
- [ ] `DILIGENT` This base estimation technique relies on `IMUBipedMatrixLieGroup` class. Implements the EKF on matrix Lie groups for proprioceptive base estimation. A first implementation is available in https://github.com/prashanthr05/bipedal-locomotion-framework/tree/feature/DLGEKF which is further fixed ifor working condition https://github.com/prashanthr05/bipedal-locomotion-framework/commits/test/DLGEKFonline

Additionally, while porting the code from https://github.com/dic-iit/element_floating-base-estimation/tree/devel/code/cpp/libraries/humanoid-estimators/src/estimation/include/HumEst/iDynTree/Estimation to BLF, I introduced a regression in the way contacts were handled. 
Earlier I was handling an arbitrary number of contacts, while the current implementations in BLF handle only the two support feet contacts.
This must be reverted to the original version of handling an arbitrary number of contacts, which scales up the use of the filters to also SLAM related problems.
- [ ] Switch to using arbitrary contact features in the existing framework in a clean way.

Other changes should include,
- [ ] Replace iDynTree objects using manif in the API
- [ ] sensorsDryReadRun feature in YarpSensorBridge

cc @S-Dafarra @traversaro @GiulioRomualdi

traversaro, 2020-11-06:
fyi @diegoferigo

prashanthr05, 2020-11-06:
A related ongoing epic https://github.com/dic-iit/component_wholebody-teleoperation/issues/297 towards testing the estimators in closed loop.

prashanthr05, 2020-12-10:
> IMUBipedMatrixLieGroup This class was dependent on manif::SE_2_3 which was merged recently. The current implementation is available at [...]

Relating this, and also considering,

> Earlier I was handling an arbitrary number of contacts, while the current implementations in BLF handle only the two support feet contacts.
This must be reverted to the original version of handling an arbitrary number of contacts, which scales up the use of the filters to multiple contacts and also EKF SLAM related problems.
 > Switch to using arbitrary contact features in the existing framework in a clean way. 

I refactored the code in [`IMUBipedMatrixLieGroup`](https://github.com/prashanthr05/bipedal-locomotion-framework/blob/feature/IMUBipedMatrixLieGroup/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/IMUBipedMatrixLieGroup.h) into a more consolidated class considering arbitrary number of contacts in [`FloatingBaseExtendedKinematicsLieGroup`](https://github.com/prashanthr05/bipedal-locomotion-framework/blob/feature/DILIGENT/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseExtendedKinematicsLieGroup.h) also with a test in [`FloatingBaseExtendedKinematicsLieGroupTest`](https://github.com/prashanthr05/bipedal-locomotion-framework/blob/feature/DILIGENT/src/Estimators/tests/FloatingBaseEstimators/FloatingBaseExtendedKinematicsLieGroupTest.cpp).

I know I have been overloading you guys with many recent PR review requests. 
I don't want to bother you more with too many new PRs given that we are already busy with our other recent tasks. 
For the moment, I will keep introducing changes towards DILIGENT considering arbitrary contacts in the same branch [prashanthr05/feature/DILIGENT](https://github.com/prashanthr05/bipedal-locomotion-framework/tree/feature/DILIGENT). 
Once everything is working, I will spit the branch into multiple small PRs (atleast class wise). Let me know if this is ok? Or would you have an alternative suggestion?

S-Dafarra, 2020-12-10:
> Relating this, and also considering,
>
>
> I refactored the code in [...]

It seems reasonable. Do not worry too much about PRs. Do not hesitate to ping from time to time if a PR got stale. It is pretty easy to forget. In any case, for sure having a PR for a class may help. Nonetheless, be careful about interdependencies. Maybe some comments or change requests may break your already existing code and may become difficult to merge the new version.

prashanthr05, 2020-12-10:
> Nonetheless, be careful about interdependencies. Maybe some comments or change requests may break your already existing code and may become difficult to merge the new version.

I have been trying to be careful with that (trying to rebase everytime over the latest updates). Let's see how it unfolds.

traversaro, 2020-12-10:
> It seems reasonable. Do not worry too much about PRs. Do not hesitate to ping from time to time if a PR got stale. It is pretty easy to forget.

Same for me.

prashanthr05, 2020-12-11:
cc @HosameldinMohamed