gbionics/bipedal-locomotion-framework #139: Keep the changelog up to date

GiulioRomualdi, 2020-11-11:
I noticed that I opened several PR without updating the changelog. For instance #61 #90. I was thinking to open a PR where I update the changelog and I also add this simple action to remind the author of the PR to update the changelog https://github.com/marketplace/actions/changelog-checker  

As far as I understood the setting `noChangelogLabel` can be used to avoid running the CI.

@traversaro @S-Dafarra @prashanthr05 do you have any suggestion?

GiulioRomualdi, 2020-11-11:
Also this one could be a valid option: https://github.com/marketplace/actions/missing-changelog-reminder

prashanthr05, 2020-11-11:
Thanks! That'd be helpful. I was also wondering about this.

prashanthr05, 2020-11-11:
YARP uses something like this https://github.com/apps/update-docs. I am not sure if it would be relevant.

GiulioRomualdi, 2020-11-11:
In #140 I used https://github.com/marketplace/actions/changelog-checker but also https://github.com/apps/update-docs is really nice. I'm open to discussion :smiley:

PR #140 (2020-12-07): Update changelog and add changelog-checker CI 
This should fix #139
closed 2020-12-07 by GiulioRomualdi