gbionics/bipedal-locomotion-framework #149: Make the python bindigs working on windows

GiulioRomualdi, 2020-11-19:
[This workflow](https://github.com/dic-iit/bipedal-locomotion-framework/runs/1417004202?check_suite_focus=true) shows the error on windows

GiulioRomualdi, 2020-11-19:
I don't know if  it's related but I drop it here: https://stackoverflow.com/questions/41748464/pytest-cannot-import-module-while-python-can

diegoferigo, 2020-11-19:
The problem is related to cyclic imports. I have no clue why it affects only windows and not the other platforms (but not being a windows user I am not aware of differences in this OS).

A nice resource I found is the one below, that basically suggests to use `import bipedal_locomotion_framework.bindings` instead of `from . import bindings`. I didn't test if with the suggestion, downstream projects can still do `from bipedal_locomotion_framework import bindings as blf`.

https://dev.to/deepjyoti30/why-you-should-keep-away-from-cyclic-imports-in-python-1n75

traversaro, 2020-11-19:
General comment: personal opinion as a predominant Windows users as of now: the only sane way of supporting Python for the bindings of a C++ projects with a lot of dependencies such as BLF is conda/conda-forge, as otherwise you never know even in which of the n-possible ways the user even installed the Python interpreter (if you ever had problems of macOS brew's python vs system's python, Windows is like that, but much worse).

traversaro, 2020-11-19:
Having said that (I am bit worried of @diegoferigo's reaction emoji, and I want to clarify) I think that is perfectly possible to fix the tests on GitHub Actions as they are, and probably @diegoferigo pointers can do that. I just am afraid that then it will not be ergonomic/easy to then expose that to the users.

diegoferigo, 2020-11-19:
In the past months I started looking to conda-forge more in depth and it seems an amazing channel to distribute binaries that is not tied to Python as many may think.

I think we will slowly start adopting it, and there's also the risk that it will become our favourite installation method. I guess that the time has not yet arrived, we still need someone that starts becoming familiar enough and starts advocating for this solution knowing the pros and cons good enough.

> Having said that (I am bit worried of @diegoferigo's reaction emoji, and I want to clarify) I think that is perfectly possible to fix the tests on GitHub Actions as they are, and probably [...]

The emoji is just because Windows always creates troubles :) Too often I think that the effort put in this platform is overly unaligned with our user base. My 2 cents is: provide basic support so that who needs it, being careful, can obtain a working setup, and then, depending on the adoption, starts improving the situation.

traversaro, 2023-07-31:
Python bindings test are working fine on Windows, see for example https://github.com/ami-iit/bipedal-locomotion-framework/actions/runs/5682006541/job/15399402563 . I thnk we can close this issue.

closed 2023-07-31 by GiulioRomualdi