gbionics/bipedal-locomotion-framework #162: Migrate `Contact` library outside `Planners` folder

prashanthr05, 2020-12-11:
In https://github.com/dic-iit/bipedal-locomotion-framework/pull/142, we identified the `Contact` class to have a more generic usage extending beyond just planning. 
So we abstracted the current Contact class into a `ContactBase` class and implemented derived classes `PlannedContact` and `EstimatedContact` classes.

In order to not break the current workflows and python bindings related merges, I created a new library `Contacts` and added only the above mentioned classes. 
However, it would make sense to eventually move other contacts-related components in the `Contact` library that are currently existing in the `Planners` folder as well into the new `Contacts` folder.
Consequently, we might also have to make changes in the python bindings generation? (do we? please correct me if I am wrong.)

cc @GiulioRomualdi @S-Dafarra  @diegoferigo

diegoferigo, 2020-12-11:
No problem for the renaming from my side, I don't have much code to migrate and I think I'm the only one that developed in the past weeks some code in Python.

> Consequently, we might also have to make changes in the python bindings generation? (do we? please correct me if I am wrong.)

Yes there are few required changes, for instance (they might not be all of them):

- [`bindings/python/Contacts.cpp`](https://github.com/dic-iit/bipedal-locomotion-framework/blob/f58eb5863e6a685d4805a8b193c7871d1b3ac1a2/bindings/python/Contacts.cpp)
- [`python/tests/test_contact.py`](https://github.com/dic-iit/bipedal-locomotion-framework/blob/f58eb5863e/bindings/python/tests/test_contact.py)
- [`python/tests/test_time_verying_dcm_planner.py`](https://github.com/dic-iit/bipedal-locomotion-framework/blob/f58eb5863e/bindings/python/tests/test_time_verying_dcm_planner.py)

prashanthr05, 2020-12-11:
Thanks @diegoferigo. I will open a PR after https://github.com/dic-iit/bipedal-locomotion-framework/pull/142/files gets merged, and add you as one of the reviewers for the python part.

prashanthr05, 2021-01-18:
https://github.com/dic-iit/bipedal-locomotion-framework/pull/142 has been merged. I will proceed with the complete migration soon.

PR #204 (2021-02-23): C++ refactoring of Contact
This is the  C++ refactoring of `Contact` class. I'm not really sure about this solution for the following reasons:
1.  Right now we are exposing `Contacts` and `Contact`. The former is in https://github.com/dic-iit/bipedal-locomotion-framework/tree/master/src/Contacts while the latter is in https://github.com/dic-iit/bipedal-locomotion-framework/tree/master/src/Planners. This is confusing :sob:
    [image]

2. Shall we move also the `ContactLists` etc in `Contacts`?
3. I do not know how to handle the transition in the bindings for the following reasons
    1. the ol [... cut]
closed 2021-02-23 by GiulioRomualdi