gbionics/bipedal-locomotion-framework #163: Bindings depend on libraries that may not be compiled

S-Dafarra, 2020-12-14:
See https://github.com/dic-iit/bipedal-locomotion-framework/blob/48a53370ea8ee729fa479f29341cbf3b8be5767f/bindings/python/CMakeLists.txt#L19-L23

but these libraries exist only if the corresponding options are enabled https://github.com/dic-iit/bipedal-locomotion-framework/blob/48a53370ea8ee729fa479f29341cbf3b8be5767f/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L177-L191

Probably some dependencies should be added here https://github.com/dic-iit/bipedal-locomotion-framework/blob/48a53370ea8ee729fa479f29341cbf3b8be5767f/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L219

S-Dafarra, 2020-12-14:
cc @GiulioRomualdi @diegoferigo

diegoferigo, 2020-12-14:
I suspect that compiling the bindings only for a subset of libraries could be very tricky and hard to maintain. I would suggest to enable the bindings only if all the dependent libraries are enabled.

S-Dafarra, 2020-12-14:
> I suspect that compiling the bindings only for a subset of libraries could be very tricky and hard to maintain. I would suggest to enable the bindings only if all the dependent libraries are enabled.

Yes, I was thinking the same, basically adding a few more dependencies on https://github.com/dic-iit/bipedal-locomotion-framework/blob/48a53370ea8ee729fa479f29341cbf3b8be5767f/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L219

GiulioRomualdi, 2021-01-25:
This issue should be fixed by #173

closed 2021-02-24 by GiulioRomualdi