gbionics/bipedal-locomotion-framework #166: Implement a TSID component

GiulioRomualdi, 2020-12-22:
In this issue, I would like to discuss the possibility to design and implement a task base inverse dynamics (TSID) component. 
The first working implementation can be found in [this branch](https://github.com/dic-iit/bipedal-locomotion-framework/tree/feature/torque_control_with_compliant_contacts). When I implemented the status of blf was not well defined, for instance, the `IParameterHandler` didn't exist yet. Furthermore, I used iDynTree matrices/vectors and this goes in contrast to #63. Thanks to https://github.com/robotology/idyntree/pull/736 we can now use `Eigen` objects in kindyn. 

It would be nice to implement the controller in a modular way. What I had in mind is to define the concept of task and then attached several tasks to the controller (a QP). For each task we should define:
1. the meaning (are we controlling the pose of a frame? is it a regularization task?)
2. the priority (in theory we may implement a cascade of QP ~ check [here](https://hal.archives-ouvertes.fr/hal-00751924/document))

It would be really nice to add the possibility to remove/add a new task while the controller is running. In this case, we probably have to recreate a new QP object but this should be transparent to the user. 

Here an example of what I've in mind. 
[image]

I would like to discuss possible desiderata with @traversaro @S-Dafarra and @prashanthr05 before starting the implementation.

S-Dafarra, 2020-12-22:
I like the dynamic loading of tasks, even if I am envisioning this more at a configuration time, rather than being online. I also like the idea of changeable priority, meaning that a task can be either considered as part of the cost or a constraint in a transparent way. Ideally, this detail (cost vs constraint) could be totally transparent to the user.

GiulioRomualdi, 2020-12-22:
> Ideally, this detail (cost vs constraint) could be totally transparent to the user.

Yes I was thinking to handle the priority with an int, the lower the higher the priority

GiulioRomualdi, 2020-12-22:
Another think we should consider is the possibility to add slack variables

traversaro, 2020-12-22:
I do not know if you checked the "official" TSID implementation at https://github.com/stack-of-tasks/tsid, it could give some nice insight of how to architecture that. Another nice and developed architecture in this area is OpenSOT. I guess it is now closed source (unless I am wrong, @EnricoMingo @alaurenzi @liesrock please correct me if I am wrong), but you can find some hints on its architecture in https://ieeexplore.ieee.org/document/7140076 and https://ieeexplore.ieee.org/abstract/document/8246954 .

GiulioRomualdi, 2020-12-22:
As far as I know in TSID only two levels of priority are implemented. For instance a task with priority:
* `0`: is considered as a constraint
* `1`: is embedded in the cost function.

GiulioRomualdi, 2020-12-22:
[**Momentum control with hierarchical inverse dynamics on a torque-controlled humanoid**](https://link.springer.com/article/10.1007/s10514-015-9476-6) can be used as a reference paper to implement a hierarchical TSID controller

GiulioRomualdi, 2021-11-15:
The component has been implemented, closing

closed 2021-11-15 by GiulioRomualdi