gbionics/bipedal-locomotion-framework #171: robotology-superbuild on Unstable branches with ROBOTOLOGY_ENABLE_DYNAMICS ON and ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS OFF compilation broken

traversaro, 2021-01-23:
In the latest nightly build of the robotology-superbuild, a few jobs failed with this error:
~~~
2021-01-23T03:21:56.2441183Z CMake Error at cmake/AddBipedalLocomotionApplication.cmake:30 (add_executable):
2021-01-23T03:21:56.2442530Z   Target "joint-trajectory-player" links to target
2021-01-23T03:21:56.2443460Z   "BipedalLocomotion::Planners" but the target was not found.  Perhaps a
2021-01-23T03:21:56.2444324Z   find_package() call is missing for an IMPORTED target, or an ALIAS target
2021-01-23T03:21:56.2444923Z   is missing?
2021-01-23T03:21:56.2445370Z Call Stack (most recent call first):
2021-01-23T03:21:56.2446490Z   utilities/joint-trajectory-player/CMakeLists.txt:5 (add_bipedal_locomotion_application)
2021-01-23T03:21:56.2447342Z 
2021-01-23T03:21:56.2447567Z 
~~~

At a first glance, all this jobs that failed with this error had ROBOTOLOGY_ENABLE_DYNAMICS ON and ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS OFF .

traversaro, 2021-01-23:
To avoid these kind of regressions, it could make sense to have in one of the CI jobs also a step like the following, that checks that the configuration and compilation works fine even if some dependencies are not enabled:
~~~
    - name: Check build if some dependencies are not enabled [Ubuntu]
      if: github.event_name != 'push' && github.event_name != 'release' && matrix.os == 'ubuntu-20.04'
      shell: bash
      run: |
        cd build
        for missing_dep in LibXml2 open62541 iDynTree Eigen3 YARP yaml-cpp; do
            echo "Testing ${missing_dep} as missing dependency."
            # Deselect missing dependency and build
            cmake -DIDJL_USES_${missing_dep}:BOOL=OFF .
            cmake --build . --config ${GHA_BUILD_TYPE}
            # Enable again dependency
            cmake -DIDJL_USES_${missing_dep}:BOOL=ON .
        done
~~~

traversaro, 2021-01-25:
Friendly ping @isorrentino @GiulioRomualdi

GiulioRomualdi, 2021-01-25:
> At a first glance, all this jobs that failed with this error had ROBOTOLOGY_ENABLE_DYNAMICS ON and ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS OFF .

`BipedalLocomotion::Planners` required casadi thus `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` should be set on. This is happening because `joint-trajectory-player` is compiled even if Planners is not built. 

This kind of machinery should be implemented to fix the problem.
https://github.com/dic-iit/bipedal-locomotion-framework/blob/9a5d2df5d3b350255cd56ac840d254caf4582f01/utilities/joint-position-tracking/CMakeLists.txt#L5

I'm opening a PR

PR #172 (2021-01-25): Add FRAMEWORK_COMPILE_JointTrajectoryPlayer cmake option
This fixes #171
closed 2021-01-25 by GiulioRomualdi