gbionics/bipedal-locomotion-framework #174: matioCppConversionsTestUnitTests fails on ubuntu 20.04

GiulioRomualdi, 2021-01-25:
```
13/52 Test #13: matioCppConversionsTestUnitTests ..................***Failed    0.00 sec
/home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/build/bin/matioCppConversionsTestUnitTests: error while loading shared libraries: libmatio.so.4: cannot open shared object file: No such file or directory

      Start 14: memcheck_matioCppConversionsTestUnitTests
14/52 Test #14: memcheck_matioCppConversionsTestUnitTests .........***Failed    0.72 sec
==11589== Memcheck, a memory error detector
==11589== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==11589== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==11589== Command: /home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/build/bin/matioCppConversionsTestUnitTests
==11589== 
/home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/build/bin/matioCppConversionsTestUnitTests: error while loading shared libraries: libmatio.so.4: cannot open shared object file: No such file or directory
==11589== Jump to the invalid address stated on the next line
==11589==    at 0x1030: ???
==11589==    by 0x401D9B7: fatal_error (dl-error-skeleton.c:78)
==11589==    by 0x401DA27: _dl_signal_exception (dl-error-skeleton.c:102)
==11589==    by 0x40106E1: _dl_map_object_deps (dl-deps.c:612)
==11589==    by 0x4004063: dl_main (rtld.c:1805)
==11589==    by 0x401CBAA: _dl_sysdep_start (dl-sysdep.c:252)
==11589==    by 0x400204B: _dl_start_final (rtld.c:449)
==11589==    by 0x400204B: _dl_start (rtld.c:539)
==11589==    by 0x4001107: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==11589==  Address 0x1030 is not stack'd, malloc'd or (recently) free'd
==11589== 
==11589== 
==11589== Process terminating with default action of signal 11 (SIGSEGV)
==11589==  Bad permissions for mapped region at address 0x1030
==11589==    at 0x1030: ???
==11589==    by 0x401D9B7: fatal_error (dl-error-skeleton.c:78)
==11589==    by 0x401DA27: _dl_signal_exception (dl-error-skeleton.c:102)
==11589==    by 0x40106E1: _dl_map_object_deps (dl-deps.c:612)
==11589==    by 0x4004063: dl_main (rtld.c:1805)
==11589==    by 0x401CBAA: _dl_sysdep_start (dl-sysdep.c:252)
==11589==    by 0x400204B: _dl_start_final (rtld.c:449)
==11589==    by 0x400204B: _dl_start (rtld.c:539)
==11589==    by 0x4001107: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==11589== 
==11589== HEAP SUMMARY:
==11589==     in use at exit: 0 bytes in 0 blocks
==11589==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated
==11589== 
==11589== All heap blocks were freed -- no leaks are possible
==11589== 
==11589== For lists of detected and suppressed errors, rerun with: -s
==11589== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

@S-Dafarra any ideas?

S-Dafarra, 2021-01-25:
It seems that ``matio`` has not been properly found:
```
/home/runner/work/bipedal-locomotion-framework/bipedal-locomotion-framework/build/bin/matioCppConversionsTestUnitTests: error while loading shared libraries: libmatio.so.4: cannot open shared object file: No such file or directory
```

GiulioRomualdi, 2021-01-25:
I was wondering if this is due to old cache on Github action 

@traversaro, do you know if it's possible to clean it?

S-Dafarra, 2021-01-25:
Note that probably ``runner.os`` is only ``Linux`` https://docs.github.com/en/actions/reference/context-and-expression-syntax-for-github-actions#runner-context in both 18.04 and 20.04. Probaly this line need to add also ``matrix.os``

GiulioRomualdi, 2021-02-24:
> Note that probably `runner.os` is only `Linux` https://docs.github.com/en/actions/reference/context-and-expression-syntax-for-github-actions#runner-context in both 18.04 and 20.04. Probaly this line [...]

This fixes the bug

closed 2021-02-24 by GiulioRomualdi