gbionics/bipedal-locomotion-framework #175: Python bindings test do not compile in GitHub CI

GiulioRomualdi, 2021-01-25:
I noticed that Python bindings test  are not compiled in GitHub CI

Indeed if I ran this command on my PC
```
cmake                -DBUILD_TESTING:BOOL=ON  
                     -DCMAKE_BUILD_TYPE=Release
                     -DFRAMEWORK_COMPILE_PYTHON_BINDINGS:BOOL=ON 
                     -DPython3_ROOT_DIR=$(python -c "import sys; print(sys.prefix)") 
                     -DFRAMEWORK_USE_Python3:BOOL=ON
                     -DFRAMEWORK_USE_pybind11:BOOL=ON  ..
```
The `FRAMEWORK_TEST_PYTHON_BINDINGS` options is set to false
```
 FRAMEWORK_TEST_PYTHON_BINDINGS   OFF - Dependencies unsatisfied: 'FRAMEWORK_COMPILE_tests' - FRAMEWORK_TEST_PYTHON_BINDINGS-NOTFOUND
```

If I ran `cmake` twice `FRAMEWORK_TEST_PYTHON_BINDINGS` is set to `ON`. I was wondering why this is happening.

GiulioRomualdi, 2021-01-25:
`FRAMEWORK_TEST_PYTHON_BINDINGS` should be `ON` if the dependencies are `ON`.

https://github.com/dic-iit/bipedal-locomotion-framework/blob/f180e7094ddcd7e3161245d751511b5d1dd21788/cmake/BipedalLocomotionFrameworkFindDependencies.cmake#L228-L230

@traversaro do you think it may be a bug in the cmake machinery?

traversaro, 2021-01-25:
Can you try to print the variables on which the option depends? I wonder if some of the became ON after the dependent option is defined

GiulioRomualdi, 2021-01-25:
I probably understood 

`FRAMEWORK_COMPILE_tests` is defined here
https://github.com/dic-iit/bipedal-locomotion-framework/blob/4234989a00a3fa1f42c022a88d4343597cd5c131/cmake/AddBipedalLocomotionUnitTest.cmake#L15-L17

However `BipedalLocomotionFrameworkFindDependencies` is included before `AddBipedalLocomotionUnitTest`. 
https://github.com/dic-iit/bipedal-locomotion-framework/blob/f180e7094ddcd7e3161245d751511b5d1dd21788/CMakeLists.txt#L20-L21

https://github.com/dic-iit/bipedal-locomotion-framework/blob/f180e7094ddcd7e3161245d751511b5d1dd21788/CMakeLists.txt#L71

So probably the first time `BipedalLocomotionFrameworkFindDependencies` is read by cmake `FRAMEWORK_COMPILE_tests` does not exist (and implicitly set to false)

PR #176 (2021-01-26): Move cmake dependencies in BipedalLocomotionFrameworkDependencies.cmake
This should fix #175
closed 2021-01-26 by GiulioRomualdi