gbionics/bipedal-locomotion-framework #185: YarpSensorBridge::getJointPositions() returns an empty vector if jointPositions size is zero

GiulioRomualdi, 2021-02-03:
If `jointPositions` size is 0 `YarpSensorBridge::getJointPositions()` return an empty vector
https://github.com/dic-iit/bipedal-locomotion-framework/blob/e2385fc708e4bbb2d4b2553601f3ffdd735d976d/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L235-L242

The code does not crash if compiled in release, if it's compiled in debug the following exception is thrown
```
blf-cartesian-trajectory-player: /usr/include/eigen3/Eigen/src/Core/DenseBase.h:257: void Eigen::DenseBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Ref<Eigen::Matrix<double, -1, 1> >; Eigen::Index = long int]: Assertion `rows == this->rows() && cols == this->cols() && "DenseBase::resize() does not actually allow one to resize."' failed.
```

apparently the `operator=()` do not resize the `eigen::ref`

@prashanthr05, @S-Dafarra do you have any clue>

GiulioRomualdi, 2021-02-03:
Related problem: https://stackoverflow.com/questions/46339455/workaround-for-resizing-eigenref

S-Dafarra, 2021-02-03:
Ref, like map or span cannot resize the underlying container because it does not own the memory. This is the reason why GenericContainer::Vector was designed in the first place 😁

GiulioRomualdi, 2021-02-03:
For the time being, we can add a check into `getJointPositions` and similar functions

prashanthr05, 2021-02-03:
If I remember right, we switched from the usage of GenericContainer to Eigen::Ref and added this warning saying that the user should pass a resized vector.

https://github.com/dic-iit/bipedal-locomotion-framework/blob/e2385fc708e4bbb2d4b2553601f3ffdd735d976d/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridge.h#L209-L211

S-Dafarra, 2021-02-03:
I think that this is strongly related to https://github.com/dic-iit/bipedal-locomotion-framework/issues/95. I think it is necessary to clearly define a guideline.

GiulioRomualdi, 2021-02-03:
Otherwise we may change 
```cpp
 bool YarpSensorBridge::getJointPositions(Eigen::Ref<Eigen::VectorXd> jointPositions, 
                                          double* receiveTimeInSeconds) 
```
into 

```cpp
Eigen::Ref<const Eigen::VectorXd> YarpSensorBridge::getJointPositions(double* receiveTimeInSeconds) 
```
or 
```cpp
GenericVector<const double> YarpSensorBridge::getJointPositions(double* receiveTimeInSeconds) 
```

S-Dafarra, 2021-02-03:
> Otherwise we may change
> 
> ```c++
>  bool YarpSensorBridge::getJointPositions(Eigen::Ref<Eigen::VectorXd> jointPositions, 
>                                           double* receiveTimeInSeconds) 
> ```
> 
> into
> 
> ```c++
> Eigen::Ref<const Eigen::VectorXd> YarpSensorBridge::getJointPositions(double* receiveTimeInSeconds) 
> ```
> 
> or
> 
> ```c++
> GenericVector<const double> YarpSensorBridge::getJointPositions(double* receiveTimeInSeconds) 
> ```

See https://github.com/dic-iit/bipedal-locomotion-framework/issues/124

GiulioRomualdi, 2025-03-10:
The checks added in https://github.com/ami-iit/bipedal-locomotion-framework/pull/946 mitigate the problem. Indeed now an error will be raised in case of wrong size of the vector

closed 2025-03-10 by GiulioRomualdi