gbionics/bipedal-locomotion-framework #188: The search of NAN values in the YarpSensorBridgeImpl seems fishy

S-Dafarra, 2021-02-04:
By chance, I was looking at this piece of code:
https://github.com/dic-iit/bipedal-locomotion-framework/blob/e2385fc708e4bbb2d4b2553601f3ffdd735d976d/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L1025-L1029

and I am a bit suspicious. Binary search requires the vector to be ordered, and in addition all comparisons with NAN return false (https://en.wikipedia.org/wiki/NaN#Comparison_with_NaN). Hence, binary search will always check half of the vector, and even if there is a NAN in that half, it would not detect it.

I suppose we would need some other method to detect it, like summing all the elements up and check if the result is NAN or not.

cc @prashanthr05 @GiulioRomualdi @traversaro @isorrentino

prashanthr05, 2021-02-04:
I think I had not considered the requirement of binary search requiring a sorted list when I implemented it.

We could simply change it to linear search and use https://en.cppreference.com/w/cpp/numeric/math/isnan for the moment, to avoid any bugs?

S-Dafarra, 2021-02-04:
> I think I had not considered the requirement of binary search requiring a sorted list when I implemented it.
> 
> We could simply change it to linear search and use https://en.cppreference.com/w/cpp/numeric/math/isnan for the moment, to avoid any bugs?

Yes, probably a linear search with a custom comparator could indeed work.

traversaro, 2021-02-04:
Can't we just do a for loop ? : )

prashanthr05, 2021-02-04:
> Can't we just do a for loop ? : )

Sorry for the poor choice of words. This is what I meant by linear search.

S-Dafarra, 2021-02-04:
> > Can't we just do a for loop ? : )
> 
> Sorry for the poor choice of words. This is what I meant by linear search.

That is basically what ``std::search`` does: https://en.cppreference.com/w/cpp/algorithm/search :grin:

closed 2021-02-04 by prashanthr05