gbionics/bipedal-locomotion-framework #189: Start using yarp-telemetry in the framework

GiulioRomualdi, 2021-02-04:
[`yarp-telemetry`](https://github.com/robotology-playground/yarp-telemetry) is a brand new library that allows logging data in a fancy and simple way.

It may be worth starting to think about how to use it in our library. A possible example is to extend `advanceable` and add a log function to store the dataset.

These are the limitations that I see
- [ ] `yarp-telemetry` currently supports circular buffer. 
- [ ] Right now the interface requires a reference to a std::vector as input. https://github.com/robotology-playground/yarp-telemetry/issues/49 A possible solution is to use `iDynTree::span` and `iDynTree::MatrixView` 

cc @nicogene @traversaro @S-Dafarra @isorrentino @prashanthr05

S-Dafarra, 2021-02-04:
> * [ ]  `yarp-telemetry` currently supports circular buffer.
> *

Why do you think this is a limitation? That's a sort of internal detail and the fact that has a limited size should be overcome with https://github.com/robotology-playground/yarp-telemetry/pull/55

GiulioRomualdi, 2021-02-04:
It may be required to have an infinitely long buffer. Assume that you run an application and you want to collect the data of the entire sequence. (You may need it for a paper) 
Given the current implementation, only the last `n` values are stored. 

In `WalkingLoggerModule` for instance, we store all the data

S-Dafarra, 2021-02-04:
> It may be required to have an infinitely long buffer. Assume that you run an application and you want to collect the data of the entire sequence. (You may need it for a paper)
> Given the current [...]

Yes, but you don't have infinite memory. So it is safer to specify a limit and eventually save multiple files when you reach a certain limit or condition. This is what https://github.com/robotology-playground/yarp-telemetry/pull/55 does :wink:

GiulioRomualdi, 2021-02-04:
Comment https://github.com/dic-iit/bipedal-locomotion-framework/issues/189#issue-801173818 updated after F2F discussion with @traversaro and @S-Dafarra

GiulioRomualdi, 2021-12-09:
> [ ] Right now the interface requires a reference to a std::vector as input. Investigate about a more flexible data type for storing data robotology/yarp-telemetry#49 A possible solution is to use iDynTree::span and iDynTree::MatrixView

Handled by https://github.com/robotology/yarp-telemetry/pull/144

S-Dafarra, 2021-12-17:
@GiulioRomualdi @isorrentino can we consider this done? Is the data stored reasonable?

GiulioRomualdi, 2021-12-17:
> @GiulioRomualdi @isorrentino can we consider this done? Is the data stored reasonable?

Waiting for https://github.com/ami-iit/bipedal-locomotion-framework/pull/487

paolo-viceconte, 2021-12-21:
We managed to use the new logger to log also the data coming from the walking. The modifications are included in the related branch: https://github.com/robotology/walking-controllers/tree/test/telemetry_logger.

###  iCubGenova09 Configuration

In order to use the new logger on iCubGenova09, we configured `icub-console` to have:
- bipedal-locomotion-framework in the [device/telemetry](https://github.com/ami-iit/bipedal-locomotion-framework/commits/device/telemetry) branch (with the `YarpRobotLoggerDevice` compilation enabled)
- YARP_telemetry in the [feature/nested_struct](https://github.com/GiulioRomualdi/yarp-telemetry/tree/feature/nested_struct) fork

and `icub-head` to have:
- walking-controllers in the [telemetry-logger](https://github.com/robotology/walking-controllers/tree/test/telemetry_logger) branch

### New logger usage

Store the logged data on `icub-console` and not on `icub-head`, due to potential memory issues when collecting long datasets. Once in the folder where you want to save the `.mat` file with the logged data, just start the new logger before running the WalkingModule with:
```
yarprobotinterface --config launch-yarp-robot-logger.xml
```

Close the logger wih a single `CTRL+C`. A `.mat` file with the logged data will be saved in the current folder.

### Logged data format

Recalling the previous version of the logger, data from the WalkingModule are currently logged using the following nested structure (where `::` defined the subfields):

```
// DCM
dcm::position::measured
dcm::position::desired
dcm::velocity::desired
// ZMP
zmp::measured
zmp::desired
zmp::desired_planner
// COM
com::position::measured
com::position::desired
com::position::desired_macumba
com::velocity::desired
// Left foot
left_foot::position::measured
left_foot::position::desired
left_foot::orientation::measured
left_foot::orientation::desired
left_foot::linear_velocity::measured
left_foot::angular_velocity::measured
left_foot::linear_force::measured
left_foot::angular_torque::measured
// Right foot
right_foot::position::measured
right_foot::position::desired
right_foot::orientation::measured
right_foot::orientation::desired
right_foot::linear_velocity::measured
right_foot::angular_velocity::measured
right_foot::linear_force::measured
right_foot::angular_torque::measured
// Joint
joint::positions::measured
joint::positions::desired
joint::positions::retargeting
joint::velocities::measured
```

We can obviously change the names of the stored quantities and the nested structure at our will. Please doublecheck whether we are missing useful data and feel free to suggest alternative data organization @S-Dafarra @GiulioRomualdi @isorrentino.

paolo-viceconte, 2021-12-21:
> The modifications are included in the related branch: https://github.com/robotology/walking-controllers/tree/test/telemetry_logger.

@S-Dafarra and I rebased the branch of the new logger on top of the clean up https://github.com/robotology/walking-controllers/pull/96. The currently most updated branch is [`telemetry_logger_rebased`](https://github.com/robotology/walking-controllers/commits/test/telemetry_logger_rebased).

We tested it on iCubGenova09 and it seems to be working fine.

paolo-viceconte, 2021-12-21:
Thanks to https://github.com/ami-iit/component_ANA-Avatar-XPRIZE/issues/446, we can directly inspect the collected data on `icub-console`.

GiulioRomualdi, 2021-12-22:
I will close it a soon as #487 this is merged

PR #487 (2022-01-10): Use telemetry in YarpLoggerDevice
You can try to use it. It depends on https://github.com/robotology/yarp-telemetry/pull/146

This closes #189 

cc @isorrentino @S-Dafarra @traversaro
closed 2022-01-10 by GiulioRomualdi