gbionics/bipedal-locomotion-framework #194: CppADTest fails to comple

GiulioRomualdi, 2021-02-07:
I've just install a clean ubuntu 20.04 distro on my pc. I installed catch2 and the superbuild. 

When I enabled the tests in blf I got the following error. The other tests compile (and run) without issues.

```
[ 66%] Building CXX object src/AutoDiff/tests/CMakeFiles/CppADTestUnitTests.dir/CppADTest.cpp.o
In file included from /home/gromualdi/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/src/AutoDiff/tests/CppADTest.cpp:11:
/home/gromualdi/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/src/AutoDiff/include/BipedalLocomotion/AutoDiff/CppAD.h:12:10: fatal error: cppad/cppad.hpp: No such file or directory
   12 | #include <cppad/cppad.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

GiulioRomualdi, 2021-02-07:
This is the called `make` command 

```
cd /home/gromualdi/robot-code/robotology-superbuild/build/src/bipedal-locomotion-framework/src/AutoDiff/tests && /usr/bin/c++  -DCATCH_CONFIG_DISABLE_MATCHERS -DCATCH_CONFIG_FAST_COMPILE -D_USE_MATH_DEFINES -I/home/gromualdi/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/src/AutoDiff/include -isystem /home/gromualdi/robot-install/include -isystem /usr/include/eigen3  -O3 -DNDEBUG   -std=gnu++17 -o CMakeFiles/CppADTestUnitTests.dir/CppADTest.cpp.o -c /home/gromualdi/robot-code/robotology-superbuild/src/bipedal-locomotion-framework/src/AutoDiff/tests/CppADTest.cpp

```

traversaro, 2021-02-08:
Interesting, the same tests work fine in the CI instead?

traversaro, 2021-02-08:
The compilation line is fishy, I would expected either `/home/gromualdi/robot-code/robotology-superbuild/build/install/include` or `/home/gromualdi/robot-code/robotology-superbuild/build/install/include/cppad` to be passed as includes, but none of those is passed, but rather  `-isystem /home/gromualdi/robot-install/include` is passed (for Catch2, perhaps?).

GiulioRomualdi, 2021-02-08:
`/home/gromualdi/robot-install/include` contains `catch2`
```
gromualdi@iiticublap109:~/robot-install$ tree
.
├── include
│   └── catch2
│       ├── catch.hpp
│       ├── catch_reporter_automake.hpp
│       ├── catch_reporter_sonarqube.hpp
│       ├── catch_reporter_tap.hpp
│       └── catch_reporter_teamcity.hpp
├── lib
│   ├── cmake
│   │   └── Catch2
│   │       ├── Catch2Config.cmake
│   │       ├── Catch2ConfigVersion.cmake
│   │       ├── Catch2Targets.cmake
│   │       ├── Catch2Targets-noconfig.cmake
│   │       ├── CatchAddTests.cmake
│   │       ├── Catch.cmake
│   │       └── ParseAndAddCatchTests.cmake
│   └── libCatch2WithMain.a
└── share
    ├── Catch2
    │   ├── gdbinit
    │   └── lldbinit
    ├── doc
    │   └── Catch2
    │       ├── assertions.md
    │       ├── benchmarks.md
    │       ├── ci-and-misc.md
    │       ├── cmake-integration.md
    │       ├── command-line.md
    │       ├── commercial-users.md
    │       ├── configuration.md
    │       ├── contributing.md
    │       ├── deprecations.md
    │       ├── event-listeners.md
    │       ├── generators.md
    │       ├── limitations.md
    │       ├── list-of-examples.md
    │       ├── logging.md
    │       ├── matchers.md
    │       ├── opensource-users.md
    │       ├── other-macros.md
    │       ├── own-main.md
    │       ├── Readme.md
    │       ├── release-notes.md
    │       ├── release-process.md
    │       ├── reporters.md
    │       ├── slow-compiles.md
    │       ├── test-cases-and-sections.md
    │       ├── test-fixtures.md
    │       ├── tostring.md
    │       ├── tutorial.md
    │       └── why-catch.md
    └── pkgconfig
        └── catch2.pc
```

in the bash I've 
```
export PATH=$PATH:$ROBOT_INSTALL/bin
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH:+${CMAKE_PREFIX_PATH}:}${ROBOT_INSTALL}
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH:+${LD_LIBRARY_PATH}:}${ROBOT_INSTALL}/lib
```

traversaro, 2021-02-08:
Can you report also the `CMakeCache.txt`? In particular I am curious on the value of the `cppad_INCLUDE_DIRS` variable that should contain `/home/gromualdi/robot-code/robotology-superbuild/build/install/include` and should pass it to the compiler via CMake's `cppad` imported target.

GiulioRomualdi, 2021-02-08:
bingo!

```
//cppad include directory                                                                           
cppad_INCLUDE_DIRS:PATH=
```

traversaro, 2021-02-08:
The logic in https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/cmake/Findcppad.cmake#L39 is particularly fragile, because if for any reason `cppad_INCLUDE_DIRS` ends up being filled with an empty string, then the cached value is always used and is never updated even if `_PC_cppad_INCLUDE_DIRS` has a valid value. Can you check the value of `_PC_cppad_INCLUDE_DIRS` ?

GiulioRomualdi, 2021-02-08:
Here you have
```
_PC_cppad_FOUND:INTERNAL=1
_PC_cppad_INCLUDEDIR:INTERNAL=NOTFOUND/lib
_PC_cppad_INCLUDE_DIRS:INTERNAL=
```

GiulioRomualdi, 2021-02-08:
This is the `cppad.pc.in` file
```
# -----------------------------------------------------------------------------
# CppAD: C++ Algorithmic Differentiation: Copyright (C) 2003-20 Bradley M. Bell
#
# CppAD is distributed under the terms of the
#              Eclipse Public License Version 2.0.
#
# This Source Code may also be made available under the following
# Secondary License when the conditions for such availability set forth
# in the Eclipse Public License, Version 2.0 are satisfied:
#       GNU General Public License, Version 2.0 or later.
# -----------------------------------------------------------------------------
#
# The following variable settings seem to by requried by the vcpkg packager,
# see https://www.gnu.org/prep/standards/html_node/Directory-Variables.html
# for the meaning of these variables.
#
prefix=@cppad_prefix@
exec_prefix=$(prefix)
includedir=${prefix}/@cppad_includedir@
libdir=${exec_prefix}/@cppad_libdir@
#
# See https://people.freedesktop.org/~dbn/pkg-config-guide.html
# for the meaning of the settings below:
#
Name:                  cppad
Description:           @cppad_description@
Version:               @cppad_version@
URL:                   @cppad_url@
#
Cflags:                -I${includedir}
Libs:                  -L{libdir} -l@cppad_lib_list@
Requires:
Libs.private           @cppad_libs_private@
Requires.private:      @cppad_requires_private@
```

`exec_prefix=$(prefix)` shouldn't be `exec_prefix=${prefix}`

traversaro, 2021-02-08:
That is quite strange, I guess you obtain the same on a clean build? In that case, it seems that cppad is found, but the include dirs are not available. To debug what's going on with pkg-config, you can try to run the following commands:
~~~
PKG_CONFIG_PATH=<robotology-superbuild>/build/install/share/pkgconfig pkg-config ccpad --exists --debug
PKG_CONFIG_PATH=<robotology-superbuild>/build/install/share/pkgconfig pkg-config cppad --print-variables --debug
PKG_CONFIG_PATH=<robotology-superbuild>/build/install/share/pkgconfig pkg-config cppad --cflags --debug
~~~

traversaro, 2021-02-08:
I did not saw you previous command, you can try to fix manually the generated file in `<robotology-superbuild>/build/install/share/pkgconfig/cppad.pc` and check if that fixes the problem (on a clean build/deleting the variables, otherwise you will have the same caching problems described before).

GiulioRomualdi, 2021-02-08:
`PKG_CONFIG_PATH=<robotology-superbuild>/build/install/share/pkgconfig pkg-config cppad --print-variables --debug` returns

```
Reading 'cppad' from file '/home/gromualdi/robot-code/robotology-superbuild/build/install/share/pkgconfig/cppad.pc'
Parsing package file '/home/gromualdi/robot-code/robotology-superbuild/build/install/share/pkgconfig/cppad.pc'
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>prefix=/home/gromualdi/robot-code/robotology-superbuild/build/install
 Variable declaration, 'prefix' has value '/home/gromualdi/robot-code/robotology-superbuild/build/install'
  line>exec_prefix=$(prefix)
 Variable declaration, 'exec_prefix' has value '$(prefix)'
  line>includedir=NOTFOUND/lib
 Variable declaration, 'includedir' has value 'NOTFOUND/lib'
  line>libdir=/lib
 Variable declaration, 'libdir' has value '/lib'
  line>
  line>
  line>
  line>
  line>Name:                  cppad
  line>Description:           Differentiation of C++ Algorithms
  line>Version:               20210000.2
  line>URL:                   https://coin-or.github.io/CppAD
  line>
  line>Cflags:                -I
  line>Libs:                  -L{libdir} -l-lcppad_lib
  line>Requires:
  line>Libs.private           
  line>Requires.private:      
Path position of 'cppad' is 1
Adding 'cppad' to list of known packages
exec_prefix
includedir
libdir
pcfiledir
prefix
```
`PKG_CONFIG_PATH=<robotology-superbuild>/build/install/share/pkgconfig pkg-config cppad --cflags --debug` returns

```
Parsing package file '/home/gromualdi/robot-code/robotology-superbuild/build/install/share/pkgconfig/cppad.pc'
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>
  line>prefix=/home/gromualdi/robot-code/robotology-superbuild/build/install
 Variable declaration, 'prefix' has value '/home/gromualdi/robot-code/robotology-superbuild/build/install'
  line>exec_prefix=$(prefix)
 Variable declaration, 'exec_prefix' has value '$(prefix)'
  line>includedir=NOTFOUND/lib
 Variable declaration, 'includedir' has value 'NOTFOUND/lib'
  line>libdir=/lib
 Variable declaration, 'libdir' has value '/lib'
  line>
  line>
  line>
  line>
  line>Name:                  cppad
  line>Description:           Differentiation of C++ Algorithms
  line>Version:               20210000.2
  line>URL:                   https://coin-or.github.io/CppAD
  line>
  line>Cflags:                -I
  line>Libs:                  -L{libdir} -l-lcppad_lib
  line>Requires:
  line>Libs.private           
  line>Requires.private:      
Path position of 'cppad' is 1
Adding 'cppad' to list of known packages
  pre-remove: cppad
 post-remove: cppad
adding CFLAGS_OTHER string ""
  pre-remove: cppad
 post-remove: cppad
 original: cppad
   sorted: cppad
adding CFLAGS_I string "-I "
returning flags string "-I"
-I

```

GiulioRomualdi, 2021-02-08:
I've just noticed that in CI we use the latest master (no tag no commits) :cry:

traversaro, 2021-02-08:
I think in CI probably we don't have the problem because catch also installs its headers in `${GITHUB_WORKSPACE}/install/deps/include`, so the include directory of cppad gets there through that.

GiulioRomualdi, 2021-02-08:
This commit of cppad fixes the problem https://github.com/GiulioRomualdi/CppAD/commit/6d10b8d7284d2d9efbee6d0ba2f6319e9b1fc73f

GiulioRomualdi, 2021-02-08:
Thank you @traversaro for the great help! Now we should decide what to do with the superbuild

traversaro, 2021-02-08:
I think we can wait for feedback on the CppAD-side, if the issue is fixed soon upstream we can just bump the version of CppAD that we use.

GiulioRomualdi, 2021-02-08:
This is the PR https://github.com/coin-or/CppAD/pull/95

GiulioRomualdi, 2021-02-08:
coin-or/CppAD#95 has been merged :rocket: 
I've also opened a PR in the superbuild https://github.com/robotology/robotology-superbuild/pull/617

Closing

closed 2021-02-08 by GiulioRomualdi