gbionics/bipedal-locomotion-framework #196: Add initialize function ti advanceable class

GiulioRomualdi, 2021-02-12:
I noticed that all the classes that inherit from advanceable have 
```cpp
bool initialize(std::shared_ptr<ParametersHandler::IParametersHandler> handler) override;
```

we may think to add this function also in the base class.

What do you think @S-Dafarra ??

S-Dafarra, 2021-02-12:
Yes, sounds good. I did not add it at the beginning because I was not sure yet if that was needed. If it is the case, I think it is a good move. I would simply avoid having it abstract. You can just provide a base implementation returning ``true`` all the times.

GiulioRomualdi, 2021-02-12:
Yes I agree

GiulioRomualdi, 2021-02-12:
The only problem is

https://github.com/dic-iit/bipedal-locomotion-framework/blob/bcb8ce2074ebc864a972c2d9133a8f45abb236e6/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseEstimator.h#L155

It wants as input also the model. I don't know if we can create another method to set the model only. @prashanthr05

S-Dafarra, 2021-02-12:
> The only problem is [...]

I don't think it is an issue, the implementation of the method you are proposing in ``FloatingBaseEstimator`` will simply return ``false`` as also the model needs to be set. The two versions can leave together thanks to overloading.

GiulioRomualdi, 2021-02-12:
Ok but in `FloatingBaseEstimation` there is also the following protected method

https://github.com/dic-iit/bipedal-locomotion-framework/blob/bcb8ce2074ebc864a972c2d9133a8f45abb236e6/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseEstimator.h#L235

prashanthr05, 2021-02-12:
> Ok but in `FloatingBaseEstimation` there is also the following protected method
> 
> https://github.com/dic-iit/bipedal-locomotion-framework/blob/bcb8ce2074ebc864a972c2d9133a8f45abb236e6/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseEstimator.h#L235

We can make it public. Indeed, I make it public in the future rolling versions so that we can use the estimator independent from the Model/KinDynComputations for scenarios of visual-inertial odometry. So the user can choose to use the corresponding function call when using a relevant estimator.

PR #267 (2021-04-01): Refactoring of advanceable
This PR refactors the advanceable class, in details:
- Now advanceable has two template parameters, Input and Output
- I Implemented Sink and Source classes as discussed in #264 
- I modified the already existing code in the repo. 

This PR closes #264 and closes #196
closed 2021-04-01 by GiulioRomualdi