gbionics/bipedal-locomotion-framework #197: Things to do in v0.0.1

GiulioRomualdi, 2021-02-17:
- [ ] Adjust the readme #148 

@S-Dafarra @traversaro @prashanthr05 @isorrentino

diegoferigo, 2021-02-17:
Are we going to follow [SemVer](https://semver.org/) to cycle the releases or do we wait a bit longer to let the APIs become more mature?

GiulioRomualdi, 2021-02-17:
> Are we going to follow SemVer to cycle the releases or do we wait a bit longer to let the APIs become more mature?

In theory yes, we can use `SemVer`. However, this shouldn't block us from the development of the library. At this point, we can easily break the API but we have to remember to change the version accordingly.

GiulioRomualdi, 2021-02-17:
By the way, at this point, we may release `0.1.0` (Check [here](https://semver.org/#how-should-i-deal-with-revisions-in-the-0yz-initial-development-phase)

diegoferigo, 2021-02-17:
Cool, I was not aware that in SemVer the releases 0.x are not API stable.

GiulioRomualdi, 2021-02-22:
`blf` v0.1.0 has been released today https://github.com/dic-iit/bipedal-locomotion-framework/releases/tag/v0.1.0

closed 2021-02-22 by GiulioRomualdi