gbionics/bipedal-locomotion-framework #206: Introduces submodules in the python bindings

GiulioRomualdi, 2021-02-24:
I was thinking that we may try to replicate the structure of the namespace with submodules in the python bindings. This will simplify the compilation of the bindings depending on the installed dependencies.

cc @traversaro @diegoferigo @paolo-viceconte

diegoferigo, 2021-02-24:
The current install tree of the bindings is the following:

```
~ via 🅒 /conda
❯ tree /usr/local/src/robotology-superbuild/build/install/lib/python3.8/dist-packages/bipedal_locomotion_framework/                          (/conda) 
/usr/local/src/robotology-superbuild/build/install/lib/python3.8/dist-packages/bipedal_locomotion_framework/
├── __init__.py
└── bindings.cpython-38-x86_64-linux-gnu.so
```

I see two ways for creating bindings with optional components:

1. Using different modules. This means a different cpython shared library for each module and a `__init__.py` that is dynamically populated by cmake (similar to [iDynTree](https://github.com/robotology/idyntree/blob/fced0b0d6635d8a1eeb87ae98e4e8e853928657a/bindings/CMakeLists.txt#L96-L121)).
1. Keeping a single module and operate on the sources by using macros to enable / disable the inclusion of binded classes ([here](https://github.com/dic-iit/bipedal-locomotion-framework/blob/566b62cb4f42b34912233e493cdd9421757b04a4/bindings/python/bipedal_locomotion_framework.cpp#L13-L51)).

Both methods have pros and cons. With different modules is more likely that something changes in the C++ side, and in order to make Python match, modules structure have to change and downstream code could be affected (even though python is quite good to prevent that). With macros, well, they are macros. Not much more to say. I would not discard anyway them since they could be quite effective in this scenario and simple to implement.

Note that 1 breaks in any case current downstream code (that, as already discussed in #162, is not much and it is ok to break it now). I'm ok with either of the solutions.

GiulioRomualdi, 2021-05-06:
This issue has been achieved by https://github.com/dic-iit/bipedal-locomotion-framework/pull/238