gbionics/bipedal-locomotion-framework #212: [Python Bindings] incompatible function arguments error for generated binding methods

prashanthr05, 2021-02-25:
I started with the bindings generation for ContactDetectors library. I start simple by creating pybind for this class SchmittTriggerUnit 
https://github.com/dic-iit/bipedal-locomotion-framework/blob/cc2fd01416bd39c3d2cc149f40ce2dc880e0cb0e/src/Contacts/include/BipedalLocomotion/ContactDetectors/SchmittTriggerDetector.h#L152 

``` cpp
void BipedalLocomotion::bindings::CreateSchmittTriggerUnit(pybind11::module& module){
    namespace py = ::pybind11;
    using namespace BipedalLocomotion::Contacts;
    py::class_<SchmittTriggerInput>(module, "SchmittTriggerInput")
        .def(py::init())
        .def_readwrite("time", &SchmittTriggerInput::time)
        .def_readwrite("value", &SchmittTriggerInput::value);
    py::class_<SchmittTriggerParams>(module, "SchmittTriggerParams")
        .def(py::init())
        .def_readwrite("onThreshold", &SchmittTriggerParams::onThreshold)
        .def_readwrite("offThreshold", &SchmittTriggerParams::offThreshold)
        .def_readwrite("switchOnAfter", &SchmittTriggerParams::switchOnAfter)
        .def_readwrite("switchOffAfter", &SchmittTriggerParams::switchOffAfter);
    py::class_<SchmittTriggerUnit>(module, "SchmittTriggerUnit")
        .def(py::init())
        .def("set_state", py::overload_cast<const bool& >(&SchmittTriggerUnit::setState))
        .def("set_state", py::overload_cast<const bool&, const double& >(&SchmittTriggerUnit::setState))
        .def("set_params", &SchmittTriggerUnit::setParams)
        .def("update", &SchmittTriggerUnit::update)
        .def("reset", &SchmittTriggerUnit::reset)
        .def("get_state", py::overload_cast<>(&SchmittTriggerUnit::getState))
        .def("get_state",
             [](SchmittTriggerUnit& impl) {
                 double switchTime;
                 bool ok = impl.getState(switchTime);
                 return std::make_tuple(ok, switchTime);
             })
        .def("get_params", &SchmittTriggerUnit::getParams);
}
```

It compiles successfully. But while calling the methods in python I get the error,
``` python
In [12]: s.set_state(True)                                                                           
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-12-ebd91c2ae8b8> in <module>
----> 1 s.set_state(True)
TypeError: set_state(): incompatible function arguments. The following argument types are supported:
    1. (self: bipedal_locomotion_framework.bindings.SchmittTriggerUnit, arg0: bool) -> None
    2. (self: bipedal_locomotion_framework.bindings.SchmittTriggerUnit, arg0: bool, arg1: float) -> None
Invoked with: True
```
Anyone has faced similar issues? @diegoferigo @GiulioRomualdi 

Update: @GiulioRomualdi suggested to not use the same names for generated python methods. I tried that as well, but still get the same error.

prashanthr05, 2021-02-25:
When @GiulioRomualdi tested on his local machine, it worked as expected.

Apparently, my poor python skills was the problem.

The problem was that  I was instantiating the object as `s = blf.SchmittTriggerUnit` instead of `s = blf.SchmittTriggerUnit()`.

Closing issue.

closed 2021-02-25 by prashanthr05

diegoferigo, 2021-02-25:
I was about to ask the full code :) Forgetting the `()` means that you were passing `self=True` to the class method (not to the instance --object-- of a class).

prashanthr05, 2021-02-25:
> I was about to ask the full code :) Forgetting the `()` means that you were passing `self=True` to the class method (not to the instance --object-- of a class).

Understood. Thanks for the explanation. :)