gbionics/bipedal-locomotion-framework #217: Install the python package in python3 folder

GiulioRomualdi, 2021-02-28:
I noticed that if you compile the python bindings from the superbuild you will find them in 
`lib/python3/dist-packages` while `blf` installs the bindings in `lib/python3.8/dist-packages`.

Is there something blocking to uniform the installation folder? 

cc @diegoferigo and @traversaro

traversaro, 2021-02-28:
@PeterBowman opened an issue on this in YARP, see https://github.com/robotology/yarp/issues/2511 . In a nutshell, it seems that iDynTree and YARP install in `<prefix>/lib/python3/dist-packages` because this is the value returned by [`distutils.sysconfig.get_python_lib(...)`](https://docs.python.org/3/distutils/apiref.html#distutils.sysconfig.get_python_lib). 

However, for some reason (that I do not know if it is related to Ubuntu/Debian packaging of Python or what) if the `<prefix>` is `/usr/local`, the  `/usr/local/lib/python3/dist-packages` directory is not part of the default search path of the Python interpreter, while `/usr/local/lib/python3.8/dist-packages` it is. 

At the superbuild level, for now I would continue to install in `<prefix>/lib/python3/dist-packages`, and to support that in blf we either switch to install there, or provide some kind of option to specify the desired Python install directory in https://github.com/robotology/robotology-superbuild/blob/v2021.02/cmake/Buildbipedal-locomotion-framework.cmake#L39 .

traversaro, 2021-02-28:
By the way, fixing the logic in https://github.com/dic-iit/bipedal-locomotion-framework/blob/master/bindings/CMakeLists.txt#L15 to use `distutils.sysconfig.get_python_lib(...)` (option A in the previous comment) to compute the Python installation prefix would also simplify the installation of bipedal-locomotion-framework in conda, see https://github.com/robotology/robotology-superbuild/issues/641 for a related comment.

traversaro, 2021-02-28:
Related issues: 
* https://github.com/robotology/robotology-superbuild/issues/641
* https://github.com/robotology/yarp/issues/2511

closed 2021-04-12 by GiulioRomualdi