gbionics/bipedal-locomotion-framework #241: Investigate differences on TimeVaryingDCMPlannerUnitTests execution time depending on dependency installation method

traversaro, 2021-03-18:
As a follow up for https://github.com/dic-iit/bipedal-locomotion-framework/issues/219, it turns out that installing ipopt+mumps via coinbrew is now sensibly faster (up to 10x) over other alternatives. In this issue, we will investigate why this is the case, as the coinbrew installation is a non-standard step that complexifies deployment. 

To have a reproducible environment, I run the tests in a dedicated GitHub Action workflow. Even if GitHub Action machines can change overtime and will not reflect exactly the hardware in which we run the walking controller on the real robot, from preliminary investigation they were able to reproduce the macroscopic performance differences in https://github.com/dic-iit/bipedal-locomotion-framework/issues/219, so it is worth to use it for the time being.

traversaro, 2021-03-18:
In https://github.com/traversaro/ipopt-walking-benchmarks/runs/2133731892 I collected the first results, with five different jobs. The tests are run 10 times to ensure that the variance across runs is not impacting the analysis. 

| Name |  Time spent executing TimeVaryingDCMPlannerUnitTests  (s)  | Description |
|:-------:|:----------------:|:-----------------------------------------------------------------------------:|
| `conda-manual` |      2.1        | All dependencies installed via conda (conda-forge, robotology-staging), bfl compiled manually | 
| `conda-superbuild` |      2.3        | mumps, ipopt installed via conda (conda-forge), casadi and bfl installed via superbuild | 
| `conda-coinbrew-superbuild` |      0.25      | mumps, ipopt installed via coinbrew (conda-forge), casadi and bfl installed via superbuild with conda-forge deps | 
| `apt-superbuild` | 0.8 | mumps, ipopt installed via apt, casadi and blf sia superbuild | 
| `apt-coinbrew-superbuild` | 0.21 | mumps, ipopt installed via coinbrew, casadi and blf sia superbuild with apt deps |

traversaro, 2021-03-18:
The nice news is that the major speedup obtained by using coinbrew is there for both apt and conda, so it could be related to the use of mumps 4 (apt and conda all use mumps 5.2.1). Interestingly, the coinbrew builds are associated with a low number of iterations of the NLP problem (see  the following outputs), so the issue it not simply that mumps 4 is faster, it is really being called a different number of times. 

## conda-manual
~~~
2021-03-17T19:02:16.3325145Z     Start 17: TimeVaryingDCMPlannerUnitTests
2021-03-17T19:02:16.3325933Z 
2021-03-17T19:02:16.3328335Z 17: Test command: /home/runner/work/ipopt-walking-benchmarks/ipopt-walking-benchmarks/bipedal-locomotion-framework/build/bin/TimeVaryingDCMPlannerUnitTests
2021-03-17T19:02:16.3330091Z 17: Test timeout computed to be: 1500
2021-03-17T19:02:16.3392688Z 17: [2021-03-17 19:02:16.338] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'linear_solver' not found.
2021-03-17T19:02:16.3395063Z 17: [2021-03-17 19:02:16.338] [blf] [info] [TimeVaryingDCMPlanner::initialize] linear_solver not found. The following parameter will be used mumps.
2021-03-17T19:02:16.3398647Z 17: [2021-03-17 19:02:16.338] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'use_external_dcm_reference' not found.
2021-03-17T19:02:16.3404653Z 17: [2021-03-17 19:02:16.338] [blf] [info] [TimeVaryingDCMPlanner::initialize] use_external_dcm_reference not found. The following parameter will be used false.
2021-03-17T19:02:16.3407116Z 17: [2021-03-17 19:02:16.338] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'gravity' not found.
2021-03-17T19:02:16.3408952Z 17: [2021-03-17 19:02:16.338] [blf] [info] [TimeVaryingDCMPlanner::initialize] gravity not found. The following parameter will be used 9.80665.
2021-03-17T19:02:16.4831375Z 17: 
2021-03-17T19:02:16.4832258Z 17: ******************************************************************************
2021-03-17T19:02:16.4835252Z 17: This program contains Ipopt, a library for large-scale nonlinear optimization.
2021-03-17T19:02:16.4836618Z 17:  Ipopt is released as open source code under the Eclipse Public License (EPL).
2021-03-17T19:02:16.4837975Z 17:          For more information visit https://github.com/coin-or/Ipopt
2021-03-17T19:02:16.4838863Z 17: ******************************************************************************
2021-03-17T19:02:16.4839414Z 17: 
2021-03-17T19:02:18.4199102Z 17:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
2021-03-17T19:02:18.4200425Z 17:        nlp_f  |   1.24ms ( 21.02us)   1.25ms ( 21.11us)        59
2021-03-17T19:02:18.4201323Z 17:        nlp_g  |   2.07ms ( 35.05us)   2.00ms ( 33.87us)        59
2021-03-17T19:02:18.4202014Z 17:     nlp_grad  | 109.00us (109.00us) 109.10us (109.10us)         1
2021-03-17T19:02:18.4202751Z 17:   nlp_grad_f  |   1.12ms ( 46.71us)   1.05ms ( 43.72us)        24
2021-03-17T19:02:18.4203454Z 17:   nlp_hess_l  |   1.71ms ( 85.60us)   1.65ms ( 82.31us)        20
2021-03-17T19:02:18.4204149Z 17:    nlp_jac_g  |   2.10ms ( 80.77us)   1.99ms ( 76.42us)        26
2021-03-17T19:02:18.4204808Z 17:        total  |   2.06 s (  2.06 s)   1.97 s (  1.97 s)         1
2021-03-17T19:02:18.4282892Z 17: ===============================================================================
2021-03-17T19:02:18.4283799Z 17: All tests passed (157 assertions in 1 test case)
2021-03-17T19:02:18.4284412Z 17: 
2021-03-17T19:02:18.4301747Z     Test #17: TimeVaryingDCMPlannerUnitTests ...   Passed    2.10 sec
~~~

## conda-superbuild
~~~
2021-03-17T19:28:54.5940652Z     Start 17: TimeVaryingDCMPlannerUnitTests
2021-03-17T19:28:54.5942230Z 
2021-03-17T19:28:54.5945279Z 17: Test command: /home/runner/work/ipopt-walking-benchmarks/ipopt-walking-benchmarks/robotology-superbuild/build/src/bipedal-locomotion-framework/bin/TimeVaryingDCMPlannerUnitTests
2021-03-17T19:28:54.5947548Z 17: Test timeout computed to be: 1500
2021-03-17T19:28:54.5998799Z 17: [2021-03-17 19:28:54.599] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'linear_solver' not found.
2021-03-17T19:28:54.6001279Z 17: [2021-03-17 19:28:54.599] [blf] [info] [TimeVaryingDCMPlanner::initialize] linear_solver not found. The following parameter will be used mumps.
2021-03-17T19:28:54.6003041Z 17: [2021-03-17 19:28:54.599] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'use_external_dcm_reference' not found.
2021-03-17T19:28:54.6004839Z 17: [2021-03-17 19:28:54.599] [blf] [info] [TimeVaryingDCMPlanner::initialize] use_external_dcm_reference not found. The following parameter will be used false.
2021-03-17T19:28:54.6007519Z 17: [2021-03-17 19:28:54.599] [blf] [error] [StdImplementation::getParameterPrivate] Parameter named 'gravity' not found.
2021-03-17T19:28:54.6009758Z 17: [2021-03-17 19:28:54.599] [blf] [info] [TimeVaryingDCMPlanner::initialize] gravity not found. The following parameter will be used 9.80665.
2021-03-17T19:28:54.7339348Z 17: 
2021-03-17T19:28:54.7340344Z 17: ******************************************************************************
2021-03-17T19:28:54.7341909Z 17: This program contains Ipopt, a library for large-scale nonlinear optimization.
2021-03-17T19:28:54.7342950Z 17:  Ipopt is released as open source code under the Eclipse Public License (EPL).
2021-03-17T19:28:54.7344317Z 17:          For more information visit https://github.com/coin-or/Ipopt
[... cut ...]
2021-03-17T19:26:50.1779977Z 17:    nlp_jac_g  |   1.01ms ( 52.89us)   1.01ms ( 53.05us)        19
2021-03-17T19:26:50.1780795Z 17:        total  |  99.68ms ( 99.68ms)  99.70ms ( 99.70ms)         1
2021-03-17T19:26:50.1833321Z 17: ===============================================================================
2021-03-17T19:26:50.1834346Z 17: All tests passed (157 assertions in 1 test case)
2021-03-17T19:26:50.1835194Z 17: 
2021-03-17T19:26:50.1842930Z     Test #17: TimeVaryingDCMPlannerUnitTests ...   Passed    0.20 sec
~~~

GiulioRomualdi, 2021-04-05:
It would be interesting also to run the test with `pardiso`.
Accordingly to the documentation `ipopt` can be used with `pardiso` https://coin-or.github.io/Ipopt/INSTALL.html#DOWNLOAD_PARDISO

However I was not able to use it since pardiso seems to be incompatible with the `lapack`/`blas` installation provided in ubuntu. Indeed when I compile coinbrew with
```
./coinbrew build Ipopt --prefix=install --with-pardiso="$HOME/robot-install/lib/libpardiso600-GNU800-X86-64.so -fopenmp -lgfortran" --test --no-prompt --verbosity=3
```

I got the following errors
```
configure:17101: checking for function pardiso_ipopt_newinterface_ in -llapack -lblas /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so -fopenmp -lgfortran  
configure:17129: gcc -o conftest -O2 -DNDEBUG    conftest.c -llapack -lblas /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so -fopenmp -lgfortran   >&5
/usr/bin/ld: /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so: undefined reference to `dtrmm_'
/usr/bin/ld: /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so: undefined reference to `lsame_'
/usr/bin/ld: /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so: undefined reference to `dlartg_'
/usr/bin/ld: /home/gromualdi/robot-install/lib/libpardiso600-GNU800-X86-64.so: undefined reference to `dswap_'
```

A possible solution is to switch to MKL intel library. I tried but the migration is not straightforward

traversaro, 2021-04-05:
Just FYI, as conda + coinbrew is approximately as fast as apt + coinbrew, a possible option is also to install mkl via conda-forge, see https://conda-forge.org/docs/maintainer/knowledge_base.html#switching-blas-implementation .

GiulioRomualdi, 2021-04-14:
I tried to run the same test with `snopt`. The time performances follow. 

TLDR. IPOPT is faster than SNOPT

```
 ==============================
    SNOPT  C interface  2.2.0   
 ==============================
 S N O P T  7.7.7    (Feb 2021)
 ==============================

 SNMEMB EXIT 100 -- finished successfully
 SNMEMB INFO 104 -- memory requirements estimated

 Trial version of SNOPT -- for evaluation or academic purposes only

 Nonlinear constraints    1697     Linear constraints       0
 Nonlinear variables      1124     Linear variables         0
 Jacobian  variables      1124     Objective variables   1124
 Total constraints        1697     Total variables       1124

 The user has defined    4695   out of    4695   constraint gradients.
 The user has defined    1124   out of    1124   objective  gradients.

        Minor NonOpt  QP mult  QP step   rgNorm Elastic QP obj     nS
          100     68  2.0E-01  1.0E+00  2.9E-11  5.3731455E+03     92

 Major Minors     Step   nCon Feasible  Optimal  MeritFunction     nS Penalty
     0    107               1  1.7E-01  7.5E-01  1.3202648E-01     99           r iT

        Minor NonOpt  QP mult  QP step   rgNorm Elastic QP obj     nS
          100     26 -2.1E-01  1.0E+00  4.0E-13 -2.6961898E+02    180

 Major Minors     Step   nCon Feasible  Optimal  MeritFunction     nS Penalty
     1    118  1.0E+00      2  8.2E-03  3.5E-02  4.5688665E+02    198 3.0E+03   r iT
     2     35  4.1E-01      4  4.9E-03  2.1E-02  2.4680235E+02    226 1.5E+02 s   i
     3     16  1.0E+00      5  8.9E-06  5.0E-04  1.6299024E+02    238 6.9E+01     i
     4     12  1.0E+00      6  3.1E-06  3.3E-04  1.2835157E+02    245 6.9E+01     i
     5      5  1.0E+00      7  5.0E-06  2.6E-04  1.0482914E+02    249 6.9E+01     i
     6      8  1.0E+00      8  2.3E-06  1.7E-04  9.2158457E+01    252 6.9E+01     i
     7      4  1.0E+00      9  3.7E-06  1.3E-04  8.2040395E+01    253 6.9E+01     i
     8      6  1.0E+00     10  3.5E-06  1.3E-04  7.5189058E+01    256 6.9E+01     i
     9      9  1.0E+00     11  2.7E-06  1.1E-04  6.9609409E+01    256 6.9E+01     i

 Major Minors     Step   nCon Feasible  Optimal  MeritFunction     nS Penalty
    10      6  1.0E+00     12  1.9E-06  1.1E-04  6.4988621E+01    259 6.9E+01     i
    11      9  1.0E+00     13  1.8E-06  1.0E-04  6.0798673E+01    259 6.9E+01     i
    12      4  1.0E+00     14  1.6E-06  9.3E-05  5.7302221E+01    262 6.9E+01   R i
    13      5  1.0E+00     15 (6.9E-08) 6.2E-05  5.5947757E+01    264 6.9E+01 s   i
    14      3  1.0E+00     16  1.0E-06  1.5E-04  5.5268399E+01    266 6.9E+01     i
    15      4  1.0E+00     17 (3.6E-07) 7.2E-05  5.3862150E+01    269 6.9E+01     i
    16      2  1.0E+00     18 (2.1E-07) 6.2E-05  5.3474768E+01    270 6.9E+01     i
    17      3  1.0E+00     19 (3.2E-07) 2.9E-05  5.2699865E+01    272 6.9E+01     i
    18      4  1.0E+00     20  2.0E-06  8.5E-05  5.2179207E+01    275 6.9E+01     i
    19      5  1.0E+00     21  2.1E-06  4.4E-05  5.1431516E+01    279 6.9E+01     i

 Major Minors     Step   nCon Feasible  Optimal  MeritFunction     nS Penalty
    20      2  1.0E+00     22  2.1E-06  6.6E-05  5.1108981E+01    280 6.9E+01     i
    21      4  1.0E+00     23 (6.1E-07) 3.3E-05  5.0595279E+01    281 6.9E+01     i
    22      4  1.0E+00     24  2.0E-06  8.1E-05  5.0411760E+01    284 6.9E+01     i
    23      3  1.0E+00     25 (8.8E-07) 2.7E-05  4.9891055E+01    286 6.9E+01   R i
    24      2  4.9E-01     27 (4.6E-07) 1.7E-05  4.9767471E+01    287 6.9E+01 s   i
    25      3  1.0E+00     28 (5.0E-08) 1.2E-05  4.9695105E+01    289 6.9E+01     i
    26      2  1.0E+00     29 (2.5E-07) 1.8E-05  4.9538009E+01    290 6.9E+01     i
    27      3  1.0E+00     30 (9.1E-07) 1.7E-05  4.9354306E+01    292 6.9E+01     i
    28      2  1.0E+00     31 (5.8E-07) 1.6E-05  4.9244687E+01    293 6.9E+01     i
    29      3  1.0E+00     32 (7.5E-07) 1.3E-05  4.9131183E+01    295 6.9E+01     i

 Major Minors     Step   nCon Feasible  Optimal  MeritFunction     nS Penalty
    30      3  1.0E+00     33  1.2E-06  1.8E-05  4.8987378E+01    297 6.9E+01     i
[... 88 lines cut ...]

 Problem name                 solver
 No. of iterations                 586   Objective            4.5910580616E+01
 No. of major iterations           101   Linear    obj. term  0.0000000000E+00
 Penalty parameter           6.855E+01   Nonlinear obj. term  4.5910580616E+01
 User function calls (total)       113
 No. of superbasics                371   No. of basic nonlinears           743
 No. of degenerate steps             6   Percentage                       1.02
 Max x                     431 4.3E+00   Max pi                   1697 1.0E+10
 Max Primal infeas        1134 2.2E-07   Max Dual infeas           722 1.9E-01
 Nonlinear constraint violn    2.2E-07

 Solution printed on file  10

 Time for MPS input                             0.00 seconds
 Time for solving problem                       2.20 seconds
 Time for solution output                       0.02 seconds
 Time for constraint functions                  0.07 seconds
 Time for objective function                    0.00 seconds
      solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
    nlp_grad  |  95.00us ( 95.00us)  94.35us ( 94.35us)         1
   nlp_jac_f  |   3.96ms ( 34.43us)   3.89ms ( 33.87us)       115
   nlp_jac_g  |   5.75ms ( 49.97us)   5.76ms ( 50.08us)       115
       total  |   2.21 s (  2.21 s)   2.21 s (  2.21 s)         1
```

traversaro, 2021-06-01:
This is still interesting to investigate, but at this point I would wait for the new Ipopt 3.14  to investigate this, in the meanwhile I remove myself as assignee to more easily track by immediate backlog.

traversaro, 2021-06-16:
I wonder if this is related: https://github.com/coin-or/Ipopt/issues/491 .

traversaro, 2021-07-04:
> It would be interesting also to run the test with `pardiso`.

FYI a conda build with pardiso mkl is being prepared in https://github.com/conda-forge/ipopt-feedstock/pull/65 . Note that pardiso mkl may be slower then normal pardiso (see https://www.pardiso-project.org/) so effectively pardisomkl and pardiso are two different solvers, but neverthless it would be nice to test it.

traversaro, 2021-07-05:
I re-run the test with latest coinbrew and latest blf and latest conda packages. Interesting, the latest coinbrew is now installing MUMPS 5.* , see https://github.com/coin-or/Ipopt/issues/479#issuecomment-850474322 . Furthermore, Ipopt has been updated to 3.14.1 in coinbrew and conda, while on apt it remains to the old version 3.11.9, see https://repology.org/project/ipopt/versions . See https://github.com/traversaro/ipopt-walking-benchmarks/runs/2983648755 . Note that is possible that also the machine instances used by GitHub Actions changed, so the only relevant thing to check are relativing timings. 

The new results are:

| Name |  Time spent executing TimeVaryingDCMPlannerUnitTests  (s)  | Description |
|:-------:|:----------------:|:----------------------------:|
| `conda-manual` |     1.85        | All dependencies installed via conda (conda-forge, robotology-staging), bfl compiled manually | 
| `conda-superbuild` |      3.5        | mumps, ipopt installed via conda (conda-forge), casadi and bfl installed via superbuild | 
| `conda-coinbrew-superbuild` |      0.9     | mumps, ipopt installed via coinbrew (conda-forge), casadi and bfl installed via superbuild with conda-forge deps | 
| `apt-superbuild` | 0.72 | mumps, ipopt installed via apt, casadi and blf sia superbuild | 
| `apt-coinbrew-superbuild` | 0.5 | mumps, ipopt installed via coinbrew, casadi and blf sia superbuild with apt deps |

traversaro, 2021-07-05:
What is really strange, even w.r.t. to previous builds, is the different betwen `conda-manual` and `conda-superbuild`. In theory the only difference there is the version of casadi (built via superbuilda vs built in conda), and yet the number of iterations and the time spent is completly different. Something is fishy.

traversaro, 2021-07-06:
One difference that is particularly strange (especially considering that is not just a timing difference, but a difference in the number of the iterations of the ipopt algorithm) is the difference between `conda-manual` and `conda-superbuild` . Both this builds use exactly the same ipopt and mumps version, and also the casadi version is the same, the only difference should be that in one case casadi is taken from conda-forge binaries, and in another it is built by the superbuild. To avoid polluting this issue again, I will open a separate issue for that.

traversaro, 2021-07-06:
As emerged in https://github.com/traversaro/ipopt-walking-benchmarks/issues/2, the major difference between apt and conda is that in apt the default blas implementation is netlib, while on conda is openblas on Linux/macOS and mkl on Windows. By switching to using netlib on conda as well, we finally get a consistent number of iterations (from https://github.com/traversaro/ipopt-walking-benchmarks/runs/2998363108):

## `conda-coinbrew-superbuild` 

~~~
 20: ******************************************************************************
20: This program contains Ipopt, a library for large-scale nonlinear optimization.
20:  Ipopt is released as open source code under the Eclipse Public License (EPL).
20:          For more information visit https://github.com/coin-or/Ipopt
20: ******************************************************************************
20: 
20:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
20:        nlp_f  | 907.00us ( 18.51us) 904.47us ( 18.46us)        49
20:        nlp_g  |   1.39ms ( 28.45us)   1.37ms ( 27.89us)        49
20:     nlp_grad  |  95.00us ( 95.00us)  94.41us ( 94.41us)         1
20:   nlp_grad_f  | 740.00us ( 33.64us) 779.66us ( 35.44us)        22
20:   nlp_hess_l  |   1.30ms ( 68.32us)   1.30ms ( 68.30us)        19
20:    nlp_jac_g  |   1.47ms ( 63.74us)   1.46ms ( 63.67us)        23
20:        total  | 358.38ms (358.38ms) 358.52ms (358.52ms)         1
20: ===============================================================================
20: All tests passed (157 assertions in 1 test case)
20: 
    Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    0.50 sec
    Start 20: TimeVaryingDCMPlannerUnitTests
~~~

## `conda-superbuild` 
~~~
 20: ******************************************************************************
20: This program contains Ipopt, a library for large-scale nonlinear optimization.
20:  Ipopt is released as open source code under the Eclipse Public License (EPL).
20:          For more information visit https://github.com/coin-or/Ipopt
20: ******************************************************************************
20: 
20:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
20:        nlp_f  | 762.00us ( 15.55us) 761.71us ( 15.55us)        49
20:        nlp_g  |   1.25ms ( 25.55us)   1.22ms ( 24.96us)        49
20:     nlp_grad  |  91.00us ( 91.00us)  91.10us ( 91.10us)         1
20:   nlp_grad_f  | 652.00us ( 29.64us) 645.81us ( 29.35us)        22
20:   nlp_hess_l  |   1.12ms ( 59.16us)   1.14ms ( 59.75us)        19
20:    nlp_jac_g  |   1.17ms ( 51.04us)   1.18ms ( 51.21us)        23
20:        total  |   1.40 s (  1.40 s)   1.37 s (  1.37 s)         1
20: ===============================================================================
20: All tests passed (157 assertions in 1 test case)
20: 
1/1 Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    1.46 sec
~~~

## `conda-manual` 
~~~
20: ******************************************************************************
20: This program contains Ipopt, a library for large-scale nonlinear optimization.
20:  Ipopt is released as open source code under the Eclipse Public License (EPL).
20:          For more information visit https://github.com/coin-or/Ipopt
20: ******************************************************************************
20: 
20:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
20:        nlp_f  | 870.00us ( 17.76us) 871.91us ( 17.79us)        49
20:        nlp_g  |   1.41ms ( 28.71us)   1.38ms ( 28.18us)        49
20:     nlp_grad  |  99.00us ( 99.00us)  99.20us ( 99.20us)         1
20:   nlp_grad_f  | 894.00us ( 40.64us) 890.21us ( 40.46us)        22
20:   nlp_hess_l  |   1.42ms ( 74.58us)   1.42ms ( 74.66us)        19
20:    nlp_jac_g  |   1.51ms ( 65.70us)   1.51ms ( 65.65us)        23
20:        total  |   1.50 s (  1.50 s)   1.47 s (  1.47 s)         1
20: ===============================================================================
20: All tests passed (157 assertions in 1 test case)
20: 
    Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    1.57 sec
~~~

[... 19 lines cut ...]
1/1 Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    0.50 sec
~~~

## `apt-superbuild` 

~~~
20: ******************************************************************************
20: This program contains Ipopt, a library for large-scale nonlinear optimization.
20:  Ipopt is released as open source code under the Eclipse Public License (EPL).
20:          For more information visit http://projects.coin-or.org/Ipopt
20: ******************************************************************************
20: 
20:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
20:        nlp_f  | 935.00us ( 15.85us) 919.16us ( 15.58us)        59
20:        nlp_g  |   1.57ms ( 26.58us)   1.57ms ( 26.62us)        59
20:     nlp_grad  | 102.00us (102.00us) 101.81us (101.81us)         1
20:   nlp_grad_f  | 988.00us ( 31.87us)   1.01ms ( 32.49us)        31
20:   nlp_hess_l  |   1.78ms ( 63.50us)   1.78ms ( 63.55us)        28
20:    nlp_jac_g  |   1.89ms ( 57.15us)   1.89ms ( 57.26us)        33
20:        total  | 588.70ms (588.70ms) 588.88ms (588.88ms)         1
20: ===============================================================================
20: All tests passed (157 assertions in 1 test case)
20: 
1/1 Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    0.71 sec
~~~

traversaro, 2021-07-06:
Except for `apt-superbuild`  that is using a really old version of Ipopt, all the other methods now exit in the same number of iterations (49). Furthermore, when coinbrew is used to compile mumps and ipopt, the performance are exactly the same on conda and on apt, with 0.5 s (even if the performance are worst then when mumps 4 was used). 

At this point, it seems much more probable that the difference between the computation time is all in the linear solver (even because the time spent in the function evaluaton between all the case is comparable, and in any case is always around 6/7 ms). In particular, the difference between the conda mumps and the coinbrew mumps (resulting in 1.5 s vs 0.5 s) need to be investigate, but it is probably an effect of the specific variant built in conda (with/without metis, and similar options) and by the specific build of mumps used in ipopt built (both mumps-mpi and mump-seq are available in conda, but only mumps-seq is linked to ipopt).

traversaro, 2021-07-06:
As a general comment on TimeVaryingDCMPlannerUnitTests  performances, probably as mentioned in https://github.com/traversaro/ipopt-walking-benchmarks/issues/2 it may be worth to try to tune termination parameters:
> The number of iterations vary a lot across blas implementations. It is 71 for mkl, 69/34 for openblas, 49 for netlib, 47 for blis. I am not sure about this, but I am afraid this is an indication that the default termination condition/threshold of ipopt may not be appropriate/ideal for this problem. **TL;DR: If we are interested in optimizing TimeVaryingDCMPlannerUnitTests planning time, it may be worth to try to tune some terminating condition, perhaps plotting the lagrangian error w.r.t. to the iteration time before.
**

traversaro, 2021-07-06:
@GiulioRomualdi while in the future we may investigate how to have a "fast" mumps also on conda, I think that the fact that coinbrew has now the same performance on conda and apt means that we can close the issue, what do you think? I also dropped a comment on how to install `netlib`-based blas in https://github.com/robotology/robotology-superbuild/issues/659 .

GiulioRomualdi, 2021-08-30:
This morning I had to install Ipopt from scratch I used CoinBrew to install MUMPS and IPOPT  (`apt-coinbrew-superbuild`)
Differently from the  original scenario https://github.com/dic-iit/bipedal-locomotion-framework/issues/219 we have

| Scenario | IPOPT version | MUMPS version |
|:----------:|:---------------:|:-------------------|
| Original https://github.com/dic-iit/bipedal-locomotion-framework/issues/219 | 3.13.4          | 4.10.0 |
| 30 August 2021 (Latest releases) | 3.14.2        |    5.4.0 Tue Apr 13 15:26:30 UTC 2021             |

Here the result
```
20: ******************************************************************************
20: This program contains Ipopt, a library for large-scale nonlinear optimization.
20:  Ipopt is released as open source code under the Eclipse Public License (EPL).
20:          For more information visit https://github.com/coin-or/Ipopt
20: ******************************************************************************
20: 
20:       solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
20:        nlp_f  | 918.00us ( 13.30us) 914.82us ( 13.26us)        69
20:        nlp_g  |   1.69ms ( 24.51us)   1.64ms ( 23.77us)        69
20:     nlp_grad  |  70.00us ( 70.00us)  70.57us ( 70.57us)         1
20:   nlp_grad_f  |   1.00ms ( 25.69us) 998.33us ( 25.60us)        39
20:   nlp_hess_l  |   1.88ms ( 52.31us)   1.88ms ( 52.21us)        36
20:    nlp_jac_g  |   1.87ms ( 46.80us)   1.88ms ( 46.95us)        40
20:        total  | 521.60ms (521.60ms) 521.60ms (521.60ms)         1
20: ===============================================================================
20: All tests passed (157 assertions in 1 test case)
20: 
1/1 Test #20: TimeVaryingDCMPlannerUnitTests ...   Passed    0.68 sec
```

traversaro, 2021-08-30:
A possible way to explore the fast open source combination for this problem is to use [spack](https://spack.io/). The main advantage of this package manager is that it compiles everything from source optimizing the code for the specific microarchitecture on the system, and you can specify several compile time options and even the version of dependency, so you can do:
~~~
spack install ipopt ^mumps@4.1.0
~~~

~~~
spack install ipopt ^mumps@5.4.0
~~~

to install two different version of ipopt with different dependencies. Unfortunatly I noticed just know that only 5.* version of mumps are supported in spack at the moment: https://spack.readthedocs.io/en/latest/package_list.html#mumps . See https://github.com/traversaro/robotology-superbuild-spack-experiments/ if you are curious on some other experiments.