gbionics/bipedal-locomotion-framework #246: Compilation failure when using master with robotology-superbuild

S-Dafarra, 2021-03-23:
I just pulled ``master`` and I have some trouble in compiling. First, the compilation stopped because we are requiring a particular version of ``manif``, then of ``OsqpEigen``. In general I think this sort of problems should occur at configure time, so I guess we should start requiring specific versions.

Once fixed that, I am still not able to compile and I get the following errors:
```
Scanning dependencies of target FloatingBaseEstimatorDevice
[ 65%] Building CXX object devices/FloatingBaseEstimatorDevice/CMakeFiles/FloatingBaseEstimatorDevice.dir/src/FloatingBaseEstimatorDevice.cpp.o
[ 66%] Linking CXX shared module ../../lib/yarp/FloatingBaseEstimatorDevice.so
[ 66%] Built target FloatingBaseEstimatorDevice
Scanning dependencies of target pybind11_blf
[ 67%] Building CXX object bindings/python/CMakeFiles/pybind11_blf.dir/bipedal_locomotion_framework.cpp.o
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp: In function ‘void pybind11_init_bindings(pybind11::module&)’:
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:42:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   42 |     py::module_ parametersHandlerModule = m.def_submodule("parameters_handler");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:43:47: error: ‘parametersHandlerModule’ was not declared in this scope
   43 |     bindings::ParametersHandler::CreateModule(parametersHandlerModule);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:45:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   45 |     py::module_ mathModule = m.def_submodule("math");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:46:34: error: ‘mathModule’ was not declared in this scope
   46 |     bindings::Math::CreateModule(mathModule);
      |                                  ^~~~~~~~~~
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:49:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   49 |     py::module_ contactsModule = m.def_submodule("contacts");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:50:38: error: ‘contactsModule’ was not declared in this scope
   50 |     bindings::Contacts::CreateModule(contactsModule);
      |                                      ^~~~~~~~~~~~~~
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:54:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   54 |     py::module_ plannersModule = m.def_submodule("planners");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:55:38: error: ‘plannersModule’ was not declared in this scope
   55 |     bindings::Planners::CreateModule(plannersModule);
      |                                      ^~~~~~~~~~~~~~
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:59:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   59 |     py::module_ robotInterfaceModule = m.def_submodule("robot_interface");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:60:44: error: ‘robotInterfaceModule’ was not declared in this scope
   60 |     bindings::RobotInterface::CreateModule(robotInterfaceModule);
      |                                            ^~~~~~~~~~~~~~~~~~~~
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:64:9: error: ‘module_’ is not a member of ‘py’; did you mean ‘module’?
   64 |     py::module_ floatingBaseEstimatorModule = m.def_submodule("floating_base_estimators");
      |         ^~~~~~~
      |         module
/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp:65:52: error: ‘floatingBaseEstimatorModule’ was not declared in this scope
   65 |     bindings::FloatingBaseEstimators::CreateModule(floatingBaseEstimatorModule);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [bindings/python/CMakeFiles/pybind11_blf.dir/build.make:63: bindings/python/CMakeFiles/pybind11_blf.dir/bipedal_locomotion_framework.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:1712: bindings/python/CMakeFiles/pybind11_blf.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/bipedal-locomotion-framework.dir/build.make:115: src/bipedal-locomotion-framework/CMakeFiles/YCMStamp/bipedal-locomotion-framework-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1967: CMakeFiles/bipedal-locomotion-framework.dir/all] Error 2
make: *** [Makefile:95: all] Error 2

```

I also deleted the build folder and let the ``robotology-superbuild`` to configure, but that still happened.

cc @GiulioRomualdi

GiulioRomualdi, 2021-03-25:
Hi @S-Dafarra you're right.

1. osqp-eigen I'm waiting for https://github.com/robotology/osqp-eigen/pull/86 so I release a new version and we can add the required version here
2. manif: unfortunately they started to use add the version only recently https://github.com/artivis/manif/pull/214

Regarding the error, I don't know in which way did you install pybind11. Probably @diegoferigo can help us

S-Dafarra, 2021-03-25:
> Regarding the error, I don't know. In which way did you install pybind11?. Probably @diegoferigo can help us

To be honest, I don't remember. What am I supposed to check?

diegoferigo, 2021-03-25:
As it is, the error is not telling me much. Maybe executing with `VERBOSE=1 make <target>` could give something more?

S-Dafarra, 2021-03-25:
> As it is, the error is not telling me much. Maybe executing with `VERBOSE=1 make <target>` could give something more?

Here:

<details>
<summary>Compilation output:</summary>

<p>

```
sdafarra@iiticublap104:~/Software/robotology-superbuild/build/src/bipedal-locomotion-framework (master)$ VERBOSE=1 make
/usr/bin/cmake -S/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework -B/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/CMakeFiles /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make -f src/TextLogging/CMakeFiles/TextLogging.dir/build.make src/TextLogging/CMakeFiles/TextLogging.dir/depend
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
cd /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/TextLogging /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/TextLogging/CMakeFiles/TextLogging.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make -f src/TextLogging/CMakeFiles/TextLogging.dir/build.make src/TextLogging/CMakeFiles/TextLogging.dir/build
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make[2]: Nothing to be done for 'src/TextLogging/CMakeFiles/TextLogging.dir/build'.
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
[  2%] Built target TextLogging
make -f src/ParametersHandler/CMakeFiles/ParametersHandler.dir/build.make src/ParametersHandler/CMakeFiles/ParametersHandler.dir/depend
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
cd /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/ParametersHandler /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/ParametersHandler/CMakeFiles/ParametersHandler.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make -f src/ParametersHandler/CMakeFiles/ParametersHandler.dir/build.make src/ParametersHandler/CMakeFiles/ParametersHandler.dir/build
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make[2]: Nothing to be done for 'src/ParametersHandler/CMakeFiles/ParametersHandler.dir/build'.
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
[  4%] Built target ParametersHandler
make -f src/YarpUtilities/CMakeFiles/YarpUtilities.dir/build.make src/YarpUtilities/CMakeFiles/YarpUtilities.dir/depend
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
cd /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/YarpUtilities /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/YarpUtilities /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/src/YarpUtilities/CMakeFiles/YarpUtilities.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make -f src/YarpUtilities/CMakeFiles/YarpUtilities.dir/build.make src/YarpUtilities/CMakeFiles/YarpUtilities.dir/build
make[2]: Entering directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make[2]: Nothing to be done for 'src/YarpUtilities/CMakeFiles/YarpUtilities.dir/build'.
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
[  7%] Built target YarpUtilities
[... cut ...]
make[2]: *** [bindings/python/CMakeFiles/pybind11_blf.dir/build.make:63: bindings/python/CMakeFiles/pybind11_blf.dir/bipedal_locomotion_framework.cpp.o] Error 1
make[2]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make[1]: *** [CMakeFiles/Makefile2:1712: bindings/python/CMakeFiles/pybind11_blf.dir/all] Error 2
make[1]: Leaving directory '/home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework'
make: *** [Makefile:130: all] Error 2
sdafarra@iiticublap104:~/Software/robotology-superbuild/build/src/bipedal-locomotion-framework (master)$ 

```

</p>

</details>

diegoferigo, 2021-03-25:
```
cd /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python && 
/usr/lib/ccache/c++  
-DSPDLOG_COMPILED_LIB -Dpybind11_blf_EXPORTS 
-I/usr/include/python2.7
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/ParametersHandler/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/Contacts/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/Planners/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/RobotInterface/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/Math/include -I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/FloatingBaseEstimators/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/Contacts/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/ContactModels/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/Planners/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/RobotInterface/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/RobotInterface/YarpImplementation/include
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/Math/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/Estimators/include 
-I/home/sdafarra/Software/robotology-superbuild/src/bipedal-locomotion-framework/src/Conversions/include 
-isystem /home/sdafarra/Software/robotology-superbuild/build/install/include 
-isystem /usr/include/eigen3  
-g -fPIC -fvisibility=hidden   -std=c++14 -std=gnu++17 
-o CMakeFiles/pybind11_blf.dir/bipedal_locomotion_framework.cpp.o 
-c /home/sdafarra/Software/robotology-superbuild/build/src/bipedal-locomotion-framework/bindings/python/bipedal_locomotion_framework.cpp
```

`-I/usr/include/python2.7` smells. pybind11 is header-only, I was expecting to find the include path of its files. Can you please tell me where the pybind11 headers are located in your system?

S-Dafarra, 2021-03-26:
In the CMake cache, there is the following:
 ```
 pybind11_DIR                     /usr/lib/cmake/pybind11                                                                                                                                                                                                                                                                
```

diegoferigo, 2021-03-26:
I'm walking in the dark, since CI and other systems seem ok, I'd tend to blame something wrong in your setup. The path you provided is for the CMake configuration of pybind. Since that is there, I expect to find the headers in `/usr/include/pybind`, likely is the package installed from apt.

If you have iDynTree working in the superbuild, maybe you can enable the unofficial pybind bindings and check if they compile fine. It's the only other package that uses pybind within the superbuild. This would confirm whether what you're experiencing is a blf-related error, or a system-wide error.

S-Dafarra, 2021-03-26:
With @GiulioRomualdi we debugged it and apparently the problem is that the version shipped with ``apt`` is too old (2.4.3, see https://repology.org/project/python:pybind11/versions). The ``module_`` has been introduced with the 2.6. See 
> py::module was renamed py::module_ to avoid issues with C++20 when used unqualified, but an alias py::module is provided for backward compatibility. #2489

in https://pybind11.readthedocs.io/en/stable/changelog.html.

So I suppose that we can use only the ``pip`` version. and force to use at least the 2.6 version from the CMake.

traversaro, 2021-03-27:
> So I suppose that we can use only the pip version. and force to use at least the 2.6 version from the CMake.

At least for robotology-superbuild/apt deployment, I would strongly prefer to avoid perturbing the global state of the system, beside installing new apt packages provided by system repos. Installing at the system level stuff with pip can influence other packages that instead may expect to use the pybind11 2.4.3 provided by apt, and I would strongly prefer to avoid so. If pybind11  is a private header-only dependency, could we think of using FetchContent to get it? However this may not work if also manif needs to find it. In that case, probably installing pybind11 via the robotology-superbuild is the best way to proceed.

S-Dafarra, 2021-03-27:
It seems that most of the problems are related to the use of `module_`. Is it fundamental to use it instead of `module` (without the backlash)?

GiulioRomualdi, 2021-03-27:
Since we are not supporting c++20 we can use `module`
No problem from my side.

GiulioRomualdi, 2021-03-27:
Now we can install pybind11 from apt (Ubuntu 20.04) 
I think we can close this issue

closed 2021-03-27 by GiulioRomualdi