gbionics/bipedal-locomotion-framework #259: Unable to load the repo in a separate project

GiulioRomualdi, 2021-03-29:
I'm trying to use blf in a separate project, however, the  

CMake fails will the following error
```
Make Error at CMakeLists.txt:98 (include):
  include could not find load file:

    AddUninstallTarget

-- Configuring incomplete, errors occurred!
```

If I don't add `find_package(BipedalLocomotionFramework 0.1.0 REQUIRED)` everiting works fine. 

I think that the problem is related to this 

https://github.com/dic-iit/bipedal-locomotion-framework/blob/4e82be2f098aa836a6dce9feb9a6c6ad5ee0f876/CMakeLists.txt#L112

However if I remove that line in blf I got
```
By not providing "Findcppad.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "cppad", but
  CMake did not find one.

  Could not find a package configuration file provided by "cppad" with any of
  the following names:

    cppadConfig.cmake
    cppad-config.cmake

  Add the installation prefix of "cppad" to CMAKE_PREFIX_PATH or set
  "cppad_DIR" to a directory containing one of the above files.  If "cppad"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/BipedalLocomotionFramework/BipedalLocomotionFrameworkConfig.cmake:27 (find_dependency)
  CMakeLists.txt:78 (find_package)
```

cc @traversaro and @S-Dafarra

S-Dafarra, 2021-03-29:
Just to understand, is this happening after https://github.com/dic-iit/bipedal-locomotion-framework/pull/253?

GiulioRomualdi, 2021-03-29:
I don't know, it's the first time I import blf in a separate project

S-Dafarra, 2021-03-29:
How is the ``BipedalLocomotionFrameworkConfig.cmake`` file in your install folder?

GiulioRomualdi, 2021-03-29:
```cmake
set(BipedalLocomotionFramework_VERSION 0.1.0)

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was BipedalLocomotionFrameworkConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)

macro(set_and_check _var _file)
  set(${_var} "${_file}")
  if(NOT EXISTS "${_file}")
    message(FATAL_ERROR "File or directory ${_file} referenced by variable ${_var} does not exist !")
  endif()
endmacro()

####################################################################################

#### Expanded from @PACKAGE_DEPENDENCIES@ by install_basic_package_files() ####

include(CMakeFindDependencyMacro)
set(CMAKE_MODULE_PATH_BK ${CMAKE_MODULE_PATH})
set(CMAKE_MODULE_PATH ${PACKAGE_PREFIX_DIR}/share/BipedalLocomotionFramework/cmake)
find_package(iDynTree 3.0.0)
find_package(Eigen3 3.2.92)
find_package(spdlog)
find_package(YARP)
find_package(Qhull 8.0.0)
find_package(cppad)
find_package(manif 0.0.3)
find_package(matioCpp)
find_package(LieGroupControllers)
find_package(OpenCV)
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH_BK})

###############################################################################

if(NOT TARGET BipedalLocomotion::ContinuousDynamicalSystem)
  include("${CMAKE_CURRENT_LIST_DIR}/BipedalLocomotionFrameworkTargets.cmake")
endif()

# Compatibility
get_property(BipedalLocomotionFramework_ContinuousDynamicalSystem_INCLUDE_DIR TARGET BipedalLocomotion::ContinuousDynamicalSystem PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_TextLogging_INCLUDE_DIR TARGET BipedalLocomotion::TextLogging PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_GenericContainer_INCLUDE_DIR TARGET BipedalLocomotion::GenericContainer PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_YarpUtilities_INCLUDE_DIR TARGET BipedalLocomotion::YarpUtilities PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ParametersHandlerYarpImplementation_INCLUDE_DIR TARGET BipedalLocomotion::ParametersHandlerYarpImplementation PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ParametersHandler_INCLUDE_DIR TARGET BipedalLocomotion::ParametersHandler PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_CommonConversions_INCLUDE_DIR TARGET BipedalLocomotion::CommonConversions PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ManifConversions_INCLUDE_DIR TARGET BipedalLocomotion::ManifConversions PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_matioCppConversions_INCLUDE_DIR TARGET BipedalLocomotion::matioCppConversions PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_Contacts_INCLUDE_DIR TARGET BipedalLocomotion::Contacts PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ContactDetectors_INCLUDE_DIR TARGET BipedalLocomotion::ContactDetectors PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_Estimators_INCLUDE_DIR TARGET BipedalLocomotion::Estimators PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_FloatingBaseEstimators_INCLUDE_DIR TARGET BipedalLocomotion::FloatingBaseEstimators PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_System_INCLUDE_DIR TARGET BipedalLocomotion::System PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_Planners_INCLUDE_DIR TARGET BipedalLocomotion::Planners PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ContactModels_INCLUDE_DIR TARGET BipedalLocomotion::ContactModels PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_AutoDiffCppAD_INCLUDE_DIR TARGET BipedalLocomotion::AutoDiffCppAD PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_RobotInterfaceYarpImplementation_INCLUDE_DIR TARGET BipedalLocomotion::RobotInterfaceYarpImplementation PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_RobotInterface_INCLUDE_DIR TARGET BipedalLocomotion::RobotInterface PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_Math_INCLUDE_DIR TARGET BipedalLocomotion::Math PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_TSID_INCLUDE_DIR TARGET BipedalLocomotion::TSID PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_IK_INCLUDE_DIR TARGET BipedalLocomotion::IK PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_ReducedModelControllers_INCLUDE_DIR TARGET BipedalLocomotion::ReducedModelControllers PROPERTY INTERFACE_INCLUDE_DIRECTORIES)
get_property(BipedalLocomotionFramework_Framework_INCLUDE_DIR TARGET BipedalLocomotion::Framework PROPERTY INTERFACE_INCLUDE_DIRECTORIES)

set(BipedalLocomotionFramework_LIBRARIES BipedalLocomotion::ContinuousDynamicalSystem BipedalLocomotion::TextLogging BipedalLocomotion::GenericContainer BipedalLocomotion::YarpUtilities BipedalLocomotion::ParametersHandlerYarpImplementation BipedalLocomotion::ParametersHandler BipedalLocomotion::CommonConversions BipedalLocomotion::ManifConversions BipedalLocomotion::matioCppConversions BipedalLocomotion::Contacts  [... cut ...] _AutoDiffCppAD_INCLUDE_DIR}" "${BipedalLocomotionFramework_RobotInterfaceYarpImplementation_INCLUDE_DIR}" "${BipedalLocomotionFramework_RobotInterface_INCLUDE_DIR}" "${BipedalLocomotionFramework_Math_INCLUDE_DIR}" "${BipedalLocomotionFramework_TSID_INCLUDE_DIR}" "${BipedalLocomotionFramework_IK_INCLUDE_DIR}" "${BipedalLocomotionFramework_ReducedModelControllers_INCLUDE_DIR}" "${BipedalLocomotionFramework_Framework_INCLUDE_DIR}")
list(REMOVE_DUPLICATES BipedalLocomotionFramework_INCLUDE_DIRS)
```

GiulioRomualdi, 2021-03-29:
Here the repo where I'm trying to use the code https://github.com/GiulioRomualdi/centroidal-mpc-walking/commit/cf289ca269ae285fecc5528abf9d7c96674fb3e3

S-Dafarra, 2021-03-29:
Ok the important part is 
```cmake
set(CMAKE_MODULE_PATH_BK ${CMAKE_MODULE_PATH})
set(CMAKE_MODULE_PATH ${PACKAGE_PREFIX_DIR}/share/BipedalLocomotionFramework/cmake)
find_package(iDynTree 3.0.0)
find_package(Eigen3 3.2.92)
find_package(spdlog)
find_package(YARP)
find_package(Qhull 8.0.0)
find_package(cppad)
find_package(manif 0.0.3)
find_package(matioCpp)
find_package(LieGroupControllers)
find_package(OpenCV)
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH_BK})
```
Maybe you can add some prints on the ``CMAKE_MODULE_PATH`` before and after the use of ``find_package(BipedalLocomotionFramework 0.1.0 REQUIRED)`` :thinking:

S-Dafarra, 2021-03-29:
> Here the repo where I'm trying to use the code [GiulioRomualdi/centroidal-mpc-walking@cf289ca](https://github.com/GiulioRomualdi/centroidal-mpc-walking/commit/cf289ca269ae285fecc5528abf9d7c96674fb3e3)

I managed to compile it with no problem. Maybe it is finding an older version of ``blf``?

GiulioRomualdi, 2021-03-29:
GiulioRomualdi/centroidal-mpc-walking@cf289ca is working fine however if I apply the following patch the compilation fails
``` patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index d1ded69..329a63c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,9 +7,6 @@ cmake_minimum_required(VERSION 3.14)
 project(CentroidalMPCWalking
   VERSION 0.0.1)
 
-find_package(Eigen3 3.2.92 REQUIRED)
-find_package(BipedalLocomotionFramework 0.1.0 REQUIRED)
-
 
 list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake)
 
@@ -68,6 +65,10 @@ add_install_rpath_support(BIN_DIRS "${CMAKE_INSTALL_PREFIX}/${CMAKE_INSTALL_BIND
   USE_LINK_PATH)
 
 
+find_package(Eigen3 3.2.92 REQUIRED)
+find_package(BipedalLocomotionFramework 0.1.0 REQUIRED)
+
+
 #Function to automatize the process of creating a new library
 include(AddCentroidalMPCWalkingLibrary)
 ```

S-Dafarra, 2021-03-29:
> ```diff
> find_package(Eigen3 3.2.92 REQUIRED)
> +find_package(BipedalLocomotionFramework 0.1.0 REQUIRED)
> ```

The same here. I am on it

S-Dafarra, 2021-03-29:
Ok, I got it. It is because I used the name ``CMAKE_MODULE_PATH_BK`` for both ``blf`` and ``matio-cpp``. I will open a PR soon.

S-Dafarra, 2021-03-29:
Related PR #262

closed 2021-03-30 by GiulioRomualdi