gbionics/bipedal-locomotion-framework #260: Discuss the desiderata of the block runner

GiulioRomualdi, 2021-03-29:
This follows #256 

The block runner is a class that runs a block in a periodic thread. It is useful if you want to run multiple blocks in the same application. 

Here I want to discuss possible desiderata to the block runner.

- [ ] FInd a way to start and stop the thread. Right now in the implementation that I proposed #256  I used atomic_bool to stop the thread. A possible solution is to use signals and slots 

cc @traversaro @S-Dafarra

S-Dafarra, 2021-03-29:
> A possible solution is to use signals and slots

I personally like the signal/slot paradigm. There was some discussion going on in https://github.com/dic-iit/bipedal-locomotion-framework/issues/44 for this. I would just avoid implementing our own signal/slot library as per https://github.com/dic-iit/bipedal-locomotion-framework/issues/44#issuecomment-633670339

I remember it was very easy to run a block in a separate thread, just a matter of a function call.

GiulioRomualdi, 2021-11-15:
https://github.com/ami-iit/bipedal-locomotion-framework/pull/272 closes the issue (for now)

closed 2021-11-15 by GiulioRomualdi