gbionics/bipedal-locomotion-framework #261: Port the matlab-whole-body-simulator in the repo

GiulioRomualdi, 2021-03-29:
In this issue, we can discuss the points required to port the `matlab-whole-body-simulator` in blf. https://github.com/dic-iit/matlab-whole-body-simulator

Blf has already the machinery to compute the dynamics of the system and integrate it. Discussing with @Giulero implementing the `whole-body-simulator` he developed in C++ should not be so complex.

Before implementing the dynamics we need
- [ ] Add the possibility to reset the integrator transparently to the user. The dynamics designed in the `matlab-whole-body-simulator` is hybrid and so we should reset the integrator when an impact with the environment occurs. 

cc @traversaro @S-Dafarra

traversaro, 2021-03-29:
The tricky aspect here is that to integrate an hybrid system with a variable step integrator you need some kind of hybrid integrator, as the so-called zero-crossing detection is part of logic that the integrator uses to determine the simulation step. If we want to go down this path an alternative to roll our our own integrator is to use an existing integrator for this, such as the one provided by [Sundials](https://github.com/LLNL/sundials) (see https://github.com/RobotLocomotion/drake/issues/1306#issuecomment-173923818 for a related comment). 

A simple example of integrating an hybrid system (in particular, a bouncing ball) with Sundials can be found in https://github.com/CATIA-Systems/FMPy/blob/e1927a128a9552f20304587c54c716750e73b05f/tests/test_cvode.py#L12 (it is in Python, but it uses Sundials bindings that match 1:1 the C version).

Another possible source of integrators (or even of inspiration if we want to roll out our own integrators, given that its code is rather commented and compressible even if it is written in C++) is Drake: https://drake.mit.edu/doxygen_cxx/group__integrators.html .

traversaro, 2021-03-29:
Sorry, I did not mentioned that for a first implementation we can avoid to do zero crossing detection, and just reset the integrator whenever the witness function/even trigger passes through zero.

nunoguedelha, 2021-04-09:
>     * The dynamics designed in the `matlab-whole-body-simulator` is hybrid and so we should reset the integrator when an impact with the environment occurs.

Hi guys, quick question: what do you mean exactly by resetting the integrator?

Just for clarifying, here's what we're doing in the `matlab-whole-body-simulator` simulator:
- We are using the MATLAB ode45 integration method, integrating the velocity $\dot{s}(t)$ to obtain the updated position $s(t+dt)$, and integrating the acceleration $\ddot{s}(t)$ to obtain the updated velocity $\dot{s}(t+dt)$.
- On the next iteration, if an impact is detected (contact vertex z position < 0), we correct the joints velocities such that the linear velocity of that vertex is 0 (this should be improved, only the vertical component of the velocity should be 0 in case of 0% restitution). After that, the integration step is done with the corrected joints velocities, and thus the vertex height stays unchanged. So this velocity correction is a reset in some way, but there is like a tolerance: although the vertex motion is stopped (no further sinking on the ground), the position is not corrected (keeps its small negative value).

this shouldn't be a problem, but if it is in some context of the BLF, one possible solution would be to reset the joint positions to the previous values.

Side note: using a zero crossing detection approach here wouldn't be trivial since the violated limit does not concern the integrated variable but rather the output of the forward kinematics.

GiulioRomualdi, 2021-04-09:
> Hi guys, quick question: what do you mean exactly by resetting the integrator?

Exactly this. 

> On the next iteration, if an impact is detected (contact vertex z position < 0), we correct the joints velocities such that the linear velocity of that vertex is 0 (this should be improved, only the [...]

traversaro, 2023-12-21:
> > [ ] Add the possibility to reset the integrator transparently to the user. The dynamics designed in the `matlab-whole-body-simulator` is hybrid and so we should reset the integrator when an impact with the environment occurs.
> 
> The tricky aspect here is that to integrate an hybrid system with a variable step integrator you need some kind of hybrid integrator, as the so-called zero-crossing detection is part of logic that the integrator uses to determine the simulation step. If we want to go down this path an alternative to roll our our own integrator is to use an existing integrator for this, such as the one provided by [Sundials](https://github.com/LLNL/sundials) (see [RobotLocomotion/drake#1306 (comment)](https://github.com/RobotLocomotion/drake/issues/1306#issuecomment-173923818) for a related comment).
> 
> A simple example of integrating an hybrid system (in particular, a bouncing ball) with Sundials can be found in https://github.com/CATIA-Systems/FMPy/blob/e1927a128a9552f20304587c54c716750e73b05f/tests/test_cvode.py#L12 (it is in Python, but it uses Sundials bindings that match 1:1 the C version).
> 
> Another possible source of integrators (or even of inspiration if we want to roll out our own integrators, given that its code is rather commented and compressible even if it is written in C++) is Drake: https://drake.mit.edu/doxygen_cxx/group__integrators.html .

fyi @S-Dafarra @diegoferigo @LoreMoretti @Gio-DS @xela-95 related to our discussion of today.

S-Dafarra, 2023-12-22:
> fyi @S-Dafarra @diegoferigo @LoreMoretti @Gio-DS @xela-95 related to our discussion of today.

I guess that it had to do with https://sundials.readthedocs.io/en/latest/cvode/Usage/index.html#rootfinding-initialization-function. It seems that you can poll whether a root has been found and change the state accordingly: https://github.com/CATIA-Systems/FMPy/blob/e1927a128a9552f20304587c54c716750e73b05f/tests/test_cvode.py#L80-L86
I wonder whether it automatically trims the step size such that the event occurs exactly at the end of an integration step.

S-Dafarra, 2023-12-22:
> I wonder whether it automatically trims the step size such that the event occurs exactly at the end of an integration step.

Most likely, yes. Reading https://computing.llnl.gov/projects/sundials/usage-notes

> To stop when the location of the discontinuity is determined by the solution (implicitly-defined discontinuity), use the rootfinding feature (available both in CVODE/CVODES and IDA/IDAS). In other words, the location of the discontinuity is the zero of some function g of the solution

Looking back at the code, it seems that it uses a different return flag: https://github.com/CATIA-Systems/FMPy/blob/e1927a128a9552f20304587c54c716750e73b05f/tests/test_cvode.py#L74-L76. Also, ``tret`` is an output, so I guess it outputs the time at which the discontinuity occurs.