gbionics/bipedal-locomotion-framework #283: Investigate the possibility to call blf methods in jupyter

GiulioRomualdi, 2021-04-21:
Discussing with @traversaro I realize that it is possible to call c++ function in jupyter using https://github.com/jupyter-xeus/xeus-cling 

I installed `xeus-cling` and  `bipedal` from conda and I was able to easily use one of the components of `blf`. 

[image]

Notice that the following lines are required to load a third-party libraries 
```cpp
#pragma cling add_include_path("/usr/include/eigen3/")
#pragma cling add_include_path("/home/gromualdi/miniforge3/envs/cling/include")
#pragma cling add_library_path("/home/gromualdi/miniforge3/envs/cling/lib")
#pragma cling load("libBipedalLocomotionFrameworkMath")
```

In theory, we should be able to remove the hardcoded path

Related issue: https://github.com/jupyter-xeus/xeus-cling/issues/393#issuecomment-824130603

cc @diegoferigo

diegoferigo, 2021-04-21:
Yes, I tried `xeus-cling` a couple of years ago and it's very cool, I remember reading an interesting technical article explaining the challenges in calling C++ interactively, but I cannot find it. TL;DR there are a lot of tricks in LLVM with namespaces for overriding variables.

GiulioRomualdi, 2024-04-08:
I'm going to close this issue for the time being, if it will be required in the future we can open a new one or just reopen this