gbionics/bipedal-locomotion-framework #285: Discuss the drawback of using `const IParametersHandler` in the Advanceable class

GiulioRomualdi, 2021-04-26:
In #267 I added
```cpp
bool initialize(std::shared_ptr<const BipedalLocomotion::ParametersHandler::IParametersHandler>)
```
method.

The signature of the method is in theory fine since a block should not change the handler in the initialization phase, however, in practice, this creates several issues.

For instance, in [`centroidal-mpc-walking`](https://github.com/GiulioRomualdi/centroidal-mpc-walking/blob/bb0f42b822f3afc41d5dec12e2e5b9bfd6bad633/src/centroidal-mpc-walking/src/WholeBodyQPBlock.cpp#L268) I tried to instantiate the `LeggedOdometry` however I'm not able to do it because the` LeggedOdometry::initialize()` wants a `non-const` `IParameterHandler`.
https://github.com/dic-iit/bipedal-locomotion-framework/blob/4ad31d8d62ffa0e3da5b7fbef77a945124c7bb4e/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseEstimator.h#L160-L161

Here two possible solutions can be implemented:
1. Change the signature of 
    https://github.com/dic-iit/bipedal-locomotion-framework/blob/4ad31d8d62ffa0e3da5b7fbef77a945124c7bb4e/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/FloatingBaseEstimator.h#L160-L161
   Into `bool initialize(std::weak_ptr<const BipedalLocomotion::ParametersHandler::IParametersHandler> handler, std::shared_ptr<iDynTree::KinDynComputations> kindyn); ` 
2. Implement the copy constructor/copy assignment operator for `IParameterHandler`.

This first solution is temporary since we may need to expand the content of a ParameterHandler and if the object is `const` this is not possible. 
On the other hand, I don't know how to implement the second solution. Since there is no way to get the type of a stored parameter is almost impossible to copy the content of a parameter handler into another one. (e.g. I don't know how to copy the content of a `YarpParameterHandler` in a `StdParameterHandler` and vice-versa)

cc @prashanthr05 @S-Dafarra and @traversaro

S-Dafarra, 2021-04-26:
I would go for the first solution. I would prefer if the user of the parameters handler does not modify anything.

GiulioRomualdi, 2021-04-26:
> I would go for the first solution. I would prefer if the user of the parameters handler does not modify anything.

I agree with that however we may need to change the content of the `handler` in the future. For instance, several classes require the `sampling_time` and the user may specify it only once. e.g.
```ini
sampling_time 0.01

[IK]
some_useful_parameter "parameter"
```

if the `ik` requires the sampling time in c++ we need to expand the parameter handler used in the ik with `sampling_time`. This is currently not possible 
```cpp
void initilizeIK(std::shared_ptr<const IParameterHandler> handler, double samplingTime)
{
    auto newHandler = std::make_shared<StdParameterHandler>();
    newHandler->set(handler); //<---------------- This is not possible
    newHandler->setParameter("sampling_time", samplingTime);
   ...........
}

S-Dafarra, 2021-04-26:
Ok, we need a method to clone then :thinking: I think that is doable for both the implementations we have at the moment

GiulioRomualdi, 2021-04-26:
You are right!

GiulioRomualdi, 2021-04-27:
https://github.com/dic-iit/bipedal-locomotion-framework/pull/288 implements what @S-Dafarra suggested in https://github.com/dic-iit/bipedal-locomotion-framework/issues/285#issuecomment-826923766

Before closing the issue we should address point 1 of https://github.com/dic-iit/bipedal-locomotion-framework/issues/285#issue-867728347