gbionics/bipedal-locomotion-framework #287: Unable to use the leggedodometry class

GiulioRomualdi, 2021-04-26:
When I try to use the legged odometry I got the following meaningless results
```
[ERROR] Model :: getFrameLink : frameIndex -1 is not valid, should be between 0 and 269
[ERROR] Model :: getFrameTransform : frameIndex -1 is not valid, should be between 0 and 269
[LeggedOdometry::updateKinematics] Initialized LeggedOdometry.
base --------------->     -nan      -nan      -nan 0.0477658  0.391864  0.918559 0.0202494
[2021-04-26 22:21:41.472] [thread: 92728] [blf] [info] COM-------------> -nan -nan -nan
[2021-04-26 22:21:41.472] [thread: 92728] [blf] [info] l_sole-------------> 0.978525 -0.00038232 0.206127
-0.148468 0.692382 0.706091
-0.142989 -0.721531 0.677456
 -nan -nan -nan

[2021-04-26 22:21:41.472] [thread: 92728] [blf] [info] r_sole-------------> 0.978518 -5.98931e-05 0.20616
-0.148805 0.691904 0.706488
-0.142686 -0.721989 0.677032
 -nan -nan -nan
```

The legged odometry is used as follows
```cpp
    if (!m_floatingBaseEstimator.initialize(handler->getGroup("FLOATING_BASE_ESTIMATOR"),
                                            kinDyn.kindyn))
    {
        BipedalLocomotion::log()->error("{} Unable to initialize the legged odometry.", logPrefix);
        return false;
    }

    m_floatingBaseEstimator.changeFixedFrame(contactListMap["left_foot"].defaultIndex(),
                                             contactListMap["left_foot"].firstContact()->pose.quat(),
                                             contactListMap["left_foot"].firstContact()->pose.translation());

    m_floatingBaseEstimator.setKinematics(m_currentJointPos, m_currentJointVel);
    m_floatingBaseEstimator.advance();

    m_baseTransform = m_floatingBaseEstimator.getOutput().basePose;
    m_baseVelocity = m_floatingBaseEstimator.getOutput().baseTwist;
```

The `ini` file contains
```ini
sampling_period_in_s      0.001

[ModelInfo]
base_link                 "root_link"
base_link_imu             "root_link"
left_foot_contact_frame   "l_sole"
right_foot_contact_frame  "r_sole"

[LeggedOdom]
initial_fixed_frame       "l_sole"
switching_pattern         "useExternal"
```
cc @prashanthr05

GiulioRomualdi, 2021-04-26:
I try to print the content of `refFrame_H_fixedFrame`
https://github.com/dic-iit/bipedal-locomotion-framework/blob/4ad31d8d62ffa0e3da5b7fbef77a945124c7bb4e/src/Estimators/src/LeggedOdometry.cpp#L456

and this is the output

```
refFrame_H_fixedFrame    0.978525 -0.00038232    0.206127        -inf
  -0.148468    0.692382    0.706091        -inf
  -0.142989   -0.721531    0.677456        -inf
          0           0           0           1
```

GiulioRomualdi, 2021-04-26:
I probably found the problem :) 

Given the configuration file I've the code enters in this `if` branch.

https://github.com/dic-iit/bipedal-locomotion-framework/blob/4ad31d8d62ffa0e3da5b7fbef77a945124c7bb4e/src/Estimators/src/LeggedOdometry.cpp#L199-L204

Here `m_pimpl->m_initialRefFrameForWorld = m_pimpl->m_initialFixedFrame;` is called however the index `m_initialRefFrameForWorldIdx` is not updated.

I think 
```cpp
m_pimpl->m_initialRefFrameForWorldIdx = m_modelComp.kinDyn()->model().getFrameIndex(m_pimpl->m_initialRefFrameForWorld);
```
should be added

prashanthr05, 2021-04-27:
Indeed, you are right. That's the first thing which is done in the `else` block, while it's missing to be initialized to the value of `m_initialFixedFrameIdx`. Just the name is being copied.

I had never tried using the class without setting the `initial_ref_frame_for_world`, that's why I would have missed it in my use-case applications.

PR #289 (2021-04-27): [FloatingBaseEstimator] Bug fix in  initialization LeggedOdometry
The frame index for world reference frame was not initialized to a proper value when it is not specified in the parameter configuration.

This PR fixes #287.
closed 2021-04-27 by GiulioRomualdi