gbionics/bipedal-locomotion-framework #290: Understand how to use leggedodometry with the knowledge of foot position

GiulioRomualdi, 2021-04-27:
While using the legged odometry I noticed a strange behavior.

I initialized the estimator with the following paramateres
```ini
sampling_period_in_s            0.001

[ModelInfo]
base_link                       "root_link"
base_link_imu                   "root_link"
left_foot_contact_frame         "l_foot"
right_foot_contact_frame        "r_foot"

[LeggedOdom]
initial_fixed_frame             "l_sole"
switching_pattern               "external"
```
Then I tried to use it as follows

```cpp
    m_floatingBaseEstimator.changeFixedFrame(contactListMap["left_foot"].defaultIndex(),
                                             contactListMap["left_foot"].firstContact()->pose.quat(),
                                             contactListMap["left_foot"].firstContact()->pose.translation());
    m_floatingBaseEstimator.setKinematics(m_currentJointPos, m_currentJointVel);
    m_floatingBaseEstimator.advance();

    m_baseTransform = m_floatingBaseEstimator.getOutput().basePose;
    m_baseVelocity = m_floatingBaseEstimator.getOutput().baseTwist;
```

Independently on the content of `contactListMap["left_foot"].firstContact()->pose`, 
`m_baseTransform` is always equal to  
```
base pose   0.0234848  -0.0798684    0.601393   -0.051003 0.000236203    0.998698 0.000178679
```

For instance (notice the different value stored in `left foot pose`). The pose vector contains `[x y z qx qy qz qw]`
```
Index 53 l_sole
left foot pose 0 7 0 0 0 0 1 
base pose   0.0234848  -0.0798684    0.601393   -0.051003 0.000236203    0.998698 0.000178679
```
or 
```
Index 53 l_sole
left foot pose 0 0.07 0 0 0 0 1
base pose   0.0234848  -0.0798684    0.601393   -0.051003 0.000236203    0.998698 0.000178679
```

Should I initialize the estimator in some way? I'm expecting that `m_floatingBaseEstimator.changeFixedFrame()` is used to translate/rotate the base.

cc @prashanthr05

prashanthr05, 2021-04-27:
Try to call the `changeFixedFrame()` after calling advance(). Since, within the `advance()` as a first step, we set the robot state for the kindyncomputations, The `changeFixedFrame()`  computes relative transforms which depends on the robot state to be set.

GiulioRomualdi, 2021-04-27:
I tried 
```cpp
    m_floatingBaseEstimator.setKinematics(m_currentJointPos, m_currentJointVel);
    m_floatingBaseEstimator.advance();

    m_floatingBaseEstimator.changeFixedFrame(contactListMap["left_foot"].defaultIndex(),
                                             contactListMap["left_foot"].firstContact()->pose.quat(),
                                             contactListMap["left_foot"].firstContact()->pose.translation());
```

but nothing changed.

Also this is not working 
```cpp
    m_floatingBaseEstimator.setKinematics(m_currentJointPos, m_currentJointVel);
    m_floatingBaseEstimator.advance();

    m_floatingBaseEstimator.changeFixedFrame(contactListMap["left_foot"].defaultIndex(),
                                             contactListMap["left_foot"].firstContact()->pose.quat(),
                                             contactListMap["left_foot"].firstContact()->pose.translation());

    m_floatingBaseEstimator.advance();
```

GiulioRomualdi, 2021-04-27:
I printed the content of the following variables

https://github.com/dic-iit/bipedal-locomotion-framework/blob/665e04d5af781f4df0c23e96c7c21444cffed6d3/src/Estimators/src/LeggedOdometry.cpp#L482-L485

```
m_world_H_fixedFrame     0 -0.07     0     0     0     0     1
fixedFrame_H_imu   0.0234823  -0.0799371     0.60138  -0.0509665 9.69215e-05      0.9987 0.000130399
world_H_imu   0.0234823   -0.149937     0.60138  -0.0509665 9.69215e-05      0.9987 0.000130399
```

Do you think are fine?

prashanthr05, 2021-04-27:
Just noticed the parameter value you were passing for `switching_pattern`.
It must be `external` -> `useExternal`. Looks like, since this is wrong, it considers the default value `latest` which chooses the latest contact from the list of contacts. However, in this case we are not filling the contact measurements using `m_floatingBaseEstimator.setContactStatus()` method, so there are not contacts and thus, the base pose doesn't seem to be updated at all.

GiulioRomualdi, 2021-05-07:
[Here](https://github.com/GiulioRomualdi/centroidal-mpc-walking/blob/c1b72da9a807adaa9b04fe38915df660b09b5c25/src/centroidal-mpc-walking/src/WholeBodyQPBlock.cpp#L269-L316) you can find a working receipt to use legged odometry plus FixedFootDetector class

closed 2021-05-07 by GiulioRomualdi