gbionics/bipedal-locomotion-framework #293: Move `KinDynComputationsDescriptor` from Estimators component into a different component

GiulioRomualdi, 2021-04-29:
`KinDynComputationsDescriptor` it's really useful to automatically build a kindyn object. I think we may move it to a different component.

https://github.com/dic-iit/bipedal-locomotion-framework/blob/47578647d11aabf15800d9c662b12d561db19eef/src/Estimators/include/BipedalLocomotion/FloatingBaseEstimators/ModelComputationsHelper.h#L23-L27

What do you suggest? @prashanthr05 @S-Dafarra @traversaro @paolo-viceconte

S-Dafarra, 2021-04-30:
What did you have in mind?

GiulioRomualdi, 2021-05-05:
> What did you have in mind?

Nothing special

prashanthr05, 2021-05-05:
We could create a Model related component. And move it there?
For example, I am adding a few Model related classes (in a private [appendix](https://github.com/dic-iit/element_floating-base-estimation/tree/feature/KinDynFusion/code/KinDynFusion/src/model/include/KinDynFusion/Model) of BLF).

S-Dafarra, 2021-05-05:
> > What did you have in mind?
> 
> Nothing special

I meant what did you have in mind :grin: Why you think it is better to move it and what would be the desiderata?

GiulioRomualdi, 2021-11-05:
At the current stage if you want to load a KinDynConputations object in `blf` you should use the `floating_base_estimators` submodule

```python
import bipedal_locomotion_framework as blf                              
kindyn = blf.floating_base_estimators.KinDynComputations()
```

However, the `FloatingBaseEstimator` library is compiled only if 

https://github.com/ami-iit/bipedal-locomotion-framework/blob/d793ac71d80ace4fc3b793db94707f7d6f8113ea/cmake/BipedalLocomotionFrameworkDependencies.cmake#L150-L152

Furthermore, the `KinDyn` object can be used also outside the submodule.

I would suggest moving it to a new submodule called `idyntree`, the submodule will contain all the future bindings related to iDynTree

In this case, it will be possible to instantiate a kindyn object with 

```python
import bipedal_locomotion_framework as blf                              
kindyn = blf.idyntree.KinDynComputations()
```

what do you think? @ami-iit/blf-developers

traversaro, 2021-11-05:
Good idea!

diegoferigo, 2021-11-05:
+1 on the submodule idea!

closed 2023-10-30 by GiulioRomualdi