gbionics/bipedal-locomotion-framework #298: Rethink the way to store frame informations in the `Contact` class

GiulioRomualdi, 2021-05-06:
This issue is the consequence of https://github.com/dic-iit/bipedal-locomotion-framework/pull/297#issuecomment-833471785

In details as @S-Dafarra suggested that having an int to indicate the `index` in the contact class may not be the best choice
https://github.com/dic-iit/bipedal-locomotion-framework/blob/005684637a8aa1f360c3aee8edab3b26edf7a654/src/Contacts/include/BipedalLocomotion/Contacts/Contact.h#L53-L56

Indeed:
1. the fact that a frame is associated with an `int` is strictly related to `iDynTree`. 
2. The point of contact may not be associated with the frame of the robot.

Discussing with @S-Dafarra and @prashanthr05 we came out with a possible solution.
Substitute the `int` with a struct that specifies the frame. For example
```cpp
struct Frame // <---------- I think that frame is not the right name of the struct
{
    std::string frameName;
    manif::SE3d contact_H_frame;
    std::string model;
}
```

We should understand if using a `std::string` to store the frame name  causes some efficiency drop when used with `iDynTree::KinDynComputations`
Indeed accordingly to the [iDynTree documentation](https://robotology.github.io/idyntree/master/classiDynTree_1_1KinDynComputations.html#a50f02bed7aa3c4b4f561d073738f0e15)

[image]

traversaro, 2021-05-06:
I think you can check (i.e. benchmark) if frame idx vs string is actually relevant in your application, consider that that code was written when C++11 was not widespread, and the default string implementation in GCC was not using the so-called Small String Optimization ( https://blogs.msmvps.com/gdicanio/2016/11/17/the-small-string-optimization/ ) . Switching to SSO std::string was the last great ABI break in C++ GCC implementation. If you need to improve the performance of string used methods, you can can try to change the implementation from linear search to a more efficient method (see https://github.com/robotology/idyntree/blob/db312bfe9f490f991250603f05cea84a8c156899/src/model/src/Model.cpp#L574).

S-Dafarra, 2021-05-06:
I think that it is up to the application to maintain a proper mapping between the name and the index, not of the interface.