gbionics/bipedal-locomotion-framework #299: Discuss the possibility to drop doygen and move to m.css

GiulioRomualdi, 2021-05-06:
In these last days, I played a bit with [m.css](https://mcss.mosra.cz/) and I was able to obtain some nice (in my opinion) website containing the documentation. 

Here some examples:
- https://robotology.github.io/osqp-eigen/
-  https://giulioromualdi.github.io/scs-eigen/

Do you have any opinion on moving from doxygen website to m.css? 

cc @S-Dafarra @prashanthr05 @traversaro

traversaro, 2021-05-06:
I wanted to share this moment with @jeljaik .

GiulioRomualdi, 2021-05-07:
Here some examples

This is the readme. (Probably we should remove the image and handle the badges in a different way) 
[image]

This is an example of the class documentation
[image]

This is the search engine 
[image]

S-Dafarra, 2021-05-07:
:heart_eyes:

closed 2021-05-08 by GiulioRomualdi

DanielePucci, 2021-05-18:
This is cool guys, well done

jeljaik, 2021-05-19:
I have to say that I agree on the decision... but I may be biased 😄

PR #300 (2021-05-08): Move from doxygen to mcss documentation
Bye-bye Doxygen and welcome mcss! 

This PR implements the machinery required to generate the documentation using mcss

In detail, the machinery generates the documentation for master and a set of tags (here we generate the documentation only for v0.1.1, since v0.1.0 is not compatible with mcss)

The following video shows how the documentation will be. 

https://user-images.githubusercontent.com/16744101/117519445-cdab0780-afa3-11eb-85bb-ecbd0c70d36c.mp4

This closes #299