gbionics/bipedal-locomotion-framework #308: Bug in setParameter method if the parameter is boolean

GiulioRomualdi, 2021-05-24:
I was trying to implement the python bindings for the `YarpImplementation` of the `ParameterHandler`. While I was writing the tests I noticed that there is a bug in the `setParameter` method when a boolean is passed as parameter.

Indeed when I try to call the `getParameter` method the following error is thrown 
```
[2021-05-24 15:43:46.880] [thread: 71068] [blf] [error] [BipedalLocomotion::YarpUtilities::getElementFromSearchable] The value named: flag is not a bool.
```
https://github.com/dic-iit/bipedal-locomotion-framework/blob/baebbb6686585117d8bc3920cd8725416f116f7f/src/YarpUtilities/include/BipedalLocomotion/YarpUtilities/Helper.tpp#L79-L85

I also created a commit where the test in c++ fails https://github.com/dic-iit/bipedal-locomotion-framework/commit/4561cead4cf579096162fc7ca0b66ea860288c0d

Printing the content of the parameters handler I got the following string. Notice that `flag` is `1` and not `True`
```
(answer_to_the_ultimate_question_of_life 42) (pi 3.14000000000000012434) (John Smith) (flag 1) ("Fibonacci Numbers" (1 1 2 3 5 8 13 21))
```

This bug does not appear if the configuration is loaded in the configuration file. 

cc @traversaro and @S-Dafarra

traversaro, 2021-05-24:
I guess it is some problem of implicit overloading?

S-Dafarra, 2021-05-24:
I would suggest checking if this happens also with direct ``yarp`` calls. I suspect that this is happening when converting a boolean to string with ``to_string`` inside ``yarp``.

GiulioRomualdi, 2021-05-24:
The problem seems related to yarp: https://github.com/robotology/yarp/issues/2584

closed 2021-05-26 by GiulioRomualdi