gbionics/bipedal-locomotion-framework #310: pybind11 bindings fail to convert numpy arrays into Eigen::Quaterniond

paolo-viceconte, 2021-05-24:
While using the pybind11 byndings, I encountered issues with functions taking `Eigen::Quaterniond` objects as arguments. 

For instance, by calling [`change_fixed_frame`](https://github.com/dic-iit/bipedal-locomotion-framework/blob/baebbb6686585117d8bc3920cd8725416f116f7f/bindings/python/FloatingBaseEstimators/src/LeggedOdometry.cpp#L78-L84) with a numpy array of doubles as `frame_orientation_in_world`, I get the following: 
```
TypeError: change_fixed_frame(): incompatible function arguments. The following argument types are supported:
    1. (self: bipedal_locomotion_framework.bindings.floating_base_estimators.LeggedOdometry, frame_index: int) -> bool
    2. (self: bipedal_locomotion_framework.bindings.floating_base_estimators.LeggedOdometry, frame_index: int, frame_orientation_in_world: Eigen::Quaternion<double, 0>, frame_position_in_world: numpy.ndarray[float64[3, 1]]) -> bool
Invoked with: <bipedal_locomotion_framework.bindings.floating_base_estimators.LeggedOdometry object at 0x7f55627bb6f0>, -1, array([-8.37623766e-03, -7.89849058e-05, -2.75573432e-03,  9.99961118e-01]), array([ 1.19725312, -1.94944606,  0.        ])
```

Passing `Eigen::Quaterniond` arguments should be properly handled in order to allow python bindings usage on these functions.

Related issue (without solution): https://stackoverflow.com/questions/66126852/how-to-use-pybind11-to-convert-between-eigenquaternion-and-numpy-ndarray

cc @GiulioRomualdi

artivis, 2021-06-21:
Hi,
This might be due to the fact that `pybind11` does not provide bindings for any `Eigen/Geometry` classes. So `pybind11` has no idea how to map the array to a quaternion. I've hit a similar issue with `manifpy` and decided not to bind the API involving `Eigen/Geometry` objects (see some discussion [here](https://github.com/artivis/manif/issues/187#issuecomment-745742452)). This is unfortunate tho.

paolo-viceconte, 2021-06-22:
Hi @artivis, 
thanks for the clarification and the pointer to the cross-link discussion. 
We are temporarily handling the issue by using `Eigen::Vector4d` objects rather than `Eigen::Quaterniond` objects and then doing proper conversions after the bindings (if needed). 
If the decision is not to bind the API with `Eigen/Geometry` objects, we will keep going this way.

GiulioRomualdi, 2024-04-08:
As @artivis mentioned, `pybind11` does not provide any bindings for `Eigen/Geometry` classes. @paolo-viceconte was successfully using what he wrote in https://github.com/ami-iit/bipedal-locomotion-framework/issues/310#issuecomment-865774757. I'm going to close this issue. Thank you all 😃

closed 2024-04-08 by GiulioRomualdi