gbionics/bipedal-locomotion-framework #317: ParametersHandlerYarpUnitTests fails in macOS

GiulioRomualdi, 2021-05-27:
The `ParametersHandlerYarpUnitTests` fails on macos
```
Error: 5-26 22:13:49.935] [thread: 24362] [blf] [error] [BipedalLocomotion::YarpUtilities::getVectorFromSearchable] The element of the list associated to the value named flags is not a Boolean.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ParametersHandlerYarpUnitTests is a Catch v2.13.6 host application.
Run with -? for options

-------------------------------------------------------------------------------
Get parameters
  Get bool Vector
-------------------------------------------------------------------------------
../src/ParametersHandler/tests/ParametersHandlerYarpTest.cpp:90
...............................................................................

../src/ParametersHandler/tests/ParametersHandlerYarpTest.cpp:93: FAILED:
  REQUIRE( parameterHandler->getParameter("flags", element) )
with expansion:
  false

Parameters: (answer_to_the_ultimate_question_of_life 42) (pi 3.14000000000000012434) (John Smith) (flag true) (flags (1 0 0 1 1 1)) ("Fibonacci Numbers" (1 1 2 3 5 8 13 21))
```
https://github.com/dic-iit/bipedal-locomotion-framework/runs/2679363744#step:10:23

Accordingly to https://github.com/dic-iit/bipedal-locomotion-framework/pull/313 

```
flags (1 0 0 1 1 1))
```
should be 
```
flags (true false false true true true))
```

this means that, in macos, the following 

https://github.com/dic-iit/bipedal-locomotion-framework/blob/8e174c293113a20c6a95d8d03d2771072ab147a9/src/ParametersHandler/YarpImplementation/include/BipedalLocomotion/ParametersHandler/YarpImplementation.tpp#L86-L88

is not true when `handler.setParameter("flags", std::vector<bool>{true ..... });` is called.

I try o debug on macos in the next few days. I will use https://github.com/GiulioRomualdi/blf-fastmac

GiulioRomualdi, 2021-05-27:
Using `blf-fastmac` I printed `decltype(element)` when a `std::vector<bool>` is passed

```cpp
        for (const auto& element : parameter)
        {
          std::cerr << "type:  " << type_name<typename std::remove_cv_t<typename std::remove_reference_t<decltype(element)>>>()  << std::endl;
        }
```

The output is 

```
type: std::__1::__bit_const_reference<std::__1::vector<bool, std::__1::allocator<bool> > >
```

that is different from `bool`.

GiulioRomualdi, 2021-05-27:
Here the reason why https://stackoverflow.com/questions/34079390/range-for-loops-and-stdvectorbool

S-Dafarra, 2021-05-27:
A vector of boolean os a particular beast, maybe it deserves an if case on its own instead of checking element by element.

GiulioRomualdi, 2021-05-27:
I was writing a template specialization only for that

closed 2021-05-27 by GiulioRomualdi