gbionics/bipedal-locomotion-framework #321: Add the possibility to compute the joint velocity in RobotControl

GiulioRomualdi, 2021-06-03:
It would be nice to implement a small filter in YarpRobotControl to get the robot velocity. We may add an optional configuration parameter (e.g. `use_estimated_velocity`)  to switch from yarp velocity to the estimated one.

A possible solution is to implement an exponential filter to compute the joint velocity

cc @isorrentino @S-Dafarra @traversaro @paolo-viceconte

S-Dafarra, 2021-06-03:
I would prefer to have a separate block to filter them. In my opinion, the sensor bridge should avoid doing any computations on the input data. It is on the user to eventually add filters

traversaro, 2021-06-03:
> 
> 
> I would prefer to have a separate block to filter them. In my opinion, the sensor bridge should avoid doing any computations on the input data. It is on the user to eventually add filters

Yes, in the past we "hided" computations inside a sensor interface, and that was not a great idea as people forgot about those filtering, and they did not considered them when optimizing parameters/debugging problems.

closed 2025-03-10 by GiulioRomualdi