gbionics/bipedal-locomotion-framework #335: Unable to use the yarp devices defined in blf

GiulioRomualdi, 2021-06-11:
When I try to call `yarpdev --list` I get the following output. Notice that the other devices work fine

```
[INFO] |yarp.dev.Drivers| Here are devices listed for your system:
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so: undefined symbol: FTIMULoggerDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so: undefined symbol: FTIMULoggerDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so: undefined symbol: FTIMULoggerDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so: undefined symbol: FTIMULoggerDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so: undefined symbol: FTIMULoggerDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin FTIMULoggerDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so: undefined symbol: ROSPublisherTestDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so: undefined symbol: ROSPublisherTestDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so: undefined symbol: ROSPublisherTestDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so: undefined symbol: ROSPublisherTestDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so:
  /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/ROSPublisherTestDevice.so: undefined symbol: ROSPublisherTestDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin ROSPublisherTestDevice
[INFO] |yarp.dev.Drivers| Device "AudioPlayerWrapper", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_AudioPlayerWrapper.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "AudioRecorderWrapper", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_AudioRecorderWrapper.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "IMURosPublisher", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_IMURosPublisher.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "JoypadControlClient", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_JoypadControlClient.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "JoypadControlServer", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_JoypadControlServer.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "MagneticFieldRosPublisher", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_MagneticFieldRosPublisher.so library), is a network wrapper.
[INFO] |yarp.dev.Drivers| Device "PoseStampedRosPublisher", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/yarp_PoseStampedRosPublisher.so library), is a network wrapper.
```

cc @prashanthr05 and @traversaro

GiulioRomualdi, 2021-06-11:
I noticed adding the prefix `yarp_` to the `library` value  in `FTIMULoggerDevice.ini` (and all the other devices) the errors disappear. However the warning remains 
```
INFO] |yarp.dev.Drivers| Here are devices listed for your system:
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin FTIMULoggerDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin FloatingBaseEstimatorDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin ROSPublisherTestDevice
```

prashanthr05, 2021-06-11:
I am using a BLF build outside of the superbuild with yarp version: YARP version 3.4.100+20201111.5+git08c1f164d

And for me running a `yarpdev --list` gives,
[image]

`yarp plugin --all` shows,

[image]

doing a `cat` on the `ini` file displays,

[image]

Maybe I should try updating YARP and see if things fail.

GiulioRomualdi, 2021-06-11:
https://github.com/dic-iit/bipedal-locomotion-framework/issues/335#issuecomment-859632646 updated

traversaro, 2021-06-11:
How is this warning: https://github.com/dic-iit/bipedal-locomotion-framework/blob/9d51b52cc16b5f341aca95ac28982a9fb3470b4b/cmake/AddBipedalLocomotionYARPDevice.cmake#L38 handled?

prashanthr05, 2021-06-11:
At the moment, there are no explicit checks, just being careful about the `CONFIGURE_PACKAGE_NAME` parameter passed to the macro being different than `NAME` parameter,

https://github.com/dic-iit/bipedal-locomotion-framework/blob/b7adab2f8dc5bd448533cac7d31633c5e4926a9b/devices/examples/FTIMULoggerDevice/CMakeLists.txt#L13

traversaro, 2021-06-11:
@GiulioRomualdi can you provide the `CMakeCache.txt` of blf in your case?

traversaro, 2021-06-11:
Furthermore, which version of YARP are you using @GiulioRomualdi ?

prashanthr05, 2021-06-11:
@traversaro, is it strange that for me the ini files are being installed like,
[image]

prashanthr05, 2021-06-11:
After updating YARP to `YARP version 3.4.101+20210611.21+git5437b92dc`, I get the following error

[image]

prashanthr05, 2021-06-11:
Removing the constructor with the `explicit` keyword,
https://github.com/dic-iit/bipedal-locomotion-framework/blob/b7adab2f8dc5bd448533cac7d31633c5e4926a9b/devices/examples/FTIMULoggerDevice/src/FTIMULoggerDevice.cpp#L18-L22

fixes the error in https://github.com/dic-iit/bipedal-locomotion-framework/issues/335#issuecomment-859671639.

GiulioRomualdi, 2021-06-11:
> Furthermore, which version of YARP are you using @GiulioRomualdi ?

YARP version 3.4.2+20-20210204.2+gitb966f9f29

traversaro, 2021-06-11:
I quickly checked with the released binaries in conda:
~~~
conda create -n blf 
conda activate blf
conda install -c conda-forge -c robotology bipedal-locomotion-framework
~~~
and the problem is present also there:
~~~
(blf) gitpod@ws-840219e8-49e2-4a4b-b398-141ed14d5ae3:/workspace/gitpod-ubuntu-20.04$ yarpdev --list
[INFO] |yarp.dev.Drivers| Here are devices listed for your system:
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gitpod/mambaforge/envs/blf/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gitpod/mambaforge/envs/blf/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /home/gitpod/mambaforge/envs/blf/lib/yarp/FloatingBaseEstimatorDevice.so:
  /home/gitpod/mambaforge/envs/blf/lib/yarp/FloatingBaseEstimatorDevice.so: undefined symbol: FloatingBaseEstimatorDevice
[WARNING] |yarp.dev.Drivers| Wrong library name for plugin FloatingBaseEstimatorDevice
~~~

GiulioRomualdi, 2021-06-13:
> Removing the constructor with the `explicit` keyword,
> https://github.com/dic-iit/bipedal-locomotion-framework/blob/b7adab2f8dc5bd448533cac7d31633c5e4926a9b/devices/examples/FTIMULoggerDevice/include/BipedalLocomotion/FTIMULoggerDevice.h#L28
> 
> https://github.com/dic-iit/bipedal-locomotion-framework/blob/b7adab2f8dc5bd448533cac7d31633c5e4926a9b/devices/examples/FTIMULoggerDevice/src/FTIMULoggerDevice.cpp#L18-L22
> 
> fixes the error in [#335 (comment)](https://github.com/dic-iit/bipedal-locomotion-framework/issues/335#issuecomment-859671639).

GiulioRomualdi, 2021-06-13:
I noticed that there are these three options in the CMake configuration
```
ENABLE_FTIMULoggerDevice                                                               
ENABLE_FloatingBaseEstimatorDe   
ENABLE_ROSPublisherTestDevice 
```

by default, there are `OFF`. If I put them `ON` and I apply https://github.com/dic-iit/bipedal-locomotion-framework/issues/335#issuecomment-859674712 
```
yarpdev --list
```
The error disappears

```
[INFO] |yarp.dev.Drivers| Device "FTIMULoggerDevice", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so library).
[INFO] |yarp.dev.Drivers| Device "FloatingBaseEstimatorDevice", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so library).
```

 Why these pare are there?

GiulioRomualdi, 2021-06-13:
https://github.com/dic-iit/bipedal-locomotion-framework/pull/337 should fix the issue

traversaro, 2021-06-14:
> I noticed that there are these three options in the CMake configuration
> 
> ```
> ENABLE_FTIMULoggerDevice                                                               
> ENABLE_FloatingBaseEstimatorDe   
> ENABLE_ROSPublisherTestDevice 
> ```
> 
> by default, there are `OFF`. If I put them `ON` and I apply [#335 (comment)](https://github.com/dic-iit/bipedal-locomotion-framework/issues/335#issuecomment-859674712)
> 
> ```
> yarpdev --list
> ```
> 
> The error disappears
> 
> ```
> [INFO] |yarp.dev.Drivers| Device "FTIMULoggerDevice", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FTIMULoggerDevice.so library).
> [INFO] |yarp.dev.Drivers| Device "FloatingBaseEstimatorDevice", available on request (found in /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/yarp/FloatingBaseEstimatorDevice.so library).
> ```
> 
> Why these parameters are there?

Good catch! I was also affected by the issue in the past and I had forgot about it. The reason is that `yarp_prepare_device` defines them (see https://github.com/robotology/yarp/blob/5313b2defd8be7237567ea56eca8dde93dfc8121/cmake/YarpPlugin.cmake#L214). If the option is off, the shared library is installed normally (and this is the confusing part) but the `YARP_DEFINE_SHARED_SUBCLASS` macro is not compiled (see https://github.com/robotology/yarp/blob/5313b2defd8be7237567ea56eca8dde93dfc8121/cmake/YarpPlugin.cmake#L492) so some necessary symbols are not defined. An alternative to the fixes in https://github.com/dic-iit/bipedal-locomotion-framework/pull/337/files is simply to set `DEFAULT ON` to the `yarp_prepare_plugin`, see for example: https://github.com/robotology/whole-body-estimators/blob/master/devices/genericSensorClient/CMakeLists.txt#L10 .

closed 2021-06-14 by GiulioRomualdi

traversaro, 2021-06-14:
I opened https://github.com/robotology/yarp/issues/2606 to mitigate the risk of this happening again.

prashanthr05, 2021-06-14:
Great, thanks @traversaro !!

PR #337 (2021-06-14): Various fixes for the yarp devices
This fixes #335 

I followed this tutorial: http://www.yarp.it/git-master/add_a_plugin.html