gbionics/bipedal-locomotion-framework #343: Understand if EIGEN_DEFAULT_TO_ROW_MAJOR used in manifpy compilation is problematic in blf Python bindings

traversaro, 2021-06-21:
Not something urgent, but apparently `manifpy` is being compiled with the `EIGEN_DEFAULT_TO_ROW_MAJOR` Eigen option (see https://github.com/artivis/manif/blob/0.0.4/python/CMakeLists.txt). 

As you can see in https://gitlab.com/libeigen/eigen/-/issues/422, this option actually breaks ABI and API, so it can create problems if `Eigen::MatrixXd` or similar classes created in `manifpy` are passed to "normal" Eigen functions compiled with the standard Eigen options. In particular some matrices **may be silently transposed**. We should understand if this can create problems in the future (at the moment, I don't understand enough of the manifpy/blf python machinery to given an appropriate opinion on this) and if this can create problems, and in that case understand how to deal with them.

traversaro, 2021-06-21:
By the way, Eigen docs are quite clear that their recommendation is never to use `EIGEN_DEFAULT_TO_ROW_MAJOR` option except for testing Eigen itself, see from https://eigen.tuxfamily.org/dox/TopicPreprocessorDirectives.html : 
> These macros are mainly meant for people developing Eigen and for testing purposes. Even though, they might be useful for power users and the curious for debugging and testing purpose, they **should not be used** by real-word code.

On the other hand, manifpy links pybind11 docs, that itself do not refer to `EIGEN_DEFAULT_TO_ROW_MAJOR`, explicitly, see https://pybind11.readthedocs.io/en/stable/advanced/cast/eigen.html#storage-orders .

traversaro, 2021-06-21:
FYI @artivis: we still did not looked into this in depth and we still do not understand/know if this can be a source of problem, but nevertheless you may still like to be aware of this.

artivis, 2021-06-21:
Thanks for the heads up.
I wasn't aware of Eigen's strong recommendation against this feature. I initially developed the binding without `EIGEN_DEFAULT_TO_ROW_MAJOR` which led to an ugly and, most importantly, error-prone API (see e.g. [here](https://github.com/artivis/manif/blob/bcd10b01a6d5a02937d9f7625f65e2d5546e10d8/test/python/pytest_manif.py#L215)). I thus imagine that reverting this would be fairly straightforward if necessary. 
If I'm not mistaken the ordering only concern matrices (e.g. here Jacobians) and thus group/tangent types are not impacted - they use vectors underneath. Furthermore, the option is only enabled for the Python lib; since there are no pointer manipulation/mapping across types like on the C++ side, I suppose this won't be trouble.
I haven't had any related issue so far but you guys definitely have a more extensive use of `manifpy` than I do, so please do feel free to open a ticket if you spot anything wobbly.